bluez/bluez #4: Plz don't check whether SDP service entry exists for a HID device

boilerman, 2020-02-17:
Hello,
I found that bluez checks if SDP service entry exists for a HID device like in the following code:
`static struct input_device *find_device(const bdaddr_t *src,
					const bdaddr_t *dst)
{
	struct btd_device *device;
	struct btd_service *service;

	device = btd_adapter_find_device(adapter_find(src), dst, BDADDR_BREDR);
	if (device == NULL)
		return NULL;

	service = btd_device_get_service(device, HID_UUID);
	if (service == NULL)
		return NULL;

	return btd_service_get_user_data(service);
}`

I mean, if (service == NULL) return NULL;

As a BT HID device can work in the Boot mode without publishing the SDP service, I think you have to allow the HID without SDP service to connect, and bluez should work in the Boot mode. (BT HID has two reporting modes: boot mode vs report mode)

For your reference, I am the author of across (www.acrosscenter.com), and am about to release a Linux version. It works well mostly except in the case of trying to pair with a Windows client. Windows PC needs to connect CTRL/INTR L2cap server to pair and load the HID's driver for a BT HID. But bluez checks if SDP service entry exists or not, and as Windows PC doesn't have the HID service, bluez disconnect L2cap connections and therefore it failed to pair with a Linux PC (which works as a HID device). So we have to make a Windows PC work as a HID device (working in Across Server mode which make SDP service available) to pair with a Linux PC.

For the L2cap connections, Windows PC seems to not send meaningful requests (no data for the INTR channel) and so if bluez get some data for the CTRL channel, you could just send a handshake and disregard them. All that I am mentioning is need only for a pairing process to succeed.
Thank you!

Regards,

boilerman, 2020-02-17:
Another solution can be to provide 2 DBus methods for the HID profile: DisableListeners and EnableListeners for the CTRL/INTR channels.

sjanc, 2020-02-25:
Hi,

Bluetooth HID Profile spec is quite clear that HID devices shall implement SDP server with required services (HID and DID).

Vudentz, 2020-03-18:
Im closing this one since it in fact mandatory to have SDP record regardless of the mode it is operating.

closed 2020-03-18 by Vudentz