bluez/bluez #8: DisplayOnly IO capability does not work with BR/EDR connections

jimbroze, 2020-04-09:
When using the DisplayOnly IO capability on a
receiving device running Bluetooth 4.1 and an initiation device with
DisplayKeyboard IO capability (android phone), the pairing 
provokes a Request Confirmation and requests a yes/no confirmation on
the receiving device. This is obviously a problem when the receiving
device is only able to display and not input and has been set as such.

I suspect that this is could be caused by the conversion of an incoming device's capability from
DisplayKeyboard to DisplayYesNo on BR/EDR connections. However, I do not have a device capable of SMP to test this. As a short term fix, is there anyway to change this behavior to default to KeyboardOnly?

This was noticed on Raspbian 2020-02-13 release (Rapsbian 10, Kernel 4.19.97) running Bluez 5.54

Vudentz, 2020-04-29:
According to the spec the combination of IO capabilities shall result in a Passkey Entry authentication method, which afaik is what RequestConfirmation is about since it gives you a passkey, the agent then can display to the user, weather this is done correctly on Android or not that is out of BlueZ control, see:

BLUETOOTH CORE SPECIFICATION Version 5.2 | Vol 3, Part C
page 1317

jimbroze, 2020-04-30:
@Vudentz I don't believe any of this control is happening on the android side.

BlueZ automatically changing the KeyboardInput capability to DisplayYesNo would, in my opinion, make for an ideal configurable setting. i.e being able to choose which capability this is downgraded to. I don't know if this is a bluetooth specification requirement or something that has been implemented purely for BlueZ?

Aside from this, however, the handling of the resulting capability combination appears to be incorrect. The core specification states that an initiator with DisplayYesNO and responder with DisplayOnly results in numeric comparison with automatic confirmation on device B (Bluez). However, Bluez is not providing this automatic confirmation and is waiting for an input which it can never receive because it is Display Only.

Vudentz, 2020-04-30:
@jimbroze The kernel uses a table to translate the IO capabilities into the authentication method:

https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/tree/net/bluetooth/smp.c#n819

I don't see any part of the code doing the translation of KeyboardInput to DisplayYesNo, perhaps the result of the passkey confirmation is actually for the device A to display so the device B can enter the passkey, once that is do device A can dismiss the dialog and the link key will be authenticated, if we drop to just works method the resulting key is to be consider unauthenticated which means it is less secure than the passkey entry.

Vudentz, 2020-06-23:
Closing this since it appears to not BlueZ's fault.

closed 2020-06-23 by Vudentz