bluez/bluez #9: bluetoothd 5.54 crash upon service authorization

delleceste, 2020-04-14:
Trying to connect to a computer configured as audio sink through bluetoothctl, after pairing and trusting in both sides bluetoothd crashes

Steps to reproduce:

> bluetoothctl

[bluetooth]# connect 00:xx:xx ...

bluetoothd segmentation fault

coredumpctl debug
           PID: 4564 (bluetoothd)
           UID: 0 (root)
           GID: 0 (root)
        Signal: 11 (SEGV)
     Timestamp: Tue 2020-04-14 14:15:44 CEST (40s ago)
  Command Line: /usr/lib/bluetooth/bluetoothd
    Executable: /usr/lib/bluetooth/bluetoothd
 Control Group: /system.slice/bluetooth.service
          Unit: bluetooth.service
         Slice: system.slice
       Boot ID: b6a0f48bcf494e3684d85d5ad94ec636
    Machine ID: 520c90fd05654925aee82c084e1546d3
      Hostname: woody
       Storage: /var/lib/systemd/coredump/core.bluetoothd.0.b6a0f48bcf494e3684d85d5ad94ec636.4564.1586866544000000000000.lz4
       Message: Process 4564 (bluetoothd) of user 0 dumped core.
                
                Stack trace of thread 4564:
                #0  0x000055d74461b65d n/a (bluetoothd + 0x2365d)
                #1  0x000055d74469e1fd n/a (bluetoothd + 0xa61fd)
                #2  0x000055d74461ea2d n/a (bluetoothd + 0x26a2d)
                #3  0x000055d74461eb79 n/a (bluetoothd + 0x26b79)
                #4  0x000055d74462012e n/a (bluetoothd + 0x2812e)
                #5  0x000055d7446250fa n/a (bluetoothd + 0x2d0fa)
                #6  0x00007f5177a1b88f g_main_context_dispatch (libglib-2.0.so.0 + 0x6b88f)
                #7  0x00007f5177a1d831 n/a (libglib-2.0.so.0 + 0x6d831)
                #8  0x00007f5177a1e843 g_main_loop_run (libglib-2.0.so.0 + 0x6e843)
                #9  0x000055d7446af166 n/a (bluetoothd + 0xb7166)
                #10 0x000055d7446af5e8 n/a (bluetoothd + 0xb75e8)
                #11 0x000055d744616941 n/a (bluetoothd + 0x1e941)
                #12 0x00007f51776e7023 __libc_start_main (libc.so.6 + 0x27023)
                #13 0x000055d74461754e n/a (bluetoothd + 0x1f54e)

GNU gdb (GDB) 9.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/lib/bluetooth/bluetoothd...
(No debugging symbols found in /usr/lib/bluetooth/bluetoothd)
[New LWP 4564]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
Core was generated by `/usr/lib/bluetooth/bluetoothd'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x000055d74461b65d in ?? ()
(gdb) bt
#0  0x000055d74461b65d in ?? ()
#1  0x000055d74469e1fd in ?? ()
#2  0x000055d74461ea2d in ?? ()
#3  0x000055d74461eb79 in ?? ()
#4  0x000055d74462012e in ?? ()
#5  0x000055d7446250fa in ?? ()
#6  0x00007f5177a1b88f in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
#7  0x00007f5177a1d831 in ?? () from /usr/lib/libglib-2.0.so.0
#8  0x00007f5177a1e843 in g_main_loop_run () from /usr/lib/libglib-2.0.so.0
#9  0x000055d7446af166 in ?? ()
#10 0x000055d7446af5e8 in ?? ()
#11 0x000055d744616941 in ?? ()
#12 0x00007f51776e7023 in __libc_start_main () from /usr/lib/libc.so.6
#13 0x000055d74461754e in ?? ()
(gdb) quit

Additional info:
* package version(s)

pacman -Q --info bluez
Name            : bluez
Version         : 5.54-1
Description     : Daemons for the bluetooth protocol stack
Architecture    : x86_64
URL             : http://www.bluez.org/
Licenses        : GPL2
Groups          : None
Provides        : None
Depends On      : libical  dbus  glib2  alsa-lib  json-c
Optional Deps   : None
Required By     : bluez-qt  gnome-bluetooth  pulseaudio-bluetooth
Optional For    : networkmanager
Conflicts With  : obexd-client  obexd-server
Replaces        : None
Installed Size  : 2.27 MiB
Packager        : Andreas Radke <andyrtr@archlinux.org>
Build Date      : Mon 16 Mar 2020 12:41:49 PM CET
Install Date    : Tue 14 Apr 2020 09:43:36 AM CEST
Install Reason  : Installed as a dependency for another package
Install Script  : No
Validated By    : Signature

* config and/or log files etc.

defaults

NOTES: if you don't trust beforehand, the crash takes place as soon as you type "yes" <enter> when Authorization for service is required.

Vudentz, 2020-04-29:
The backtrace is not the helpful without debug symbols, also if possible try reproducing this with the latest since there have been some fixes for A2DP in the meantime.

Vudentz, 2020-06-23:
Closing this since there has been any response to debug it further.

closed 2020-06-23 by Vudentz