bluez/bluez #10: bluetoothctl terminal issues

dj0405, 2020-04-15:
Ubuntu 18.04
bluetoothctl version: 5.54

Initially I had an old Bluez installed (no issues), upgraded to the latest stable release 5.54 using the following instructions:

    wget http://www.kernel.org/pub/linux/bluetooth/bluez-5.54.tar.xz
    tar xf bluez-5.54.tar.xz
    cd bluez-5.54/
    ./configure --enable-library
    make
    sudo make install
    sudo systemctl enable bluetooth

The problem is bluetoothctl shows weird characters in the terminal. It still functions as expected but the characters are messed up! I attached a screenshot for your review.
[image]

Any ideas why is this happening? Missing fonts?

Thank you!

Vudentz, 2020-04-29:
I suspect this has something to do with the terminal, tilix and gnome-terminal seems to work properly.

AndresRReina, 2021-03-08:
I also get the funny chars next to color codes. Connecting through ssh to RPI4 running Ubuntu 20. My terminal is Terminator. I've never seen this behavior with any other command neither local or remote.

Vudentz, 2021-03-08:
This seems to be with readline or the terminal not really supporting RL_PROMPT_START_IGNORE/RL_PROMPT_END_IGNORE: https://wiki.hackzine.org/development/misc/readline-color-prompt.html

AndresRReina, 2021-03-09:
You are right. After reading about it, I understand readline  should be eating those characters. Do you know why would it not do so? I'm only using packages from the Ubuntu 20 repository, I think readline should be up to date then.

dpward, 2021-08-16:
> gnome-terminal seems to work properly.

FWIW this does affect gnome-terminal under RHEL 8.

noahfriedman, 2022-03-17:
It's completely inappropriate to hardcode terminal escape sequences directly into any application instead of using a terminal capability library like ncurses to look up the correct sequences, or whether the terminal is capable of color at all.

I mostly interact with bluetoothctl with dumb terminals and the escape sequence noise has been irritating me for **years**

macmpi, 2023-10-12:
@Vudentz would it be possible to have an option which disable overall coloring and escape sequences ([NO_COLOR](https://no-color.org/) or al)?
This makes scripting (`expect`/`sexpect`) unnecessarily painfull.
Thanks for consideration.

Vudentz, 2023-10-24:
> @Vudentz would it be possible to have an option which disable overall coloring and escape sequences ([NO_COLOR](https://no-color.org/) or al)? This makes scripting (`expect`/`sexpect`) unnecessarily painfull. Thanks for consideration.

Yeah, or we need some means to detect if the underline terminal supports colors or not.

Vudentz, 2023-10-24:
How about this:

```
diff --git a/src/shared/shell.h b/src/shared/shell.h
index 87fb5c415f20..a9a635bda959 100644
--- a/src/shared/shell.h
+++ b/src/shared/shell.h
@@ -10,14 +10,14 @@
 #include <getopt.h>
 #include <stdbool.h>
 
-#define COLOR_OFF      "\001\x1B[0m\002"
-#define COLOR_RED      "\001\x1B[0;91m\002"
-#define COLOR_GREEN    "\001\x1B[0;92m\002"
-#define COLOR_YELLOW   "\001\x1B[0;93m\002"
-#define COLOR_BLUE     "\001\x1B[0;94m\002"
-#define COLOR_BOLDGRAY "\001\x1B[1;30m\002"
-#define COLOR_BOLDWHITE        "\001\x1B[1;37m\002"
-#define COLOR_HIGHLIGHT        "\001\x1B[1;39m\002"
+#define COLOR_OFF      "\x1B[0m"
+#define COLOR_RED      "\x1B[0;91m"
+#define COLOR_GREEN    "\x1B[0;92m"
+#define COLOR_YELLOW   "\x1B[0;93m"
+#define COLOR_BLUE     "\x1B[0;94m"
+#define COLOR_BOLDGRAY "\x1B[1;30m"
+#define COLOR_BOLDWHITE        "\x1B[1;37m"
+#define COLOR_HIGHLIGHT        "\x1B[1;39m"
```

noahfriedman, 2023-10-25:
Again, I have to ask, do you not understand what ncurses/terminfo is for?  Terminal capabilities are only 40-year old technology.

Vudentz, 2023-10-25:
> Again, I have to ask, do you not understand what ncurses/terminfo is for? Terminal capabilities are only 40-year old technology.

We won't be rewriting bluetoothctl using ncurses anytime soon, we are using libreadline which doesn't seem to handle the use of colors that well (https://wiki.hackzine.org/development/misc/readline-color-prompt.html), or if you have a better idea please enlight us.

Btw, I get that you may be frustated that this issue has been known for a while, but I don't think you are really helping by stating it is a 40-year old technology, etc, specially because it makes no difference if readline can't really calculate the prompt length properly without the use of escape sequences when colors are used.

macmpi, 2023-10-27:
> How about this:

Could not test it yet, sorry.
However, wouldn't it be possible to simply filter-out all those color codes if a certain parameter is passed or env variable set (like NO_COLOR)?

closed 2023-11-20 by BluezTestBot