bluez/bluez #11: RFE: synthesize hid device for fido2 tokens

poettering, 2020-05-12:
Hmm, so bluez already does some hid stuff. It would make a ton of sense I think if it would also synthesize hid devices for fido2 tokens. Maybe that's something that is interesting to the bluez project?

Eventually Android apparently will expose a fido2 bluetooth token, and it would be awesome to hook that up to any bluez running system.

There's prior code for this:

https://github.com/freeu2f/u2fhid2bt

And I think it would be excellent if bluez would just support this out of the box?

/cc @npmccallum

npmccallum, 2020-05-12:
I'm happy to relicense `u2fhid2bt` code if you want to use any of it. That being said. It isn't particularly *good* code. It was just an experiment to see if it was possible. It is.

The only outstanding issue is this one: https://github.com/freeu2f/u2fhid2bt/blob/master/uhid.c#L312

Vudentz, 2020-05-13:
I would have no problem adding a plugin for it, we already have uhid abstration that we use for HoG so it should be pretty easy to have this being handled inside bluetoothd.

@npmccallum Is 0000fffd-0000-1000-8000-00805f9b34fb UUID reserved for use only on fido2 tokens?

npmccallum, 2020-05-24:
@Vudentz

The [FIDO Alliance](https://fidoalliance.org) created a series of [specifications](https://fidoalliance.org/specifications). Originally, these were called U2F (v1.0, v1.1 and v1.2) and were widely implemented by YubiKeys and other hardware tokens. This was designed to make authentication in the browser easier and more secure. Because browsers didn't want to write drivers for every combination of hardware and OS, instead they merely framed APDU (i.e. smartcard) commands over USB HID. This achieved fairly wide success.

Extensions to the U2F protocol were added, including NFC and Bluetooth LE to increase usage on mobile devices. These new transports merely framed the same APDU messages in a different way for each technology.

This is where `u2fhid2bt` comes in. For each BLE FIDO device announced by `bluez`, we create an emulated UHID device. Since the "upper" protocol is the same and only the "lower" transport differs, we just proxy the UHID requests over the BLE framing. It all pretty much just works.

Ultimately, the "upper" part of the specification needed improvement. They split the "upper" protocol into two pieces: WebAuthn (standardized by the w3c) and CTAP (standardized by FIDO; a.k.a. FIDO2). Although CTAP changes the "upper" protocol extensively (to use CBOR rather than APDU), the "lower" framing is (I think...) unchanged.

Therefore, although this is designed for U2F v1.2, it should also work for FIDO2. However, there are no FIDO2 bluetooth devices yet, so I can't test this.

Now, to answer your specific question, `0000fffd-0000-1000-8000-00805f9b34fb` is the long form of the short UUID `0xfffd`. It is defined here: https://fidoalliance.org/specs/fido-v2.0-ps-20190130/fido-client-to-authenticator-protocol-v2.0-ps-20190130.html#ble-fido-service

This UUID remains unchanged from U2F and should work for all U2F and FIDO2 (CTAP) devices since the framing remains (I think...) unchanged.

alainmic, 2020-05-25:
No problem for adding a plugin for this as well.

@Luiz, FFFD is allocated to the FIDO alliance for this:
https://www.bluetooth.com/specifications/assigned-numbers/16-bit-uuids-for-sdos/