bluez/bluez #12: Unable to connect multiple BLE devices

ramondeklein, 2020-05-13:
I am using BlueZ on a Raspberry PI to provide some GATT services. It seems that only one BLE phone can connect to my GATT service. The second BLE phone can only connect, when the first one disconnects. Is this a BlueZ limitation?

When using [Paypal Gatt](https://github.com/paypal/gatt/blob/master/examples/server_lnx.go#L21) library, I can make multiple simultaneous connections. I would rather use the BlueZ implementation, because I don't want to use Go.

Vudentz, 2020-05-13:
That is normally controlled by the kernel which uses the controller states to determine if we can continue to advertise while already connected:

https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/tree/net/bluetooth/hci_request.c#n1234

ramondeklein, 2020-05-14:
We're on an older kernel and it bails out when there is a single connection:
http://git.toradex.com/cgit/linux-toradex.git/tree/net/bluetooth/hci_request.c?h=toradex_4.14-2.0.x-imx#n925. I guess the [Paypal Gatt](https://github.com/paypal/gatt/blob/master/examples/server_lnx.go#L21) library bypasses the kernel and directly manages the interface. That library has been abandoned years ago, so I don't want to use it. We'll check on a desktop PC to try and make it work and consider backporting it.

Thanks for your support. It's very much appreciated.

ramondeklein, 2020-05-14:
We also tried it on a PC with the Ubuntu 20.04 that has the `is_advertising_allowed` check. It is running Linux kernel v5.4.0-29 and BlueZ v5.53 and we're still not able to connect. Do we need an even newer kernel or am I missing something else?

Vudentz, 2020-05-14:
Well if the controller is saying it doesn't support there is nothing we can do, you can check what is the supported states the controller has with use of btmon and then forcing a reset:

< HCI Command: LE Read Supported States (0x08|0x001c) plen 0                                                                                                                              #33 [hci0] 10:25:17.077389
> HCI Event: Command Complete (0x0e) plen 12                                                                                                                                              #34 [hci0] 10:25:17.078268
      LE Read Supported States (0x08|0x001c) ncmd 1
        Status: Success (0x00)
        States: 0x000003ffffffffff
...
          Connectable Advertising State
            and Connection State (Slave Role)
            and Master Role & Slave Role

If these states are not supported then is_advertising_allowed will fail, btw if I recall this correctly you will need to trigger advertise again in order to be able to receive more connections.

ramondeklein, 2020-05-15:
We have upgraded our embedded device ([Colibri i.MX 6ULL](https://www.toradex.com/computer-on-modules/colibri-arm-family/nxp-imx6ull)) to Linux kernel v5.4.28 and it reports states **0x000003ffffffffff**. We are now able to connect multiple clients to our GATT service, so it works on this hardware and kernel version. We will upgrade our kernel and BlueZ version in our final firmware.

I also tested it on an older Toshiba Tecra laptop with Ubuntu 20.04 LTS (kernel 5.4.0), but it reports only flags **0x000000000000ffff** and doesn't work.

**For users that want to determine the flags**

Start `btmon` and open another window and open `bluetoothctl`. Select the proper controller (if required) and type `power off` and `power on`.

This is the report of the Colibri device that supports multiple connections:
```
< HCI Command: LE Read Supported States (0x08|0x001c) plen 0                                                                                                                                                            #31 [hci0] 21.362934
> HCI Event: Command Complete (0x0e) plen 12                                                                                                                                                                            #32 [hci0] 21.364039
      LE Read Supported States (0x08|0x001c) ncmd 1
        Status: Success (0x00)
        States: 0x000003ffffffffff
          Non-connectable Advertising State
          Scannable Advertising State
          Connectable Advertising State
          High Duty Cycle Directed Advertising State
          Passive Scanning State
          Active Scanning State
          Initiating State
            and Connection State (Master Role)
          Connection State (Slave Role)
          Non-connectable Advertising State
            and Passive Scanning State
          Scannable Advertising State
            and Passive Scanning State
          Connectable Advertising State
            and Passive Scanning State
          High Duty Cycle Directed Advertising State
            and Passive Scanning State
          Non-connectable Advertising State
            and Active Scanning State
          Scannable Advertising State
            and Active Scanning State
          Connectable Advertising State
            and Active Scanning State
          High Duty Cycle Directed Advertising State
            and Active Scanning State
          Non-connectable Advertising State
            and Initiating State
          Scannable Advertising State
            and Initiating State
          Non-connectable Advertising State
            and Connection State (Master Role)
          Scannable Advertising State
            and Connection State (Master Role)
          Non-connectable Advertising State
            and Connection State (Slave Role)
          Scannable Advertising State
            and Connection State (Slave Role)
          Passive Scanning State
            and Initiating State
          Active Scanning State
            and Initiating State
          Passive Scanning State
            and Connection State (Master Role)
          Active Scanning State
            and Connection State (Master Role)
          Passive Scanning State
            and Connection State (Slave Role)
          Active Scanning State
            and Connection State (Slave Role)
          Initiating State
            and Connection State (Master Role)
            and Master Role & Master Role
          Low Duty Cycle Directed Advertising State
          Low Duty Cycle Directed Advertising State
            and Passive Scanning State
          Low Duty Cycle Directed Advertising State
            and Active Scanning State
          Connectable Advertising State
            and Initiating State
            and Master Role & Slave Role
          High Duty Cycle Directed Advertising State
            and Initiating State
            and Master Role & Slave Role
          Low Duty Cycle Directed Advertising State
            and Initiating State
            and Master Role & Slave Role
          Connectable Advertising State
            and Connection State (Master Role)
            and Master Role & Slave Role
          High Duty Cycle Directed Advertising State
            and Connection State (Master Role)
            and Master Role & Slave Role
          Low Duty Cycle Directed Advertising State
            and Connection State (Master Role)
            and Master Role & Slave Role
          Connectable Advertising State
            and Connection State (Slave Role)
            and Master Role & Slave Role
          High Duty Cycle Directed Advertising State
            and Connection State (Slave Role)
            and Slave Role & Slave Role
          Low Duty Cycle Directed Advertising State
            and Connection State (Slave Role)
            and Slave Role & Slave Role
          Initiating State
            and Connection State (Slave Role)
            and Master Role & Slave Role
```

closed 2020-05-18 by Vudentz