bluez/bluez #13: Incomplete A2DP endpoint capabilities cached due to incorrect session version

smunaut, 2020-05-13:
When first connecting a device, the A2DP session version with be 0x0100 (because the SDP aren't available yet IIUC).

This means that during the discover phase all the end points are going to be checked with GetCapabilities instead of GetAllCapabilities (if the device is AVDTP 1.3 compliant) and so some capabilities will be missing (like Delay reporting). And those incomplete caps will be cached and thus the device will not be queried again.

On subsequent connections the session->version will be 0x0103 but then it's too late, the endpoint capabilities will be taken from the cache and not re-queried with GetAllCapabilities.

Vudentz, 2020-05-18:
Proposed patches:

https://lore.kernel.org/linux-bluetooth/20200518205636.139184-1-luiz.dentz@gmail.com/T/#t

Vudentz, 2020-05-21:
Merged.

closed 2020-05-21 by Vudentz