bluez/bluez #17: Can't set volume of Airpods Pro

mbirth, 2020-06-13:
I had an issue where connected Airpod Pro bluetooth headphones only ran at ~50% volume. I tried all "tips" I found on the Internet and the only one that worked was [this one](https://unix.stackexchange.com/a/562381/123507). It seems there is a problem with [`volume_exists`](/bluez/bluez/blob/master/profiles/audio/transport.c#L632) so it doesn't resolve to TRUE and thus no volume control is possible.

Once this is patched to return TRUE, setting and querying the volume of the Airpod Pro works great. From what I could gather this problem also occurs with other headsets (Sony, beats, etc.).

Vudentz, 2020-06-23:
Usually this happens if the device don't support AVRCP absolute volume control feature, there maybe other ways to control the volume using other profiles, AT command, etc, but afaik they are not part of standard, that said it would be great to have the HCI traces when connecting so we can analyse if that is really happening or there is other problem going on.

mbirth, 2020-06-24:
Here you go: [hcitrace.zip](https://github.com/bluez/bluez/files/4822618/hcitrace.zip)

This is the whole trace for opening the Airpods case and connecting to them using Blueman (right-click --> Connect).

vexcat, 2020-06-25:
If it helps, here are the profiles/services offered by my Airpods:
```
[AirPods]# info E4:76:84:90:36:17
Device E4:76:84:90:36:17 (public)
	Name: AirPods
	Alias: AirPods
	Class: 0x00240418
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Service Discovery Serve.. (00001000-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (74ec2172-0bad-4d01-8f77-997b2be0722a)
	Modalias: bluetooth:v004Cp200Fd416C
```
[Making volume_exists always return true](https://unix.stackexchange.com/questions/437468/bluetooth-headset-volume-too-low-only-in-arch/562381#562381) also works for me (I wrote this answer), allowing the setting of volume on `/org/bluez/hci0/dev_E4_76_84_90_36_17/fd0`'s `MediaTransport1` interface which does not otherwise work despite the AirPods' AVRCP Target support. This also happens on my Google Home Mini which supports the same AVRCP Target profile. It would be nice to see this be properly fixed, it seems that many others are having the same issues with "quiet airpods" on their linux machines.

hadess, 2020-06-26:
`AT+BRSF` is the AT command to use to change the volume on Apple accessories: https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf

That might need to be implemented directly in PulseAudio rather than in bluez. This hack shows how the battery status would be implemented, which also uses AT commands:
https://github.com/sre/pulseaudio/commit/d66b66d20e9bc73e6d0ca89283cf2b5675304b00

dimidagd, 2020-07-15:
> If it helps, here are the profiles/services offered by my Airpods:
>
> ```
> [AirPods]# info E4:76:84:90:36:17
> Device E4:76:84:90:36:17 (public)
> 	Name: AirPods
> 	Alias: AirPods
> 	Class: 0x00240418
> 	Icon: [...]
This is the output after hexediting bluetoothd, restarting bluetooth service and sending the dbus command

` dd@dd-Hpbook  ~  dbus-send --print-reply --system --dest=org.bluez /org/bluez/hci0/dev_80:82:23:A3:14:8C/fd0 org.freedesktop.DBus.Properties.Set string:org.bluez.MediaTransport1 string:Volume variant:uint16:10
dbus[42846]: arguments to dbus_message_new_method_call() were incorrect, assertion "_dbus_check_is_valid_path (path)" failed in file ../../../dbus/dbus-message.c line 1366.
This is normally a bug in some application using the D-Bus library.

  D-Bus not built with -rdynamic so unable to print a backtrace
[1]    42846 abort (core dumped)  dbus-send --print-reply --system --dest=org.bluez    string:Volume 
`

SebiderSushi, 2020-08-30:
According to my tests which might be flawed version `5.54` has this issue, however if i build from master@4eae282b706a2f38 it is possible to read/write the Volume attribute of my AirPods:
```
dbus-send --print-reply --system --dest=org.bluez \
   /org/bluez/hci1/dev_<airpods mac address>/sep1/fd0 \
   org.freedesktop.DBus.Properties.Set \
   string:org.bluez.MediaTransport1 \
   string:Volume variant:uint16:111 \
```

Could [d1f3beaaf1](https://github.com/bluez/bluez/commit/d1f3beaaf1949e41fbceba7b56b410bf2fe64879#diff-300ff51326f180e4b69ba502dbc1343cL637) have fixed this?

Vudentz, 2020-09-01:
> According to my tests which might be flawed version `5.54` has this issue, however if i build from master@ [4eae282](https://github.com/bluez/bluez/commit/4eae282b706a2f38d30968599945331413a92a0d) it is possible to read/write the Volume attribute of my AirPods:
> 
> ```
> dbus-send --print-reply --system --dest=org.bluez \
>    /org/bluez/hci1/dev_<airpods mac address>/sep1/fd0 \
>    org.freedesktop.DBus.Properties.Set \
>    string:org.bluez.MediaTransport1 \
>    string:Volume variant:uint16:111 \
> ```
> 
> Edit: If i revert [df7d3fa](https://github.com/bluez/bluez/commit/df7d3fa5002373f7f955219defc7d0bc794a5a6c), then the issue reappears
> 
> Re: PulseAudio
> Do they even implement any sort of Bluetooth Absolute Volume? Never seen this while using any of my bluetooth devices.

Afaik PulseAudio does not support this yet, so the volume will be applied locally to the stream itself rather than at the headset side.

Vudentz, 2020-09-01:
Closing this since upstream already contains the fixes to change the volume via MediaTransport1.

closed 2020-09-01 by Vudentz

SebiderSushi, 2020-09-01:
> > Re: PulseAudio
> > Do they even implement any sort of Bluetooth Absolute Volume? Never seen this while using any of my bluetooth devices.
> 
> Afaik PulseAudio does not support this yet, so the volume will be applied locally to the stream itself rather than at the headset side.

Okay, thank you!

mehulpanchal007, 2020-10-25:
I'm still having this issue on Fedora 32 and I can't find a way to send dbus command because no /org directory exists in Fedora 32

SebiderSushi, 2021-01-26:
@mehulpanchal007 are you referring to an `/org` directory in the filesystem?

You can tinker with your D-Bus using the "D-Bus debugger for GNOME". It is packaged as `d-feet` on virtually every popular Linux Distribution.

This is actually how i first set that Volume attribute for my AirPods IIRC.

[Screenshot of how this looks like](https://user-images.githubusercontent.com/23618858/105803398-b10f3480-5f9d-11eb-8cda-a7358715c69f.png)

SebiderSushi, 2021-01-26:
> Afaik PulseAudio does not support this yet, so the volume will be applied locally to the stream itself rather than at the headset side.

Also for anyone interested this is the relevant Merge Request for PulseAudio Bluetooth Absolute Volume feature which is actively being worked on: https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/239