bluez/bluez #18: Sennheiser Momentum True Wireless 2 (MTW2) not setting the Player Settings bits and crashing

rbergami, 2020-06-21:
Sennheiser Momentum True Wireless 2 headset are paired correctly but crash after connection.

This is a log of 

bluez-5.54-1.fc32.x86_64
On Fedora 32

Same issues as reported at:
https://www.spinics.net/lists/linux-bluetooth/msg84920.html

I can connect the headphones if I disable AVRCP by adding --noplugin=avrcp to bluetoothd command line.

rbergami, 2020-06-21:
@Vudentz

rbergami, 2020-06-21:
This is related to  #b7fe84c4582a54e098c0f52500d0ed7396026d05

Lastique, 2020-06-21:
As noted in the kernel bugzilla:

> Should be fixed by bluez commits b7fe84c4582a54e098c0f52500d0ed7396026d05 and ba91d0ad7379fc7b064af911229e50ffa1a16169.

PS: Is this GitHub project the right place for reporting issues in bluez?

rbergami, 2020-06-21:
i can confirm the patch works fine. I rebuilt the RPMs from source on fedora 32 applying the two commits below. 
I however still see a lot of message on the bluetooth service and can't enable a2dp source on the headset.
```

Jun 21 15:50:53 lnxbox bluetoothd[1108]: Bluetooth management interface 1.15 initialized
Jun 21 15:51:11 lnxbox bluetoothd[1108]: Endpoint registered: sender=:1.119 path=/MediaEndpoint/A2DPSink/sbc
Jun 21 15:51:11 lnxbox bluetoothd[1108]: Endpoint registered: sender=:1.119 path=/MediaEndpoint/A2DPSource/sbc
Jun 21 15:51:31 lnxbox bluetoothd[1108]: Unable to get Headset Voice gateway SDP record: Device or resource busy
Jun 21 15:51:31 lnxbox bluetoothd[1108]: connect error: Device or resource busy (16)
Jun 21 15:51:31 lnxbox bluetoothd[1108]: a2dp-source profile connect failed for 00:1B:66:B6:05:A1: Device or resource busy
Jun 21 15:51:33 lnxbox bluetoothd[1108]: Unable to get Headset Voice gateway SDP record: Device or resource busy
Jun 21 15:51:33 lnxbox bluetoothd[1108]: connect error: Device or resource busy (16)
Jun 21 15:51:33 lnxbox bluetoothd[1108]: a2dp-source profile connect failed for 00:1B:66:B6:05:A1: Device or resource busy
Jun 21 15:51:34 lnxbox bluetoothd[1108]: Unable to get Headset Voice gateway SDP record: Device or resource busy
Jun 21 15:51:34 lnxbox bluetoothd[1108]: connect error: Device or resource busy (16)
Jun 21 15:51:34 lnxbox bluetoothd[1108]: a2dp-source profile connect failed for 00:1B:66:B6:05:A1: Device or resource busy
Jun 21 15:51:35 lnxbox bluetoothd[1108]: Unable to get Headset Voice gateway SDP record: Device or resource busy
Jun 21 15:51:35 lnxbox bluetoothd[1108]: connect error: Device or resource busy (16)
Jun 21 15:51:35 lnxbox bluetoothd[1108]: a2dp-source profile connect failed for 00:1B:66:B6:05:A1: Device or resource busy
Jun 21 15:51:36 lnxbox bluetoothd[1108]: Unable to get Headset Voice gateway SDP record: Device or resource busy
Jun 21 15:51:36 lnxbox bluetoothd[1108]: connect error: Device or resource busy (16)
Jun 21 15:51:36 lnxbox bluetoothd[1108]: a2dp-source profile connect failed for 00:1B:66:B6:05:A1: Device or resource busy
Jun 21 15:51:37 lnxbox bluetoothd[1108]: Unable to get Headset Voice gateway SDP record: Device or resource busy
Jun 21 15:51:37 lnxbox bluetoothd[1108]: connect error: Device or resource busy (16)
Jun 21 15:51:37 lnxbox bluetoothd[1108]: a2dp-source profile connect failed for 00:1B:66:B6:05:A1: Device or resource busy
Jun 21 15:51:45 lnxbox bluetoothd[1108]: /org/bluez/hci0/dev_00_1B_66_B6_05_A1/sep2/fd0: fd(69) ready
Jun 21 15:52:11 lnxbox bluetoothd[1108]: Unable to get connect data for Headset Voice gateway: getpeername: Transport endpoint is not connected (107)
Jun 21 15:52:11 lnxbox bluetoothd[1108]: a2dp-source profile connect failed for 00:1B:66:B6:05:A1: Device or resource busy
Jun 21 15:52:30 lnxbox bluetoothd[1108]: Unable to get Headset Voice gateway SDP record: Device or resource busy
Jun 21 15:52:30 lnxbox bluetoothd[1108]: connect error: Device or resource busy (16)
Jun 21 15:52:30 lnxbox bluetoothd[1108]: a2dp-source profile connect failed for 00:1B:66:B6:05:A1: Device or resource busy
Jun 21 15:52:32 lnxbox bluetoothd[1108]: Unable to get Headset Voice gateway SDP record: Device or resource busy
Jun 21 15:52:32 lnxbox bluetoothd[1108]: connect error: Device or resource busy (16)
Jun 21 15:52:32 lnxbox bluetoothd[1108]: a2dp-source profile connect failed for 00:1B:66:B6:05:A1: Device or resource busy
Jun 21 15:52:35 lnxbox bluetoothd[1108]: connect error: Device or resource busy (16)
lines 24445-24496/24496 (END)
```

rbergami, 2020-06-21:
@Lastique can you email me? If you own this headset I'd like to discuss how to solve the issues on a2dp-source.

Lastique, 2020-06-21:
I do have the headset, but I don't use a2dp-source. I can provide some testing, but I'm not a bluez developer, nor am I knowledgeable in Bluetooth specs. We can discuss here, if my assistance is needed.

Vudentz, 2020-06-22:
> As noted in the kernel bugzilla:
> 
> > Should be fixed by bluez commits [b7fe84c](https://github.com/bluez/bluez/commit/b7fe84c4582a54e098c0f52500d0ed7396026d05) and [ba91d0a](https://github.com/bluez/bluez/commit/ba91d0ad7379fc7b064af911229e50ffa1a16169).
> 
> PS: Is this GitHub project the right place for reporting issues in bluez?

@rbergami @Lastique We are testing out if github would be more convenient to report issues, that said we patches are already applied so I wonder if the problem is still reproducible with them or the problem still persist?

Lastique, 2020-06-22:
@Vudentz The headset works for me with the two patches applied in A2DP sink mode (i.e. I can hear audio through the headset).

I'm not sure how to test the A2DP source mode. When I select "A2DP Source: SBC" in pactl, nothing happens, the MTW2 device doesn't appear in the PulseAudio input devices. In syslog I can only see this:

```
Jun 22 21:29:08 hostname pulseaudio[2001]: Refused to switch profile to a2dp_source_sbc: Not connected
```

No messages from bluetoothd. BTW, I'm using a patched PulseAudio Bluetooth module from [this](https://github.com/EHfive/pulseaudio-modules-bt) project.

I think, A2DP source problem needs to be reported in a separate issue. Though I'm not sure, where that issue belongs to - PulseAudio or bluez.

rbergami, 2020-06-22:
@Vudentz patches work like a charm. Thanks! 

> I think, A2DP source problem needs to be reported in a separate issue. Though I'm not sure, where that issue belongs to - PulseAudio or bluez.

Seems PulseAudio AFAIK

Vudentz, 2020-09-01:
@rbergami can we close this one or there is still something wrong at BlueZ side?

rbergami, 2020-09-02:
@Vudentz yes please. Thanks for the support.

closed 2020-09-02 by rbergami