bluez/bluez #19: bluetoothd crashes with Logitech MX Keys (with a fix already applied)

levaidaniel, 2020-06-27:
I've had reported a similar issue a couple of days back for which [Luiz Augusto von Dentz issued a fix](https://lore.kernel.org/linux-bluetooth/20200625211125.2239435-1-luiz.dentz@gmail.com/T/#u), and now I'm using that patch on top of bluez 5.54.
Most of the time now *bluetoothd* doesn't crash when I turn on the keyboard, but unfortunately it still happens sometimes.

The backtrace and the logs follow:

```
Reading symbols from /usr/lib/bluetooth/bluetoothd...
[New LWP 3970196]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
Core was generated by `/usr/lib/bluetooth/bluetoothd'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  gatt_db_unref (db=0x36413a43442f6874) at src/shared/gatt-db.c:420
420             if (__sync_sub_and_fetch(&db->ref_count, 1))
(gdb) list
415     void gatt_db_unref(struct gatt_db *db)
416     {
417             if (!db)
418                     return;
419
420             if (__sync_sub_and_fetch(&db->ref_count, 1))
421                     return;
422
423             gatt_db_destroy(db);
424     }
(gdb) bt
#0  gatt_db_unref (db=0x36413a43442f6874) at src/shared/gatt-db.c:420
#1  gatt_db_unref (db=0x36413a43442f6874) at src/shared/gatt-db.c:415
#2  0x0000005579b9dda8 in batt_reset (batt=0x5586490dd0) at profiles/battery/battery.c:86
#3  0x0000005579c11938 in complete_notify_request (data=0x55864967c0) at src/shared/gatt-client.c:1563
#4  enable_ccc_callback (opcode=1 '\001', pdu=0x0, length=0, user_data=0x55864967c0) at src/shared/gatt-client.c:1637
#5  0x0000005579c0b47c in disc_att_send_op (data=0x55864abeb0) at src/shared/att.c:417
#6  0x0000005579c05c18 in queue_remove_all (queue=<optimized out>, function=function@entry=0x0, user_data=user_data@entry=0x0, destroy=destroy@entry=0x5579c0b450 <disc_att_send_op>) at src/shared/queue.c:353
#7  0x0000005579c0d7b8 in disconnect_cb (io=<optimized out>, user_data=0x55864afae0) at src/shared/att.c:635
#8  0x0000005579c1a6a0 in watch_callback (channel=<optimized out>, cond=<optimized out>, user_data=<optimized out>) at src/shared/io-glib.c:170
#9  0x0000007fba32466c in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
#10 0x0000007fba324af8 in ?? () from /usr/lib/libglib-2.0.so.0
#11 0x0000007fba324f00 in g_main_loop_run () from /usr/lib/libglib-2.0.so.0
#12 0x0000005579c1ae30 in mainloop_run () at src/shared/mainloop-glib.c:79
#13 0x0000005579c1b34c in mainloop_run_with_signal (func=<optimized out>, user_data=0x0) at src/shared/mainloop-notify.c:201
#14 0x0000005579b64c3c in main (argc=<optimized out>, argv=<optimized out>) at src/main.c:770
(gdb) bt full
#0  gatt_db_unref (db=0x36413a43442f6874) at src/shared/gatt-db.c:420
No locals.
#1  gatt_db_unref (db=0x36413a43442f6874) at src/shared/gatt-db.c:415
No locals.
#2  0x0000005579b9dda8 in batt_reset (batt=0x5586490dd0) at profiles/battery/battery.c:86
No locals.
#3  0x0000005579c11938 in complete_notify_request (data=0x55864967c0) at src/shared/gatt-client.c:1563
        notify_data = 0x55864967c0
#4  enable_ccc_callback (opcode=1 '\001', pdu=0x0, length=0, user_data=0x55864967c0) at src/shared/gatt-client.c:1637
        notify_data = 0x55864967c0
        __PRETTY_FUNCTION__ = "enable_ccc_callback"
#5  0x0000005579c0b47c in disc_att_send_op (data=0x55864abeb0) at src/shared/att.c:417
        op = 0x55864abeb0
#6  0x0000005579c05c18 in queue_remove_all (queue=<optimized out>, function=function@entry=0x0, user_data=user_data@entry=0x0, destroy=destroy@entry=0x5579c0b450 <disc_att_send_op>) at src/shared/queue.c:353
        tmp = 0x558648f6a0
        entry = 0x0
        count = 9
#7  0x0000005579c0d7b8 in disconnect_cb (io=<optimized out>, user_data=0x55864afae0) at src/shared/att.c:635
        chan = 0x55864afae0
        att = 0x55864ac6e0
        err = 110
        len = 4
#8  0x0000005579c1a6a0 in watch_callback (channel=<optimized out>, cond=<optimized out>, user_data=<optimized out>) at src/shared/io-glib.c:170
        watch = <optimized out>
        result = <optimized out>
        destroy = <optimized out>
#9  0x0000007fba32466c in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
No symbol table info available.
#10 0x0000007fba324af8 in ?? () from /usr/lib/libglib-2.0.so.0
No symbol table info available.
#11 0x0000007fba324f00 in g_main_loop_run () from /usr/lib/libglib-2.0.so.0
No symbol table info available.
#12 0x0000005579c1ae30 in mainloop_run () at src/shared/mainloop-glib.c:79
No locals.
#13 0x0000005579c1b34c in mainloop_run_with_signal (func=<optimized out>, user_data=0x0) at src/shared/mainloop-notify.c:201
        data = 0x558647e170
        io = 0x558647f330
        ret = <optimized out>
#14 0x0000005579b64c3c in main (argc=<optimized out>, argv=<optimized out>) at src/main.c:770
        context = <optimized out>
        err = 0x0
        sdp_flags = <optimized out>
        gdbus_flags = <optimized out>
        __func__ = "main"
```

```
jún 27 09:17:04 bluetoothd[540701]: Endpoint registered: sender=:1.53 path=/MediaEndpoint/A2DPSource/sbc
jún 27 09:17:04 bluetoothd[540701]: Endpoint registered: sender=:1.53 path=/MediaEndpoint/A2DPSink/sbc
jún 27 09:17:04 bluetoothd[540701]: Failed to set privacy: Rejected (0x0b)
jún 27 09:17:04 bluetoothd[540701]: Bluetooth management interface 1.16 initialized
jún 27 09:17:04 bluetoothd[540701]: Starting SDP server
jún 27 09:17:04 systemd[1]: Started Bluetooth service.
jún 27 09:17:04 bluetoothd[540701]: Bluetooth daemon 5.54
jún 27 09:17:04 systemd[1]: Starting Bluetooth service...
jún 27 09:17:04 systemd[1]: Stopped Bluetooth service.
jún 27 09:17:04 systemd[1]: bluetooth.service: Scheduled restart job, restart counter is at 3.
jún 27 09:17:04 systemd[1]: bluetooth.service: Failed with result 'core-dump'.
jún 27 09:17:04 systemd[1]: bluetooth.service: Main process exited, code=dumped, status=11/SEGV
jún 27 09:17:03 bluetoothd[3970196]: [57B blob data]
jún 27 09:17:03 bluetoothd[3970196]: Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error
jún 27 09:17:03 bluetoothd[3970196]: Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
jún 27 09:17:03 bluetoothd[3970196]: Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
jún 27 09:17:03 bluetoothd[3970196]: Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
jún 27 09:17:03 bluetoothd[3970196]: Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
jún 27 09:17:03 bluetoothd[3970196]: Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
jún 27 09:17:03 bluetoothd[3970196]: Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
```

levaidaniel, 2020-06-29:
A new backtrace from today, seems different than the previous:

```
Reading symbols from /usr/lib/bluetooth/bluetoothd...
[New LWP 2621608]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
Core was generated by `/usr/lib/bluetooth/bluetoothd'.
Program terminated with signal SIGABRT, Aborted.
#0  0x0000007f97be0758 in raise () from /usr/lib/libc.so.6
(gdb) list
654                             "Specify an explicit path to the config file", "FILE"},
655             { "compat", 'C', 0, G_OPTION_ARG_NONE, &option_compat,
656                                     "Provide deprecated command line interfaces" },
657             { "experimental", 'E', 0, G_OPTION_ARG_NONE, &option_experimental,
658                                     "Enable experimental interfaces" },
659             { "nodetach", 'n', G_OPTION_FLAG_REVERSE,
660                                     G_OPTION_ARG_NONE, &option_detach,
661                                     "Run with logging in foreground" },
662             { "version", 'v', 0, G_OPTION_ARG_NONE, &option_version,
663                                     "Show version information and exit" },
(gdb) bt
#0  0x0000007f97be0758 in raise () from /usr/lib/libc.so.6
#1  0x0000007f97bcda68 in abort () from /usr/lib/libc.so.6
#2  0x0000007f97c19e60 in __libc_message () from /usr/lib/libc.so.6
#3  0x0000007f97c21234 in malloc_printerr () from /usr/lib/libc.so.6
#4  0x0000007f97c219bc in unlink_chunk.isra () from /usr/lib/libc.so.6
#5  0x0000007f97c21ab0 in malloc_consolidate () from /usr/lib/libc.so.6
#6  0x0000007f97c23a48 in _int_malloc () from /usr/lib/libc.so.6
#7  0x0000007f97c260c4 in calloc () from /usr/lib/libc.so.6
#8  0x0000007f97c187dc in open_memstream () from /usr/lib/libc.so.6
#9  0x0000007f97c76918 in __vsyslog_internal () from /usr/lib/libc.so.6
#10 0x0000007f97c76f5c in __vsyslog_chk () from /usr/lib/libc.so.6
#11 0x000000557ad42780 in vsyslog (__ap=..., __fmt=0x557adc1098 "Unable to register %s interface for %s", __pri=3) at /usr/include/bits/syslog.h:47
#12 error (format=format@entry=0x557adc1098 "Unable to register %s interface for %s") at src/log.c:61
#13 0x000000557ad38110 in batt_io_ccc_written_cb (att_ecode=<optimized out>, user_data=0x5589d65ce0) at profiles/battery/battery.c:157
#14 0x000000557adab938 in complete_notify_request (data=0x5589d832c0) at src/shared/gatt-client.c:1563
#15 enable_ccc_callback (opcode=1 '\001', pdu=0x0, length=0, user_data=0x5589d832c0) at src/shared/gatt-client.c:1637
#16 0x000000557ada547c in disc_att_send_op (data=0x5589d549d0) at src/shared/att.c:417
#17 0x000000557ad9fc18 in queue_remove_all (queue=<optimized out>, function=function@entry=0x0, user_data=user_data@entry=0x0, destroy=destroy@entry=0x557ada5450 <disc_att_send_op>) at src/shared/queue.c:353
#18 0x000000557ada77b8 in disconnect_cb (io=<optimized out>, user_data=0x5589d85ff0) at src/shared/att.c:635
#19 0x000000557adb46a0 in watch_callback (channel=<optimized out>, cond=<optimized out>, user_data=<optimized out>) at src/shared/io-glib.c:170
#20 0x0000007f97ee666c in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
#21 0x0000007f97ee6af8 in ?? () from /usr/lib/libglib-2.0.so.0
#22 0x0000007f97ee6f00 in g_main_loop_run () from /usr/lib/libglib-2.0.so.0
#23 0x000000557adb4e30 in mainloop_run () at src/shared/mainloop-glib.c:79
#24 0x000000557adb534c in mainloop_run_with_signal (func=<optimized out>, user_data=0x0) at src/shared/mainloop-notify.c:201
#25 0x000000557acfec3c in main (argc=<optimized out>, argv=<optimized out>) at src/main.c:770
(gdb) bt full
#0  0x0000007f97be0758 in raise () from /usr/lib/libc.so.6
No symbol table info available.
#1  0x0000007f97bcda68 in abort () from /usr/lib/libc.so.6
No symbol table info available.
#2  0x0000007f97c19e60 in __libc_message () from /usr/lib/libc.so.6
No symbol table info available.
#3  0x0000007f97c21234 in malloc_printerr () from /usr/lib/libc.so.6
No symbol table info available.
#4  0x0000007f97c219bc in unlink_chunk.isra () from /usr/lib/libc.so.6
No symbol table info available.
#5  0x0000007f97c21ab0 in malloc_consolidate () from /usr/lib/libc.so.6
No symbol table info available.
#6  0x0000007f97c23a48 in _int_malloc () from /usr/lib/libc.so.6
No symbol table info available.
#7  0x0000007f97c260c4 in calloc () from /usr/lib/libc.so.6
No symbol table info available.
#8  0x0000007f97c187dc in open_memstream () from /usr/lib/libc.so.6
No symbol table info available.
#9  0x0000007f97c76918 in __vsyslog_internal () from /usr/lib/libc.so.6
No symbol table info available.
#10 0x0000007f97c76f5c in __vsyslog_chk () from /usr/lib/libc.so.6
[... 23 lines cut ...]
        err = 110
        len = 4
#19 0x000000557adb46a0 in watch_callback (channel=<optimized out>, cond=<optimized out>, user_data=<optimized out>) at src/shared/io-glib.c:170
        watch = <optimized out>
        result = <optimized out>
        destroy = <optimized out>
#20 0x0000007f97ee666c in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
No symbol table info available.
#21 0x0000007f97ee6af8 in ?? () from /usr/lib/libglib-2.0.so.0
No symbol table info available.
#22 0x0000007f97ee6f00 in g_main_loop_run () from /usr/lib/libglib-2.0.so.0
No symbol table info available.
#23 0x000000557adb4e30 in mainloop_run () at src/shared/mainloop-glib.c:79
No locals.
#24 0x000000557adb534c in mainloop_run_with_signal (func=<optimized out>, user_data=0x0) at src/shared/mainloop-notify.c:201
        data = 0x5589d53180
        io = 0x5589d543c0
        ret = <optimized out>
#25 0x000000557acfec3c in main (argc=<optimized out>, argv=<optimized out>) at src/main.c:770
        context = <optimized out>
        err = 0x0
        sdp_flags = <optimized out>
        gdbus_flags = <optimized out>
        __func__ = "main"
```

Vudentz, 2020-07-02:
I send another fix for this, let me know if you are still able to reproduce the crash, also try disabling the battery plugin with -P battery, that seems to be where is crashing now.

levaidaniel, 2020-07-02:
After applying 8689322d482911978ea4166e3c594107a1afee14 this is a new backtrace:
```
Reading symbols from /usr/lib/bluetooth/bluetoothd...
[New LWP 1100008]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
Core was generated by `/usr/lib/bluetooth/bluetoothd'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x000000556dcafb54 in queue_remove_all (queue=0x2d303030382d3030, function=function@entry=0x556dcb7af0 <notify_set_ecode>, user_data=0xe, destroy=destroy@entry=0x556dcb7ac8 <complete_notify_request>) at src/shared/queue.c:327
327             entry = queue->head;
(gdb) 322               unsigned int count = 0;
323
324             if (!queue)
325                     return 0;
326
327             entry = queue->head;
328
329             if (function) {
330                     while (entry) {
331                             void *data;
(gdb) #0  0x000000556dcafb54 in queue_remove_all (queue=0x2d303030382d3030, function=function@entry=0x556dcb7af0 <notify_set_ecode>, user_data=0xe, destroy=destroy@entry=0x556dcb7ac8 <complete_notify_request>) at src/shared/queue.c:327
#1  0x000000556dcbb9bc in enable_ccc_callback (opcode=1 '\001', pdu=0x0, length=0, user_data=0x5595390a90) at src/shared/gatt-client.c:1638
#2  0x000000556dcb547c in disc_att_send_op (data=0x5595381580) at src/shared/att.c:417
#3  0x000000556dcafc18 in queue_remove_all (queue=<optimized out>, function=function@entry=0x0, user_data=user_data@entry=0x0, destroy=destroy@entry=0x556dcb5450 <disc_att_send_op>) at src/shared/queue.c:353
#4  0x000000556dcb77b8 in disconnect_cb (io=<optimized out>, user_data=0x55953937a0) at src/shared/att.c:635
#5  0x000000556dcc3708 in watch_callback (channel=<optimized out>, cond=<optimized out>, user_data=<optimized out>) at src/shared/io-glib.c:170
#6  0x0000007fbbd0966c in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
#7  0x0000007fbbd09af8 in ?? () from /usr/lib/libglib-2.0.so.0
#8  0x0000007fbbd09f00 in g_main_loop_run () from /usr/lib/libglib-2.0.so.0
#9  0x000000556dcc3e98 in mainloop_run () at src/shared/mainloop-glib.c:79
#10 0x000000556dcc43b4 in mainloop_run_with_signal (func=<optimized out>, user_data=0x0) at src/shared/mainloop-notify.c:201
#11 0x000000556dc0ec3c in main (argc=<optimized out>, argv=<optimized out>) at src/main.c:770
(gdb) #0  0x000000556dcafb54 in queue_remove_all (queue=0x2d303030382d3030, function=function@entry=0x556dcb7af0 <notify_set_ecode>, user_data=0xe, destroy=destroy@entry=0x556dcb7ac8 <complete_notify_request>) at src/shared/queue.c:327
        entry = <optimized out>
        count = 0
#1  0x000000556dcbb9bc in enable_ccc_callback (opcode=1 '\001', pdu=0x0, length=0, user_data=0x5595390a90) at src/shared/gatt-client.c:1638
        notify_data = 0x5595390a90
        __PRETTY_FUNCTION__ = "enable_ccc_callback"
#2  0x000000556dcb547c in disc_att_send_op (data=0x5595381580) at src/shared/att.c:417
        op = 0x5595381580
#3  0x000000556dcafc18 in queue_remove_all (queue=<optimized out>, function=function@entry=0x0, user_data=user_data@entry=0x0, destroy=destroy@entry=0x556dcb5450 <disc_att_send_op>) at src/shared/queue.c:353
        tmp = 0x5595380d40
        entry = 0x0
        count = 10
#4  0x000000556dcb77b8 in disconnect_cb (io=<optimized out>, user_data=0x55953937a0) at src/shared/att.c:635
        chan = 0x55953937a0
        att = 0x5595390940
        err = 110
        len = 4
#5  0x000000556dcc3708 in watch_callback (channel=<optimized out>, cond=<optimized out>, user_data=<optimized out>) at src/shared/io-glib.c:170
        watch = <optimized out>
        result = <optimized out>
        destroy = <optimized out>
#6  0x0000007fbbd0966c in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
No symbol table info available.
#7  0x0000007fbbd09af8 in ?? () from /usr/lib/libglib-2.0.so.0
No symbol table info available.
#8  0x0000007fbbd09f00 in g_main_loop_run () from /usr/lib/libglib-2.0.so.0
No symbol table info available.
#9  0x000000556dcc3e98 in mainloop_run () at src/shared/mainloop-glib.c:79
No locals.
#10 0x000000556dcc43b4 in mainloop_run_with_signal (func=<optimized out>, user_data=0x0) at src/shared/mainloop-notify.c:201
        data = 0x5595363240
        io = 0x5595363260
        ret = <optimized out>
#11 0x000000556dc0ec3c in main (argc=<optimized out>, argv=<optimized out>) at src/main.c:770
        context = <optimized out>
        err = 0x0
        sdp_flags = <optimized out>
        gdbus_flags = <optimized out>
[... 274 lines cut ...]
júl 02 21:13:18 bluetoothd[1113608]: bt_uhid_send: Invalid argument (22)
júl 02 21:13:18 bluetoothd[1113608]: bt_uhid_send: Invalid argument (22)
júl 02 21:13:18 bluetoothd[1113608]: bt_uhid_send: Invalid argument (22)
júl 02 21:13:18 bluetoothd[1113608]: bt_uhid_send: Invalid argument (22)
júl 02 21:13:18 bluetoothd[1113608]: bt_uhid_send: Invalid argument (22)
júl 02 21:13:18 bluetoothd[1113608]: bt_uhid_send: Invalid argument (22)
júl 02 21:13:18 bluetoothd[1113608]: bt_uhid_send: Invalid argument (22)
júl 02 21:13:18 bluetoothd[1113608]: bt_uhid_send: Invalid argument (22)
júl 02 21:13:18 bluetoothd[1113608]: bt_uhid_send: Invalid argument (22)
júl 02 21:13:18 bluetoothd[1113608]: bt_uhid_send: Invalid argument (22)
júl 02 21:13:18 bluetoothd[1113608]: bt_uhid_send: Invalid argument (22)
júl 02 21:13:19 bluetoothd[1113608]: bt_uhid_send: Invalid argument (22)
[... cut ...]
júl 02 21:13:26 bluetoothd[1113608]: bt_uhid_send: Invalid argument (22)
júl 02 21:13:26 bluetoothd[1113608]: bt_uhid_send: Invalid argument (22)
júl 02 21:13:26 bluetoothd[1113608]: bt_uhid_send: Invalid argument (22)
júl 02 21:13:43 bluetoothd[1113608]: bt_uhid_send: Invalid argument (22)
júl 02 21:13:43 bluetoothd[1113608]: bt_uhid_send: Invalid argument (22)
júl 02 21:13:43 bluetoothd[1113608]: bt_uhid_send: Invalid argument (22)
júl 02 21:13:43 bluetoothd[1113608]: bt_uhid_send: Invalid argument (22)
```

hadess, 2020-07-08:
The best way to find the problem would be to run bluetooth under valgrind and reproduce the problem. The backtraces you see are different because it crashes because of the memory corruption caused by a double-free rather than the double-free itself.

levaidaniel, 2020-07-08:
Do you have anywhere I can upload valgrind's output?

hadess, 2020-07-08:
You can upload it here as a text attachment, or copy just the last lines that show the actual error, not all the noise before it.

levaidaniel, 2020-07-09:
Does it make sense that now that I'm running it with valgrind (`valgrind --error-limit=no --num-callers=30 --track-origins=yes --keep-stacktraces=alloc-and-free /usr/lib/bluetooth/bluetoothd --nodetach --debug=src/gatt-client.c`) in the foreground it does not want to crash? :confused:

levaidaniel, 2020-07-10:
Hah, speak of the devil!

I attached a lengthy (5662 lines) output from `script(1)` that it recorded in the terminal window.
The command was: `valgrind --error-limit=no --num-callers=30 --track-origins=yes --keep-stacktraces=alloc-and-free /usr/lib/bluetooth/bluetoothd --nodetach --debug=src/gatt-client.c`

And the last 100 lines were:
```
==2669670==    by 0x12BC3B: main (main.c:770)
==2669670==
==2669670== Invalid read of size 4
==2669670==    at 0x1D3EEC: bt_att_unref (att.c:1257)
==2669670==    by 0x1D486B: disconnect_cb (att.c:660)
==2669670==    by 0x1E0707: watch_callback (io-glib.c:170)
==2669670==    by 0x48E963B: g_main_context_dispatch (in /usr/lib/libglib-2.0.so.0.6400.4)
==2669670==    by 0x48E9AC7: ??? (in /usr/lib/libglib-2.0.so.0.6400.4)
==2669670==    by 0x48E9ECF: g_main_loop_run (in /usr/lib/libglib-2.0.so.0.6400.4)
==2669670==    by 0x1E0E97: mainloop_run (mainloop-glib.c:79)
==2669670==    by 0x1E13B3: mainloop_run_with_signal (mainloop-notify.c:201)
==2669670==    by 0x12BC3B: main (main.c:770)
==2669670==  Address 0x7d360a0 is 0 bytes inside a block of size 144 free'd
==2669670==    at 0x484A2E0: free (vg_replace_malloc.c:540)
==2669670==    by 0x1697BB: g_attrib_unref (gattrib.c:195)
==2669670==    by 0x1CC90B: queue_foreach (queue.c:220)
==2669670==    by 0x1D484B: disconnect_cb (att.c:657)
==2669670==    by 0x1E0707: watch_callback (io-glib.c:170)
==2669670==    by 0x48E963B: g_main_context_dispatch (in /usr/lib/libglib-2.0.so.0.6400.4)
==2669670==    by 0x48E9AC7: ??? (in /usr/lib/libglib-2.0.so.0.6400.4)
==2669670==    by 0x48E9ECF: g_main_loop_run (in /usr/lib/libglib-2.0.so.0.6400.4)
==2669670==    by 0x1E0E97: mainloop_run (mainloop-glib.c:79)
==2669670==    by 0x1E13B3: mainloop_run_with_signal (mainloop-notify.c:201)
==2669670==    by 0x12BC3B: main (main.c:770)
==2669670==  Block was alloc'd at
==2669670==    at 0x484914C: malloc (vg_replace_malloc.c:309)
==2669670==    by 0x1CCD07: btd_malloc (util.c:47)
==2669670==    by 0x1D30EB: bt_att_new (att.c:1223)
==2669670==    by 0x169663: g_attrib_new (gattrib.c:114)
==2669670==    by 0x1ACE57: device_attach_att (device.c:5071)
==2669670==    by 0x1796FB: connect_cb (gatt-database.c:651)
==2669670==    by 0x16BD13: server_cb (btio.c:264)
==2669670==    by 0x16BD13: server_cb (btio.c:239)
==2669670==    by 0x48E963B: g_main_context_dispatch (in /usr/lib/libglib-2.0.so.0.6400.4)
==2669670==    by 0x48E9AC7: ??? (in /usr/lib/libglib-2.0.so.0.6400.4)
==2669670==    by 0x48E9ECF: g_main_loop_run (in /usr/lib/libglib-2.0.so.0.6400.4)
==2669670==    by 0x1E0E97: mainloop_run (mainloop-glib.c:79)
==2669670==    by 0x1E13B3: mainloop_run_with_signal (mainloop-notify.c:201)
==2669670==    by 0x12BC3B: main (main.c:770)
==2669670==
==2669670== Invalid write of size 4
==2669670==    at 0x1D3EF4: bt_att_unref (att.c:1257)
==2669670==    by 0x1D486B: disconnect_cb (att.c:660)
==2669670==    by 0x1E0707: watch_callback (io-glib.c:170)
==2669670==    by 0x48E963B: g_main_context_dispatch (in /usr/lib/libglib-2.0.so.0.6400.4)
==2669670==    by 0x48E9AC7: ??? (in /usr/lib/libglib-2.0.so.0.6400.4)
==2669670==    by 0x48E9ECF: g_main_loop_run (in /usr/lib/libglib-2.0.so.0.6400.4)
==2669670==    by 0x1E0E97: mainloop_run (mainloop-glib.c:79)
==2669670==    by 0x1E13B3: mainloop_run_with_signal (mainloop-notify.c:201)
==2669670==    by 0x12BC3B: main (main.c:770)
==2669670==  Address 0x7d360a0 is 0 bytes inside a block of size 144 free'd
==2669670==    at 0x484A2E0: free (vg_replace_malloc.c:540)
==2669670==    by 0x1697BB: g_attrib_unref (gattrib.c:195)
==2669670==    by 0x1CC90B: queue_foreach (queue.c:220)
==2669670==    by 0x1D484B: disconnect_cb (att.c:657)
==2669670==    by 0x1E0707: watch_callback (io-glib.c:170)
==2669670==    by 0x48E963B: g_main_context_dispatch (in /usr/lib/libglib-2.0.so.0.6400.4)
==2669670==    by 0x48E9AC7: ??? (in /usr/lib/libglib-2.0.so.0.6400.4)
==2669670==    by 0x48E9ECF: g_main_loop_run (in /usr/lib/libglib-2.0.so.0.6400.4)
==2669670==    by 0x1E0E97: mainloop_run (mainloop-glib.c:79)
==2669670==    by 0x1E13B3: mainloop_run_with_signal (mainloop-notify.c:201)
==2669670==    by 0x12BC3B: main (main.c:770)
==2669670==  Block was alloc'd at
[... 18 lines cut ...]
bluetoothd[2669670]: src/gatt-client.c:create_services() Exporting objects for GATT services: E6:25:BB:XX:XX:XX
bluetoothd[2669670]: src/gatt-client.c:service_create() Exported GATT service: /org/bluez/hci0/dev_E6_25_BB_XX_XX_XX/service000a
bluetoothd[2669670]: src/gatt-client.c:characteristic_create() Exported GATT characteristic: /org/bluez/hci0/dev_E6_25_BB_XX_XX_XX/service000a/char000b
bluetoothd[2669670]: src/gatt-client.c:descriptor_create() Exported GATT characteristic descriptor: /org/bluez/hci0/dev_E6_25_BB_XX_XX_XX/service000a/char000b/desc000d
bluetoothd[2669670]: src/gatt-client.c:service_create() Exported GATT service: /org/bluez/hci0/dev_E6_25_BB_XX_XX_XX/service000e
bluetoothd[2669670]: src/gatt-client.c:characteristic_create() Exported GATT characteristic: /org/bluez/hci0/dev_E6_25_BB_XX_XX_XX/service000e/char000f
bluetoothd[2669670]: src/gatt-client.c:characteristic_create() Exported GATT characteristic: /org/bluez/hci0/dev_E6_25_BB_XX_XX_XX/service000e/char0011
bluetoothd[2669670]: src/gatt-client.c:characteristic_create() Exported GATT characteristic: /org/bluez/hci0/dev_E6_25_BB_XX_XX_XX/service000e/char0013
[... cut ...]
bluetoothd[2669670]: bt_uhid_send: Invalid argument (22)
bluetoothd[2669670]: bt_uhid_send: Invalid argument (22)
```

These lines were the last ones after it seemingly crashed, but in the attached file the last ~100 lines are after terminating valgrind (and thus bluetoothd), so look for this ^^^ somewhere above where there's the `^Cbluetoothd[2669670]: Terminating` line in the file.

Hope it makes sense :+1: 
[valgrind_record.txt](https://github.com/bluez/bluez/files/4901851/valgrind_record.txt)

Vudentz, 2020-07-15:
Line numbers seems to be off, are you sure you running against the latest tree?

Vudentz, 2020-07-16:
Try with the following patch: https://github.com/Vudentz/BlueZ/commit/e8204d987d862b1b44fbc85470154ab4eb02cce3

hadess, 2020-07-16:
> Try with the following patch: [Vudentz/BlueZ@e8204d9](https://github.com/Vudentz/BlueZ/commit/e8204d987d862b1b44fbc85470154ab4eb02cce3)

`cancel` not `cancal` in the commit message.

I don't see a need for the first hunk.

The `bt_att_cancel()` conditional should probably be in a separate function if you need to have a goto label inside it...

Thanks for fixing my code though :)

levaidaniel, 2020-07-16:
@Vudentz No it was 5.54 with your patch(es) applied.

But this is with latest git pull and your Vudentz/BlueZ@e8204d9

First things first, I almost immediately got this crash with valgrind running:
```
[...]
==3460338== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==3460338==  Access not within mapped region at address 0x70
==3460338==    at 0x4941B38: g_io_channel_unix_get_fd (in /usr/lib/libglib-2.0.so.0.6400.4)
==3460338==    by 0x161207: bt_io_get_type (btio.c:106)
==3460338==    by 0x162D73: bt_io_get (btio.c:1615)
==3460338==    by 0x155C8B: report_map_read_cb (hog-lib.c:991)
==3460338==    by 0x15D3DF: read_blob_helper (gatt.c:804)
==3460338==    by 0x15F4A3: attrib_callback_result (gattrib.c:273)
==3460338==    by 0x15F4A3: attrib_callback_result (gattrib.c:250)
==3460338==    by 0x1BE333: disc_att_send_op (att.c:420)
==3460338==    by 0x1BA357: queue_remove_all (queue.c:354)
==3460338==    by 0x1C02FF: disconnect_cb (att.c:658)
==3460338==    by 0x1CABD7: watch_callback (io-glib.c:170)
==3460338==    by 0x48E963B: g_main_context_dispatch (in /usr/lib/libglib-2.0.so.0.6400.4)
==3460338==    by 0x48E9AC7: ??? (in /usr/lib/libglib-2.0.so.0.6400.4)
==3460338==    by 0x48E9ECF: g_main_loop_run (in /usr/lib/libglib-2.0.so.0.6400.4)
==3460338==    by 0x1CB24B: mainloop_run (mainloop-glib.c:79)
==3460338==    by 0x1CB683: mainloop_run_with_signal (mainloop-notify.c:201)
==3460338==    by 0x129567: main (main.c:950)
==3460338==  If you believe this happened as a result of a stack
==3460338==  overflow in your program's main thread (unlikely but
==3460338==  possible), you can try to increase the size of the
==3460338==  main thread stack using the --main-stacksize= flag.
==3460338==  The main thread stack size used in this run was 8388608.
==3460338==
==3460338== HEAP SUMMARY:
==3460338==     in use at exit: 654,939 bytes in 2,386 blocks
==3460338==   total heap usage: 30,622 allocs, 28,236 frees, 4,029,103 bytes allocated
==3460338==
==3460338== LEAK SUMMARY:
==3460338==    definitely lost: 208 bytes in 4 blocks
==3460338==    indirectly lost: 0 bytes in 0 blocks
==3460338==      possibly lost: 0 bytes in 0 blocks
==3460338==    still reachable: 654,731 bytes in 2,382 blocks
==3460338==         suppressed: 0 bytes in 0 blocks
==3460338== Rerun with --leak-check=full to see details of leaked memory
==3460338==
==3460338== For lists of detected and suppressed errors, rerun with: -s
==3460338== ERROR SUMMARY: 4 errors from 4 contexts (suppressed: 0 from 0)
Segmentation fault (core dumped)
```
Even valgrind crashed at the end... :smile: 

Full output is attached.
[valgrind_record-2020-07-16.txt](https://github.com/bluez/bluez/files/4931489/valgrind_record-2020-07-16.txt)

Secondly, do you think it would be possible to figure out why is it so painful to start using this keyboard (Logitech MX Keys) with bluez? I mean, with 5.54 it was after about 1 minute after connecting the device until it _settled_, but with this latest git pull, it just goes on and on, for minutes on end with disconnecting and reconnecting, and it's kind of impossible to use it. E.g.:
```
bluetoothd[3519307]: Bluetooth daemon 5.54
bluetoothd[3519307]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[3519307]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[3519307]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[3519307]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[3519307]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[3519307]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[3519307]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[3519307]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[3519307]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[3519307]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[3519307]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[3519307]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[3519307]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[3519307]: src/main.c:parse_controller_config() Key file does not have group “Controller”
[... 332 lines cut ...]
bluetoothd[3519307]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
bluetoothd[3519307]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
bluetoothd[3519307]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
bluetoothd[3519307]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
bluetoothd[3519307]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
bluetoothd[3519307]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
bluetoothd[3519307]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
bluetoothd[3519307]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
[... cut ...]
bluetoothd[3519307]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
bluetoothd[3519307]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
bluetoothd[3519307]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
[...]
```

Don't get me wrong, I'm glad it uncovers a bunch of bugs during device disconnect :stuck_out_tongue_winking_eye: , but can we somehow figure out what it is that bluez doesn't like about this device's connection process?

Cheers,
Dani

Vudentz, 2020-07-16:
@levaidaniel Well it seems pretty clear that the link is very unstable and it constantly gets disconnected and connected, have you tried with a different controller? That said we shouldn't be crashing, so Im trying to fix that first before we move on to why the connection is so unstable.

Vudentz, 2020-07-20:
@levaidaniel Ive applied a few more fixes that I found while investigating this issue, let me know if still have the same issue.

levaidaniel, 2020-07-22:
@Vudentz Thanks for the heads-up. I've been running with commit 6ea1b038eedeca31c3796ac921e25509d48d70ec from the 17th for a couple of days now, and the crashes seem to have disappeared.
Plus, I don't know if this is just a placebo effect, but it seems that the initial number of disconnections when turning on the keyboard have been reduced to only a couple (1-2 in the first minute). Anyway, I need to test this latter a bit further, but I must thank you for these fixes, these actually made _bluez_ a lot more usable on my end!

levaidaniel, 2020-07-23:
Maybe I was a bit over-optimistic re disconnections :smile: 
But in any case, what do you say we close this, because there were no crashes so far, and I open a new issue for the iffy (dis)connecting when powering on the keyboard?

closed 2020-08-08 by Vudentz