bluez/bluez #22: obexd: Crash if outgouing transfer is cancelled by peer

d-grigorev, 2020-06-30:
Steps to reproduce:
1. Send a file big enough using obex bluetooth
2. Cancel transfer on receiving side

I discovered it on BlueZ 5.50 but as I see from the commit history, the bug is still there.

Backtrace :
__GI___pthread_mutex_lock (mutex=0x65732f74) at pthread_mutex_lock.c:67
0xecc6eeda in dbus_connection_get_object_path_data () from /usr/lib/libdbus-1.so.3
0x000457d4 in g_dbus_emit_property_changed_full () at gdbus/object.c:1794
0x00045868 in g_dbus_emit_property_changed () at gdbus/object.c:1832
0x000367f0 in transfer_set_status () at obexd/client/transfer.c:211
0x0003681e in transfer_set_status () at obexd/client/transfer.c:206
xfer_complete () at obexd/client/transfer.c:672
0x00022df6 in transfer_complete () at gobex/gobex-transfer.c:103
0x00022f44 in transfer_abort_response () at gobex/gobex-transfer.c:124
0x00020a0e in handle_response () at gobex/gobex.c:1128

d-grigorev, 2020-06-30:
Recently I posted a [patch](https://lore.kernel.org/linux-bluetooth/20200616183033.5243-1-d.grigorev@omprussia.ru/) to linux-bluetooth mailing  list but apparently it got lost due to high rate. Where is the correct place for patches?

Vudentz, 2020-07-01:
I will apply it shortly, I thought I had applied it already.

d-grigorev, 2020-07-01:
Great! Thanks.

Vudentz, 2020-07-02:
Applied

closed 2020-07-02 by Vudentz