bluez/bluez #23: Can't see registered player when introspecting

lapinozz, 2020-07-01:
So I've registered a player using the python example
It seems to register fine as it doesn't error anymore and the log in "/var/log/syslog" says it's registered

But when I introspect with `gdbus introspect  --system --dest org.bluez --object-path /org --recurse` I don't see it
Am I missing something?

The reason I want to register a player is for my headphone which connects but doesn't have a player so the volume key have no effects.
I checked the code and it seems like it's checking if it has a player if it does it updates the Volume property.

Thanks!

Vudentz, 2020-07-01:
Registered player would be on its connection id not under org.bluez if that is what you expect. The headset is still expected to send button presses for volume up and down, or you saying that has no effect in your system?

lapinozz, 2020-07-01:
I'm not sure what you mean by "its connection id"?

Regardless, I do see that the volume get sent and process
[image]

But I would like to receive that update programmatically. 
It doesn't seem to change the system sound level on my ubuntu machine

Vudentz, 2020-07-10:
That is for the headset to notify its volume changes, changes to that would be notified to the MediaTransport directly:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/media-api.txt#n651

Since it can only be changed by the application that acquired the transport you can change it directly, PulseAudio exposes it own volume control so you can access the volume via the audio card created for your headset, perhaps you are saying that doesn't working for some reason.

Vudentz, 2020-07-16:
The following patch should have fixed the volume:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=df7d3fa5002373f7f955219defc7d0bc794a5a6c

closed 2020-07-16 by Vudentz