bluez/bluez #30: Not all characteristics are listed for the service

BerserkerDotNet, 2020-07-28:
Hi :)

### Environment
**Device:** Raspberry Pi 4 and 3
**OS:** Raspbian GNU/Linux 10 (buster) tried ubuntu as well
**Hardware:** [Makeblock Ultimate 2.0](https://www.makeblock.com/steam-kits/mbot-ultimate)
**Bluez version:** 5.54

### Issue
1. Connect to the device  [Makeblock Ultimate 2.0](https://www.makeblock.com/steam-kits/mbot-ultimate)
1. List all services
1. Pick the service in question 
1. List characteristics

**Expected:**
2 characteristics should be present. I found it by looking at Makeblock's source code for Android and iOS app.
Additionally, when using [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en_US) it correctly lists 2 characteristics under the service in question.
[image]

**Actual:**
1 characteristic is present
[image]

### Initial debugging

When using `btgatt-client` tool I get this error:
```
pi@raspberrypi:~/bluez-5.54/tools $ ./btgatt-client -v -d 00:0D:19:12:75:22
btgatt-client: Opening L2CAP LE connection on ATT channel:
         src: 00:00:00:00:00:00
        dest: 00:0D:19:12:75:22
Connecting to device... Done
...
[GATT client]# att: (chan 0xc82570) ATT response received: 0x09
[GATT client]# att: (chan 0xc82570) ATT op 0x08
[GATT client]# att: < 08 15 00 ff ff 03 28                             ......(         
[GATT client]# att: (chan 0xc82570) ATT received: 5
[GATT client]# att: > 01 08 15 00 0a                                   .....           
[GATT client]# att: (chan 0xc82570) ATT response received: 0x01
[GATT client]# gatt: Characteristics found: 7
[GATT client]# gatt: start: 0x0002, end: 0x0003, value: 0x0003, props: 0x0a, uuid: 00002a00-0000-1
[GATT client]# gatt: start: 0x0004, end: 0x0005, value: 0x0005, props: 0x02, uuid: 00002a01-0000-1
[GATT client]# gatt: start: 0x0006, end: 0x0008, value: 0x0007, props: 0x02, uuid: 00002a04-0000-1
[GATT client]# gatt: start: 0x0009, end: 0x000c, value: 0x000a, props: 0x20, uuid: 00002a05-0000-1
[GATT client]# gatt: start: 0x000d, end: 0x000f, value: 0x000e, props: 0x10, uuid: 0000ffe2-0000-1
[GATT client]# gatt: start: 0x0010, end: 0x0013, value: 0x0011, props: 0x04, uuid: 0000ffe3-0000-1
[GATT client]# gatt: start: 0x0014, end: 0xffff, value: 0x0015, props: 0x06, uuid: 0000ffe5-0000-1
[GATT client]# Service Added - UUID: 00001800-0000-1000-8000-00805f9b34fb start: 0x0001 end: 0x0007
[GATT client]# Service Added - UUID: 00001801-0000-1000-8000-00805f9b34fb start: 0x0008 end: 0x000b
[GATT client]# Service Added - UUID: 0000ffe1-0000-1000-8000-00805f9b34fb start: 0x000c end: 0x0010
[GATT client]# gatt: Failed to insert characteristic at 0x0011
[GATT client]# gatt: service disappeared: start 0x0013 end 0xffff
[GATT client]# gatt: Failed to initialize gatt-client
[GATT client]# GATT discovery procedures failed - error code: 0x0a
```
According to the BT Core spec `0x0a` is `Attribute Not Found - No attribute found within the given attribute handle range`

Using the `gatttool` I can see that there are 4 services:
```
[00:0D:19:12:75:22][LE]> primary
attr handle: 0x0001, end grp handle: 0x0007 uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle: 0x0008, end grp handle: 0x000b uuid: 00001801-0000-1000-8000-00805f9b34fb
attr handle: 0x000c, end grp handle: 0x0010 uuid: 0000ffe1-0000-1000-8000-00805f9b34fb
attr handle: 0x0013, end grp handle: 0xffff uuid: 0000ffe4-0000-1000-8000-00805f9b34fb
```
Service that I need is `0000ffe1-0000-1000-8000-00805f9b34fb` and it's group handle ends at `0x0010`.
Listing characteristics lists the following:
```
handle: 0x0002, char properties: 0x0a, char value handle: 0x0003, uuid: 00002a00-0000-1000-8000-00805f9b34fb
handle: 0x0004, char properties: 0x02, char value handle: 0x0005, uuid: 00002a01-0000-1000-8000-00805f9b34fb
handle: 0x0006, char properties: 0x02, char value handle: 0x0007, uuid: 00002a04-0000-1000-8000-00805f9b34fb
handle: 0x0009, char properties: 0x20, char value handle: 0x000a, uuid: 00002a05-0000-1000-8000-00805f9b34fb
handle: 0x000d, char properties: 0x10, char value handle: 0x000e, uuid: 0000ffe2-0000-1000-8000-00805f9b34fb
handle: 0x0010, char properties: 0x04, char value handle: 0x0011, uuid: 0000ffe3-0000-1000-8000-00805f9b34fb
handle: 0x0014, char properties: 0x06, char value handle: 0x0015, uuid: 0000ffe5-0000-1000-8000-00805f9b34fb
```
Notice that missing characteristic is `0000ffe3-0000-1000-8000-00805f9b34fb` and it is right on the edge of the service's group handle address. Maybe that is the reason for `0x0a` aka `Attribute Not Found`.

One other thing to mention, if I use `gatttool` to write to the `0x11` (`0000ffe3-0000-1000-8000-00805f9b34fb`) characteristics, it works and returns a result:
```
[00:0D:19:12:75:22][LE]> char-write-cmd 0x11 ff5504010101070a
Notification handle = 0x000e value: ff 55 01 02 ee 69 a4 41 0d 0a 
```

Let me know what other diagnostics I can run to generate more information.

Vudentz, 2020-08-08:
I would check with bluetoothd/bluetoothctl and the latest git to see if there is any change to this, note that it does find the characteristic:

`[GATT client]# gatt: start: 0x0010, end: 0x0013, value: 0x0011, props: 0x04, uuid: 0000ffe3-0000-1`

But it could not insert it on the database:

`[GATT client]# gatt: Failed to insert characteristic at 0x0011`

Which is why it doesn't display it.

cspensky, 2021-01-06:
After much digging, it appears to be BlueZ trying to read the battery status of my iPhone when I connect.  

I attached [PacketLogger to my iPhone](https://www.bluetooth.com/blog/a-new-way-to-debug-iosbluetooth-applications/) and it looks like Apple requires authentication to read that characteristic.
[image]

Odd behavior indeed and the fact that it doing it by default, without a flag is pretty annoying.  Is this a Bleak thing?  Any insights into how to turn this off would be great.  Otherwise, I plan to continue to dig, and potentially submit a PR to the BlueZ folks.