bluez/bluez #32: Continuous reconnect when first turning on/waking up Logitech MX Keys

levaidaniel, 2020-08-04:
I have a Logitech MX Keys keyboard and when I turn it off, and turn it on after a while (or just leave it there for e.g. the night and wake it up), it disconnects/reconnects for minutes with bluez after it _settles_ and becomes usable. Also, I have to continuously push buttons for it to reconnect, otherwise it doesn't do anything after disconnecting for some reason.
So a typical day begins with me pushing a button (doesn't seem to matter which) for minutes, and noticing when what I type doesn't get lost and the input is continuous so that I know that it has connected for sure.
Here's some debug log from bluez (I attached the full log file to this issue), running with `/usr/lib/bluetooth/bluetoothd -P battery --debug=profiles/input/hog-lib.c` (I had to disable the battery module, because it was erroneous; see #19 ):
```
aug 04 10:43:29  bluetoothd[9651]: profiles/input/hog-lib.c:bt_hog_attach() HoG discovering characteristics
aug 04 10:43:29  bluetoothd[9651]: profiles/input/hog-lib.c:info_read_cb() bcdHID: 0x0111 bCountryCode: 0x00 Flags: 0x03
aug 04 10:43:29  bluetoothd[9651]: profiles/input/hog-lib.c:report_reference_cb() Report 0x002f: id 0x01 type input
aug 04 10:43:30  bluetoothd[9651]: profiles/input/hog-lib.c:report_reference_cb() Report 0x0033: id 0x01 type output
aug 04 10:43:30  bluetoothd[9651]: profiles/input/hog-lib.c:report_reference_cb() Report 0x0036: id 0x03 type input
aug 04 10:43:30  bluetoothd[9651]: profiles/input/hog-lib.c:report_reference_cb() Report 0x003a: id 0x02 type input
aug 04 10:43:30  bluetoothd[9651]: profiles/input/hog-lib.c:report_reference_cb() Report 0x003e: id 0x11 type input
aug 04 10:43:30  bluetoothd[9651]: profiles/input/hog-lib.c:report_reference_cb() Report 0x0042: id 0x11 type output
aug 04 10:43:30  bluetoothd[9651]: profiles/input/hog-lib.c:proto_mode_read_cb() HoG is operating in Report Protocol Mode
aug 04 10:43:32  bluetoothd[9651]: profiles/input/hog-lib.c:report_ccc_written_cb() Report characteristic descriptor written: notifications enabled
aug 04 10:43:32  bluetoothd[9651]: profiles/input/hog-lib.c:report_ccc_written_cb() Report characteristic descriptor written: notifications enabled
aug 04 10:43:32  bluetoothd[9651]: profiles/input/hog-lib.c:report_ccc_written_cb() Report characteristic descriptor written: notifications enabled
aug 04 10:43:32  bluetoothd[9651]: profiles/input/hog-lib.c:report_ccc_written_cb() Report characteristic descriptor written: notifications enabled
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb() HoG inspecting report map
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb() Report MAP:
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          05 01
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          09 06
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          a1 01
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          85 01
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          05 07
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          19 e0
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          29 e7
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          15 00
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          25 01
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          75 01
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          95 08
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          81 02
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          95 05
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          05 08
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          19 01
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          29 05
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          91 02
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          95 01
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          75 03
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          91 01
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          95 06
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          75 08
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          15 00
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          26 a4 00
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          05 07
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          19 00
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          2a a4 00
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          81 00
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          c0
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          05 01
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          09 02
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          a1 01
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb() Report Map parsing failed at 65
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          85
aug 04 10:43:34  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb() Failed to connection details: getpeername: Transport endpoint is not connected (107)
aug 04 10:43:35  bluetoothd[9651]: profiles/input/hog-lib.c:bt_hog_attach() HoG discovering characteristics
aug 04 10:43:36  bluetoothd[9651]: profiles/input/hog-lib.c:info_read_cb() bcdHID: 0x0111 bCountryCode: 0x00 Flags: 0x03
aug 04 10:43:36  bluetoothd[9651]: profiles/input/hog-lib.c:report_reference_cb() Report 0x002d: id 0x01 type input
aug 04 10:43:36  bluetoothd[9651]: profiles/input/hog-lib.c:report_reference_cb() Report 0x0031: id 0x02 type input
aug 04 10:43:36  bluetoothd[9651]: profiles/input/hog-lib.c:report_reference_cb() Report 0x0035: id 0x11 type input
aug 04 10:43:36  bluetoothd[9651]: profiles/input/hog-lib.c:report_reference_cb() Report 0x0039: id 0x11 type output
aug 04 10:43:36  bluetoothd[9651]: profiles/input/hog-lib.c:proto_mode_read_cb() HoG is operating in Report Protocol Mode
aug 04 10:43:36  bluetoothd[9651]: profiles/input/hog-lib.c:report_ccc_written_cb() Report characteristic descriptor written: notifications enabled
aug 04 10:43:36  bluetoothd[9651]: profiles/input/hog-lib.c:report_ccc_written_cb() Report characteristic descriptor written: notifications enabled
aug 04 10:43:36  bluetoothd[9651]: profiles/input/hog-lib.c:report_ccc_written_cb() Report characteristic descriptor written: notifications enabled
aug 04 10:43:37  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb() HoG inspecting report map
aug 04 10:43:37  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb() Report MAP:
aug 04 10:43:37  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          05 01
aug 04 10:43:37  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          09 06
aug 04 10:43:37  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          a1 01
aug 04 10:43:37  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          85 01
[... 128 lines cut ...]
aug 04 10:43:42  bluetoothd[9651]: profiles/input/hog-lib.c:report_ccc_written_cb() Report characteristic descriptor written: notifications enabled
aug 04 10:43:42  bluetoothd[9651]: profiles/input/hog-lib.c:report_ccc_written_cb() Report characteristic descriptor written: notifications enabled
aug 04 10:43:42  bluetoothd[9651]: profiles/input/hog-lib.c:report_ccc_written_cb() Report characteristic descriptor written: notifications enabled
[... cut ...]
aug 04 10:43:42  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          15 00
aug 04 10:43:42  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          25 01
aug 04 10:43:42  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          75 01
aug 04 10:43:42  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          95 08
aug 04 10:43:42  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          81 02
aug 04 10:43:42  bluetoothd[9651]: profiles/input/hog-lib.c:report_map_read_cb()          95 05
[...]
```

Vudentz, 2020-08-08:
Is that a regression since the last time you meantioned it was working much better.

levaidaniel, 2020-08-09:
That was just wishful thinking, unfortunately. I'm not sure what this depends on, but sometimes the keyboard wakes up after a couple of seconds of (re)connecting, but mostly it's just a pain.
I'd be happy to provide more details if you tell me where to look, what debug option(s) to turn on, etc...

levaidaniel, 2020-08-14:
This is another thing that happens, where it doesn't connect at all and I need to actually turn the keyboard off and on again, or switch over and back to and from another pairing setup (it has three "slots" you can use to connect to multiple devices and you switch between them with the dedicated buttons on the keyboard):
```
aug 14 10:19:25 bluetoothd[450]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
aug 14 10:19:25 bluetoothd[450]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
aug 14 10:19:25 bluetoothd[450]: profiles/input/hog-lib.c:report_map_read_cb() HoG inspecting report map
aug 14 10:19:25 bluetoothd[450]: profiles/input/hog-lib.c:report_map_read_cb() Report Map read failed: Request attribute has encountered an unlikely error
aug 14 10:19:25 bluetoothd[450]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
aug 14 10:19:25 bluetoothd[450]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
aug 14 10:19:25 bluetoothd[450]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
aug 14 10:19:25 bluetoothd[450]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
aug 14 10:19:25 bluetoothd[450]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
aug 14 10:19:25 bluetoothd[450]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
aug 14 10:19:25 bluetoothd[450]: profiles/input/hog-lib.c:proto_mode_read_cb() Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error
aug 14 10:19:27 bluetoothd[450]: profiles/input/hog-lib.c:bt_hog_attach() HoG discovering characteristics
aug 14 10:19:29 bluetoothd[450]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
aug 14 10:19:29 bluetoothd[450]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
aug 14 10:19:29 bluetoothd[450]: profiles/input/hog-lib.c:report_map_read_cb() HoG inspecting report map
aug 14 10:19:29 bluetoothd[450]: profiles/input/hog-lib.c:report_map_read_cb() Report Map read failed: Request attribute has encountered an unlikely error
aug 14 10:19:29 bluetoothd[450]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
aug 14 10:19:29 bluetoothd[450]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
aug 14 10:19:29 bluetoothd[450]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
aug 14 10:19:29 bluetoothd[450]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
aug 14 10:19:29 bluetoothd[450]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
aug 14 10:19:29 bluetoothd[450]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
aug 14 10:19:29 bluetoothd[450]: profiles/input/hog-lib.c:proto_mode_read_cb() Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error
aug 14 10:19:31 bluetoothd[450]: profiles/input/hog-lib.c:bt_hog_attach() HoG discovering characteristics
aug 14 10:19:33 bluetoothd[450]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
aug 14 10:19:33 bluetoothd[450]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
aug 14 10:19:33 bluetoothd[450]: profiles/input/hog-lib.c:report_map_read_cb() HoG inspecting report map
aug 14 10:19:33 bluetoothd[450]: profiles/input/hog-lib.c:report_map_read_cb() Report Map read failed: Request attribute has encountered an unlikely error
aug 14 10:19:33 bluetoothd[450]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
aug 14 10:19:33 bluetoothd[450]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
aug 14 10:19:33 bluetoothd[450]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
aug 14 10:19:33 bluetoothd[450]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
[... cut ...]
aug 14 10:20:18 bluetoothd[450]: profiles/input/hog-lib.c:report_ccc_written_cb() Report characteristic descriptor written: notifications enabled
aug 14 10:20:18 bluetoothd[450]: profiles/input/hog-lib.c:report_ccc_written_cb() Report characteristic descriptor written: notifications enabled
aug 14 10:20:18 bluetoothd[450]: profiles/input/hog-lib.c:report_ccc_written_cb() Report characteristic descriptor written: notifications enabled
aug 14 10:20:18 bluetoothd[450]: profiles/input/hog-lib.c:report_ccc_written_cb() Report characteristic descriptor written: notifications enabled
```

levaidaniel, 2020-08-17:
I've attached some more logs with a `btmon` output, maybe it helps a bit. The timestamps correlate, and the **MX Keys'** MAC address is the E6:25:BB one.
[bluetoothd_debug-profiles_input_hog-lib.c-2020-08-17.log](https://github.com/bluez/bluez/files/5086534/bluetoothd_debug-profiles_input_hog-lib.c-2020-08-17.log)
[btmon_capture-2020-08-17.log](https://github.com/bluez/bluez/files/5086585/btmon_capture-2020-08-17.log)

Vudentz, 2021-01-08:
@levaidaniel Perhaps it is worth check if this problem is fixed by:

https://patchwork.kernel.org/project/bluetooth/list/?series=411499

closed 2025-09-01 by github-actions[bot]