bluez/bluez #33: GATT server Handle values causes iOS apps to crash

Climax777, 2020-08-25:
Hi

I've had trouble with iOS when I fix handle values to my services and characteristics. I had to fix the Handle values in order to get Android's cache to play nicely with my GATT server.

The moment I set the Handle value to a fixed value, iOS won't connect. In fact, it actually causes any of the bluetooth apps I tried to crash. It works perfectly with Android, and it correctly uses the Handles I set without requiring a cache clear on Android's side.

I've used the `example_gatt_server.py` file and added fixed handles in `get_properties` for services and characteristics.

So in essence: am I somehow setting the Handle values improperly, or is there a compatibility issue with Handles and iOS?

Vudentz, 2020-09-01:
Not sure how we can help here, except if the handles are somehow overlapping, do you have a dump of the resulting database?

Vudentz, 2020-09-16:
No response over 15 days.

closed 2020-09-16 by Vudentz