bluez/bluez #34: slow point-to-multipoint file transfer speed over l2cap

opopops, 2020-08-26:
I'm facing to a problem when I set up a point-to-multipoint l2cap connection between a master and multiples slaves.

First, on a fresh bluez install, I pair the master device with only one slave, and I transfer a file over L2CAP, the file speed transfer is around 200KB/s. When I connect a second device to the master and try to transfer a file over l2cap connection the file transfer speed is now divided by 2 (~100KB/s), even if there is no activities on the first connection, If I connect a third device the file transfer speed is divided by 3... 

how can I disable this behavior and share dynamically the bandwidth? 

Bluez 5.50 is installed on all my devices.

Vudentz, 2020-09-01:
Im afraid this is down to link layer, it might be picking a packet type that drops the bandwidth in order to attend other connections as well, perhaps check with other controllers to see if that helps.

Vudentz, 2020-09-16:
No reponse over 15 days.

closed 2020-09-16 by Vudentz