bluez/bluez #36: 3rd party sixaxis fails to connect

rderooy, 2020-09-04:
Fedora 32, bluez 5.54, kernel 5.8.4.

I have both an official Sony PS3 DS3 sixaxis controller, and two identical 3rd party controllers.

I can connect the official Sony controller just fine, although it does take a while from pressing the PS button, until it manages to successfully pair. Watching bluetoothctl output shows that it simply takes a long time before the MAC address gets listed.

But the 3rd party controllers are problematic. They work just fine with the PS3 or when plugged in via USB. But over Bluetooth it is impossible to get them to connect. And even if you get them to connect after many tries, they almost immediately again disconnect.

The 3rd party controllers are branded as "Under Control" and "Proxima Plus".

```
 sudo lsusb -v -d 054c:0268
Bus 003 Device 044: ID 054c:0268 Sony Corp. Batoh Device / PlayStation 3 Controller
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x054c Sony Corp.
  idProduct          0x0268 Batoh Device / PlayStation 3 Controller
  bcdDevice            1.00
  iManufacturer           1 Sony
  iProduct                2 PLAYSTATION(R)3 Controller
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     148
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```
kernel messages on USB plug:
```
Sep 04 13:34:45 e93z kernel: usb 3-2.1: new full-speed USB device number 47 using xhci_hcd
Sep 04 13:34:45 e93z kernel: usb 3-2.1: New USB device found, idVendor=054c, idProduct=0268, bcdDevice= 1.00
Sep 04 13:34:45 e93z kernel: usb 3-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Sep 04 13:34:45 e93z kernel: usb 3-2.1: Product: PLAYSTATION(R)3 Controller
Sep 04 13:34:45 e93z kernel: usb 3-2.1: Manufacturer: Sony
Sep 04 13:34:45 e93z kernel: input: Sony PLAYSTATION(R)3 Controller Motion Sensors as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.1/3-2.1:1.0/0003:054C:0268.0030/input/input118
Sep 04 13:34:45 e93z kernel: input: Sony PLAYSTATION(R)3 Controller as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.1/3-2.1:1.0/0003:054C:0268.0030/input/input117
Sep 04 13:34:45 e93z kernel: sony 0003:054C:0268.0030: input,hiddev99,hidraw6: USB HID v81.11 Joystick [Sony PLAYSTATION(R)3 Controller] on usb-0000:00:14.0-2.1/input0
Sep 04 13:34:45 e93z mtp-probe[164880]: checking bus 3, device 47: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.1"
Sep 04 13:34:45 e93z mtp-probe[164880]: bus: 3, device: 47 was not an MTP device
Sep 04 13:34:45 e93z systemd[1]: Starting IIO Sensor Proxy service...
Sep 04 13:34:45 e93z bluetoothd[157879]: sixaxis: compatible device connected: Sony PLAYSTATION(R)3 Controller (054C:0268 /sys/devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.1/3-2.1:1.0/0003:054C:0268.0030/hidraw/hidraw6)
Sep 04 13:34:45 e93z bluetoothd[157879]: sixaxis: setting up new device
Sep 04 13:34:45 e93z bluetoothd[157879]: Authentication attempt without agent
Sep 04 13:34:45 e93z systemd-udevd[164884]: js0: Process '/usr/lib/snapd/snap-device-helper add snap_retroarch_retroarch /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.1/3-2.1:1.0/0003:054C:0268.0030/input/input117/js0 13:0' failed with exit code 1.
Sep 04 13:34:45 e93z /usr/libexec/gdm-x-session[1737]: (II) config/udev: Adding input device Sony PLAYSTATION(R)3 Controller (/dev/input/js0)
Sep 04 13:34:45 e93z /usr/libexec/gdm-x-session[1737]: (II) No input driver specified, ignoring this device.
Sep 04 13:34:45 e93z /usr/libexec/gdm-x-session[1737]: (II) This device may have been added with another device file.
Sep 04 13:34:45 e93z systemd[1]: Started IIO Sensor Proxy service.
Sep 04 13:34:45 e93z audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=iio-sensor-proxy comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Sep 04 13:34:45 e93z /usr/libexec/gdm-x-session[1737]: (II) config/udev: Adding input device Sony PLAYSTATION(R)3 Controller Motion Sensors (/dev/input/event16)
Sep 04 13:34:45 e93z /usr/libexec/gdm-x-session[1737]: (II) No input driver specified, ignoring this device.
Sep 04 13:34:45 e93z /usr/libexec/gdm-x-session[1737]: (II) This device may have been added with another device file.
Sep 04 13:34:45 e93z systemd-udevd[164886]: event17: Process '/usr/lib/snapd/snap-device-helper add snap_retroarch_retroarch /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.1/3-2.1:1.0/0003:054C:0268.0030/input/input117/event17 13:81' failed with exit code 1.
Sep 04 13:34:45 e93z /usr/libexec/gdm-x-session[1737]: (II) config/udev: Adding input device Sony PLAYSTATION(R)3 Controller (/dev/input/event17)
Sep 04 13:34:45 e93z /usr/libexec/gdm-x-session[1737]: (II) No input driver specified, ignoring this device.
Sep 04 13:34:45 e93z /usr/libexec/gdm-x-session[1737]: (II) This device may have been added with another device file.
Sep 04 13:34:45 e93z mtp-probe[164902]: checking bus 3, device 47: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.1"
Sep 04 13:34:45 e93z mtp-probe[164902]: bus: 3, device: 47 was not an MTP device
Sep 04 13:34:45 e93z upowerd[834]: treating change event as add on /sys/devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.1/3-2.1:1.0/0003:054C:0268.0030/power_supply/sony_controller_battery_33:22:23:00:05:42
Sep 04 13:34:45 e93z systemd[1]: iio-sensor-proxy.service: Succeeded.
Sep 04 13:34:45 e93z audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=iio-sensor-proxy comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Sep 04 13:34:45 e93z gsd-power[2105]: Call to iio-proxy failed: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name is not activatable
```

bluetoothctl output for 3rd paty controller:
```
[Sony PLAYSTATION(R)3 Controller]# info 33:22:23:00:05:42 
Device 33:22:23:00:05:42 (public)
	Name: Sony PLAYSTATION(R)3 Controller
	Alias: Sony PLAYSTATION(R)3 Controller
	Class: 0x00001f00
	Paired: no
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v054Cp0268d0000
```

When I plug it in with the Gnome Bluetooth settings open, I get a pop-up telling me that the PS3 controller wants to connect to this device, and if I want to allow it. In fact, it asks this EVERY single time it gets plugged in, even if you allow it (perhaps it should remember it?).

rderooy, 2020-09-04:
Comparing lsusb and kernel messages between the authentic Sony controller and the 3rd party controller, they appear identical.

But looking at bluetoothctl, I notice one difference. The Class is different.

Real Sony PS3 controller:
```
[Sony PLAYSTATION(R)3 Controller]# info E0:75:0A:8B:0E:04 
Device E0:75:0A:8B:0E:04 (public)
	Name: Sony PLAYSTATION(R)3 Controller
	Alias: Sony PLAYSTATION(R)3 Controller
	Class: 0x00000508
	Icon: input-gaming
	Paired: no
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v054Cp0268d0000
```

aguaviva, 2021-03-26:
Hi Robert,

Good to hear you can pair the Sony PS3 controller. I am so looking forward to doing the same but it wort work for me. Could you please explain what did you do? I can't wait to use play some MSX games with it :)

Cooky-12, 2021-08-30:
Hi, I made a patch to make fake PS3 controller work, though it may not a right fix, but my fake PS3 controller work in bluetooth mode now.
Have a try.
https://aur.archlinux.org/cgit/aur.git/tree/fake-ps3.patch?h=bluez-plugins-fix

mirh, 2024-09-28:
Does it work now? Otherwise you could try with each of the Windows drivers that we have code available.
https://forums.pcsx2.net/Thread-XInput-Wrapper-for-DS3-and-Play-com-USB-Dual-DS2-Controller
https://github.com/nefarius/ScpToolkit
https://github.com/nefarius/BthPS3
Fakes are so.. random.

rderooy, 2024-09-29:
I was not quite sure any longer how I set it up 4 years ago. This is what I did today on a clean Fedora installation:

- install the `bluez-tools` RPM **before** you try to pair the device (otherwise delete the pairing)
- plug in the device via USB, and press the PS button, and you should get a bluetooth connect prompt, together with a prompt for a PIN code. But neither the default random PIN code offered by Gnome, nor the supposed correct "0000" code would work, and the player indicator LEDs would keep blinking, and after a bit it would prompt for the PIN code again.
- At this point the USB cable is no longer needed, and each press of the PS button will cause it to try to connect, which will fail each time.

Trying the 3rd party controllers fared even worse. I would get the connection request, which I could accept, but not the PIN code prompt. Again the player LEDs would keep blinking and after a bit they would turn off.

So it seems like support for PS3 controllers may have regressed further.

The good thing, at least, is that PS3 controllers are easily emulated with any more modern controller.

mirh, 2024-09-29:
Mhh.. what is your bt dongle? I know their support got a lot of highs and lows during the last few years.

rderooy, 2024-09-29:
It is indeed not on the same hardware that I did the testing today, compared to 4 years ago.

Today's system was an AMD Zen based ThinkPad laptop with integrated Bluetooth. With USB device ID 10ab:9309.

Testing 4 years ago was on an Intel based ThinkPad X1 Yoga Gen1, that is in a cupboard somewhere, but I'm pretty sure has an Intel Bluetooth controller (it for sure had Intel WiFi).

mirh, 2024-10-01:
Sorry, I meant the support for cheap CSR dongle knockoffs had ups and downs. 
But those seem pretty legit (even though I'll grant it's the first time I hear about Universal Scientific Industrial)...

So I don't really know other than trying windows or older linux distros to see if \*something\* can at least make a difference.

rderooy, 2025-07-05:
Ohh nice, auto closing unsolved issues

closed 2025-10-07 by github-actions[bot]