bluez/bluez #37: failed to connect Huawei Mouse

ArisAachen, 2020-09-10:
-- Logs begin at Fri 2020-09-04 12:12:06 CST, end at Tue 2020-09-08 19:40:29 CST. --
9月 08 14:02:33 qw010 bluetoothd[2212]: Bluetooth daemon 5.50
9月 08 14:02:33 qw010 bluetoothd[2212]: Starting SDP server
9月 08 14:02:33 qw010 bluetoothd[2212]: Bluetooth management interface 1.16 initialized
9月 08 14:02:33 qw010 bluetoothd[2212]: Sap driver initialization failed.
9月 08 14:02:33 qw010 bluetoothd[2212]: sap-server: Operation not permitted (1)
9月 08 14:02:39 qw010 bluetoothd[2212]: Endpoint registered: sender=:1.69 path=/MediaEndpoint/A2DPSource
9月 08 14:02:39 qw010 bluetoothd[2212]: Endpoint registered: sender=:1.69 path=/MediaEndpoint/A2DPSink
9月 08 19:38:53 qw010 bluetoothd[2212]: Terminating
9月 08 19:38:53 qw010 bluetoothd[2212]: Endpoint unregistered: sender=:1.69 path=/MediaEndpoint/A2DPSource
9月 08 19:38:53 qw010 bluetoothd[2212]: Endpoint unregistered: sender=:1.69 path=/MediaEndpoint/A2DPSink
9月 08 19:38:53 qw010 bluetoothd[2212]: Stopping SDP server
9月 08 19:38:53 qw010 bluetoothd[2212]: Exit
9月 08 19:38:53 qw010 bluetoothd[8888]: Bluetooth daemon 5.50
9月 08 19:38:53 qw010 bluetoothd[8888]: src/main.c:parse_config() parsing /etc/bluetooth/main.conf
9月 08 19:38:53 qw010 bluetoothd[8888]: src/main.c:parse_config() Key file does not have key “DiscoverableTimeout” in group “General”
9月 08 19:38:53 qw010 bluetoothd[8888]: src/main.c:parse_config() Key file does not have key “PairableTimeout” in group “General”
9月 08 19:38:53 qw010 bluetoothd[8888]: src/main.c:parse_config() Key file does not have key “Privacy” in group “General”
9月 08 19:38:53 qw010 bluetoothd[8888]: src/main.c:parse_config() Key file does not have key “Name” in group “General”
9月 08 19:38:53 qw010 bluetoothd[8888]: src/main.c:parse_config() Key file does not have key “Class” in group “General”
9月 08 19:38:53 qw010 bluetoothd[8888]: src/main.c:parse_config() Key file does not have key “DeviceID” in group “General”
9月 08 19:38:53 qw010 bluetoothd[8888]: src/main.c:parse_config() Key file does not have key “ReverseServiceDiscovery” in group “General”
9月 08 19:38:53 qw010 bluetoothd[8888]: src/main.c:parse_config() Key file does not have key “MinEncKeySize” in group “GATT”
9月 08 19:38:53 qw010 bluetoothd[8888]: src/adapter.c:adapter_init() sending read version command
9月 08 19:38:53 qw010 bluetoothd[8888]: Starting SDP server
9月 08 19:38:53 qw010 bluetoothd[8888]: src/sdpd-service.c:register_device_id() Adding device id record for 0002:1d6b:0246:0532
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:plugin_init() Loading builtin plugins
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:add_plugin() Loading hostname plugin
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:add_plugin() Loading wiimote plugin
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:add_plugin() Loading autopair plugin
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:add_plugin() Loading policy plugin
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:add_plugin() Loading neard plugin
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:add_plugin() Loading sap plugin
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:add_plugin() Loading a2dp plugin
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:add_plugin() Loading avrcp plugin
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:add_plugin() Loading network plugin
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:add_plugin() Loading input plugin
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:add_plugin() Loading hog plugin
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:add_plugin() Loading health plugin
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:add_plugin() Loading gap plugin
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:add_plugin() Loading scanparam plugin
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:add_plugin() Loading deviceinfo plugin
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:add_plugin() Loading midi plugin
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:add_plugin() Loading battery plugin
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:plugin_init() Loading plugins /usr/lib/x86_64-linux-gnu/bluetooth/plugins
9月 08 19:38:53 qw010 bluetoothd[8888]: src/plugin.c:add_plugin() Loading sixaxis plugin
9月 08 19:38:53 qw010 bluetoothd[8888]: profiles/health/hdp.c:hdp_manager_start() Starting Health manager
9月 08 19:38:53 qw010 bluetoothd[8888]: profiles/input/suspend-none.c:suspend_init()
9月 08 19:38:53 qw010 bluetoothd[8888]: profiles/network/manager.c:read_config() /etc/bluetooth/network.conf: Key file does not have key “DisableSecurity” in group “General”
9月 08 19:38:53 qw010 bluetoothd[8888]: profiles/network/manager.c:read_config() Config options: Security=true
9月 08 19:38:53 qw010 bluetoothd[8888]: plugins/neard.c:neard_init() Setup neard plugin
9月 08 19:38:53 qw010 bluetoothd[8888]: plugins/hostname.c:read_dmi_fallback() chassis: laptop
9月 08 19:38:53 qw010 bluetoothd[8888]: plugins/hostname.c:read_dmi_fallback() major: 0x01 minor: 0x03
9月 08 19:38:53 qw010 bluetoothd[8888]: plugins/sixaxis.c:sixaxis_init()
9月 08 19:38:53 qw010 bluetoothd[8888]: src/main.c:main() Entering main loop
9月 08 19:38:53 qw010 bluetoothd[8888]: src/rfkill.c:rfkill_event() RFKILL event idx 0 type 1 op 0 soft 0 hard 0
9月 08 19:38:53 qw010 bluetoothd[8888]: Bluetooth management interface 1.16 initialized
9月 08 19:38:53 qw010 bluetoothd[8888]: src/adapter.c:read_version_complete() sending read supported commands command
9月 08 19:38:53 qw010 bluetoothd[8888]: src/adapter.c:read_version_complete() sending read index list command
9月 08 19:38:53 qw010 bluetoothd[8888]: src/rfkill.c:rfkill_event() RFKILL event idx 1 type 2 op 0 soft 0 hard 0
9月 08 19:38:53 qw010 bluetoothd[8888]: src/adapter.c:read_commands_complete() Number of commands: 67
9月 08 19:38:53 qw010 bluetoothd[8888]: src/adapter.c:read_commands_complete() Number of events: 35
9月 08 19:38:53 qw010 bluetoothd[8888]: src/adapter.c:read_commands_complete() enabling kernel-side connection control
9月 08 19:38:53 qw010 bluetoothd[8888]: src/adapter.c:read_index_list_complete() Number of controllers: 1
9月 08 19:38:53 qw010 bluetoothd[8888]: src/adapter.c:read_index_list_complete() Found index 0
9月 08 19:38:53 qw010 bluetoothd[8888]: src/adapter.c:index_added() index 0
9月 08 19:38:53 qw010 bluetoothd[8888]: src/adapter.c:btd_adapter_new() System name: BlueZ 5.50
9月 08 19:38:53 qw010 bluetoothd[8888]: src/adapter.c:btd_adapter_new() Major class: 0
9月 08 19:38:53 qw010 bluetoothd[8888]: src/adapter.c:btd_adapter_new() Minor class: 0
9月 08 19:38:53 qw010 bluetoothd[8888]: src/adapter.c:btd_adapter_new() Modalias: usb:v1D6Bp0246d0532
[... 317 lines cut ...]
9月 08 19:39:59 qw010 bluetoothd[8888]: profiles/audio/a2dp.c:a2dp_source_server_remove() path /org/bluez/hci0
9月 08 19:39:59 qw010 bluetoothd[8888]: profiles/audio/a2dp.c:a2dp_sink_server_remove() path /org/bluez/hci0
9月 08 19:39:59 qw010 bluetoothd[8888]: profiles/audio/avrcp.c:avrcp_target_server_remove() path /org/bluez/hci0
9月 08 19:39:59 qw010 bluetoothd[8888]: src/adapter.c:adapter_service_remove() /org/bluez/hci0
9月 08 19:39:59 qw010 bluetoothd[8888]: src/adapter.c:remove_uuid() sending remove uuid command for index 0
9月 08 19:39:59 qw010 bluetoothd[8888]: src/sdpd-service.c:remove_record_from_server() Removing record with handle 0x10002
9月 08 19:39:59 qw010 bluetoothd[8888]: profiles/audio/avrcp.c:avrcp_controller_server_remove() path /org/bluez/hci0
9月 08 19:39:59 qw010 bluetoothd[8888]: src/adapter.c:adapter_service_remove() /org/bluez/hci0
9月 08 19:39:59 qw010 bluetoothd[8888]: src/adapter.c:remove_uuid() sending remove uuid command for index 0
9月 08 19:39:59 qw010 bluetoothd[8888]: src/sdpd-service.c:remove_record_from_server() Removing record with handle 0x10001
9月 08 19:39:59 qw010 bluetoothd[8888]: profiles/network/manager.c:nap_server_remove() path /org/bluez/hci0
9月 08 19:39:59 qw010 bluetoothd[8888]: profiles/network/manager.c:gn_server_remove() path /org/bluez/hci0
9月 08 19:39:59 qw010 bluetoothd[8888]: profiles/network/manager.c:panu_server_remove() path /org/bluez/hci0
9月 08 19:39:59 qw010 bluetoothd[8888]: profiles/network/server.c:path_unregister() Unregistered interface org.bluez.NetworkServer1 on path /org/bluez/hci0
9月 08 19:39:59 qw010 bluetoothd[8888]: src/adapter.c:btd_adapter_unref() Freeing adapter /org/bluez/hci0
9月 08 19:39:59 qw010 bluetoothd[8888]: src/adapter.c:adapter_free() 0xfe4680
9月 08 19:39:59 qw010 bluetoothd[8888]: Stopping SDP server
[... cut ...]
9月 08 19:40:00 qw010 bluetoothd[9040]: Starting SDP server
9月 08 19:40:00 qw010 bluetoothd[9040]: Bluetooth management interface 1.16 initialized
9月 08 19:40:00 qw010 bluetoothd[9040]: Sap driver initialization failed.
9月 08 19:40:00 qw010 bluetoothd[9040]: sap-server: Operation not permitted (1)
9月 08 19:40:00 qw010 bluetoothd[9040]: Endpoint registered: sender=:1.69 path=/MediaEndpoint/A2DPSource
9月 08 19:40:00 qw010 bluetoothd[9040]: Endpoint registered: sender=:1.69 path=/MediaEndpoint/A2DPSink

ArisAachen, 2020-09-10:
[bluetooth]# pair E5:4F:38:27:34:DD
Attempting to pair with E5:4F:38:27:34:DD
[CHG] Device E5:4F:38:27:34:DD Connected: yes
[CHG] Device E5:4F:38:27:34:DD Connected: no
Failed to pair: org.bluez.Error.AuthenticationCanceled

ArisAachen, 2020-09-10:
sudo service bluetooth status                                                                                                                                                     
● bluetooth.service - Bluetooth service                                                                                                                        
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)                                                                     
   Active: active (running) since Thu 2020-09-03 13:17:45 CST; 48min ago                                                                                       
     Docs: man:bluetoothd(8)                                                                                                                                   
Main PID: 4124 (bluetoothd)                                                                                                                                   
   Status: "Running"                                                                                                                                          
    Tasks: 1 (limit: 4915)                                                                                                                                    
   Memory: 4.0M                                                                                                                                                
   CGroup: /system.slice/bluetooth.service                                                                                                                     
           └─4124 /usr/lib/bluetooth/bluetoothd                                                                                                               
                                                                                                                                                               
9月 03 13:17:45 qw010 systemd[1]: Starting Bluetooth service...                                                                                                
9月 03 13:17:45 qw010 bluetoothd[4124]: Bluetooth daemon 5.50
9月 03 13:17:45 qw010 systemd[1]: Started Bluetooth service.
9月 03 13:17:45 qw010 bluetoothd[4124]: Starting SDP server
9月 03 13:17:45 qw010 bluetoothd[4124]: Bluetooth management interface 1.16 initialized
9月 03 13:17:45 qw010 bluetoothd[4124]: Sap driver initialization failed.
9月 03 13:17:45 qw010 bluetoothd[4124]: sap-server: Operation not permitted (1)
9月 03 13:18:11 qw010 bluetoothd[4124]: Endpoint registered: sender=:1.77 path=/MediaEndpoint/A2DPSource
9月 03 13:18:11 qw010 bluetoothd[4124]: Endpoint registered: sender=:1.77 path=/MediaEndpoint/A2DPSink

KingsWorldProject, 2020-10-10:
Use hcidump to grab the failed log

Vudentz, 2021-01-21:
No HCI has been provided, closing.

closed 2021-01-21 by Vudentz