bluez/bluez #38: IOS phone are not able to write characteristics to bluez

p1ng0o, 2020-09-23:
Hi,

I still observe the same issue described in the mailing list bellow:
https://marc.info/?l=linux-bluetooth&m=152154683807490&w=2

When an IOS device is connected to the bluez GATT server, bluez systematically try to read the battery level characteristic. (since it's advertised by the Iphone)

The read request failed due to the lack of pairing. (used in a app with just std ble connection)

Then, all the write_request_with_response made by the Iphone, fails.

The only founded workaround is to disable the battery profile in bluez binary.
Is there something messy in the pending message list ? A failed read shouldn't avoid to send write response.

Thanks

Vudentz, 2020-11-13:
This is by design actually, since the remote is telling that pairing is required ATT socket will be blocked until the pairing is completed.

closed 2020-11-13 by Vudentz

lirengen10, 2020-12-25:
Hi Antoine Aubert,
I also saw this issue at bluez5.50 , can you please tell me how to disable the battery profile  in bluez binary.
Thanks.