bluez/bluez #39: Meshctl sub-add with vendor id ?

Ripitchip, 2020-09-29:
To my surprise, I saw that meshctl doesn't provide a sub-set with [cid] parameter.. ?
If I look to all commands in menu config, I can read this :

- pub-set <ele_addr> <pub_addr> <app_idx> <per (step|res)> <re-xmt (cnt|per)> [cid]
- bind <ele_idx> <app_idx> <mod_id> [cid]
- sub-add <ele_addr> <sub_addr>

Counterfeiter, 2020-10-21:
Same surprise here... 
Is there any (spec) reason for that? Also the mesh-client command has the CID addition.

@Ripitchip did you add it since your report?

Ripitchip, 2020-10-21:
@Counterfeiter 

No I didn't try to add anything. For two reasons :
1) It seems it was already added, if you check the commits for example [here](https://github.com/bluez/bluez/commit/5474314d895e802ede83c9862c81013707249be2). That's why I was so suprised.
2) If you use the nRF Mesh application, it's possible to bind, publish and subscribe a model vendor. So that's it, I left meshctl for nRF Mesh app, it's enough so far for my project.

Best regards

bgix, 2021-01-11:
meshctl is on it way towards deprecation, although we may preserve some form of it as a test GATT Client app to test remote Mesh Proxy Server and PB-GATT implementations. 

For ongoing mesh development with BlueZ, we are encouraging users to use the bluetooth-meshd daemon, and the tools that talk to the daemon:  tools/mesh-cfgclient and test/test-mesh

hashing-cell, 2021-11-18:
Hi, I have submitted a commit for this issue, so hopefully in the new update this is no longer an issue

closed 2025-08-31 by github-actions[bot]