bluez/bluez #40: bluez 5.55 unit/test-mesh-crypto fails

ncopa, 2020-10-06:
The test suite failed on alpine linux x86_64 (musl libc 1.2.1).
```
Linux ncopa-edge-x86_64 4.19.80-0-vanilla #1-Alpine SMP Fri Oct 18 11:27:53 UTC 2019 x86_64 GNU/Linux
```

```
...
FAIL: unit/test-mesh-crypto
============================================================================
Testsuite summary for bluez 5.55
============================================================================
# TOTAL: 26
# PASS:  25
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0
============================================================================
See ./test-suite.log
============================================================================
```

Content of test-suite.log:
```
==================================
   bluez 5.55: ./test-suite.log
==================================

# TOTAL: 26
# PASS:  25
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0

.. contents:: :depth: 2

FAIL: unit/test-mesh-crypto
===========================

[0;94m[8.1.1 s1 SALT generation function][0m
Salt Input           = test
Salt Input           = 74657374
s1(Salt)             = b73cefbd641ef2ea598c2b6efb62f79c
                       00000000000000000000000000000000 => [0;91mFAIL[0m
FAIL unit/test-mesh-crypto (exit status: 1)
```

ncopa, 2020-10-06:
Testsuite passes on aarch64, ppc64le and x86 (32bit), so it seems like it fails if there is AES hardware support?

ncopa, 2020-10-06:
And it passes on my i9 work desktop. This is weird. Why does it fail on that one build server, but works everywhere else I try?

This is the CPU it fails on:
```
...
processor       : 47
vendor_id       : GenuineIntel
cpu family      : 6
model           : 79
model name      : Intel(R) Xeon(R) CPU E5-2650 v4 @ 2.20GHz
stepping        : 1
microcode       : 0xb00001e
cpu MHz         : 1197.893
cache size      : 30720 KB
physical id     : 1
siblings        : 24
core id         : 13
cpu cores       : 12
apicid          : 59
initial apicid  : 59
fpu             : yes
fpu_exception   : yes
cpuid level     : 20
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid dca sse4_1 sse4_2 x2apic movbe popcnt aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb cat_l3 cdp_l3 invpcid_single pti tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm cqm rdt_a rdseed adx smap intel_pt xsaveopt cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local dtherm ida arat pln pts
bugs            : cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs
bogomips        : 4391.07
clflush size    : 64
cache_alignment : 64
address sizes   : 46 bits physical, 48 bits virtual
power management:
```

bgix, 2020-10-07:
I need to see the Crypto settings on the Kernel build...  This is usually the problem.

For the 8.1.1 (s1 Salt) test, a minimum AES, CMAC and the User APIs need to be enabled. I believe all of the required Kernel Build Options for Mesh are included in the README file in .../bluez/README.  This have been winnowed down over time as the minimum options.

closed 2021-01-08 by Vudentz