bluez/bluez #42: HoG profile accept failed if non-encrypted/paired mode

v01d, 2020-10-08:
I'm developing a HoG device and since I wanted to try connection without pairing/encryption I disabled `LEAutoSecurity`. However, when doing this the hog profile is not accepted for my device according to bluetoothd log. You can see the log here:

```
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/adapter.c:connected_callback() hci0 device AA:AA:AA:AA:AA:AA connected eir_len 22
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/gatt-database.c:connect_cb() New incoming LE ATT connection
Oct  8 15:15:35 pop-os bluetoothd[50800]: attrib/gattrib.c:g_attrib_ref() 0x55a0e9e92210: g_attrib_ref=1 
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:load_gatt_db() Restoring AA:AA:AA:AA:AA:AA gatt database from file
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:load_service() loading service: 0x0001, end: 0x0005, uuid: 00001800-0000-1000-8000-00805f9b34fb
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:load_service() loading service: 0x0006, end: 0x0008, uuid: 0000180a-0000-1000-8000-00805f9b34fb
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:load_service() loading service: 0x0009, end: 0x0013, uuid: 00001812-0000-1000-8000-00805f9b34fb
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:load_service() loading service: 0x0014, end: 0x0017, uuid: 0000180f-0000-1000-8000-00805f9b34fb
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:load_service() loading service: 0x0018, end: 0xffff, uuid: 00001801-0000-1000-8000-00805f9b34fb
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:load_chrc() loading characteristic handle: 0x0002, value handle: 0x0003, properties 0x0002 value:  uuid: 00002a00-0000-1000-8000-00805f9b34fb
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:load_chrc() loading characteristic handle: 0x0004, value handle: 0x0005, properties 0x0002 value:  uuid: 00002a01-0000-1000-8000-00805f9b34fb
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:load_chrc() loading characteristic handle: 0x0007, value handle: 0x0008, properties 0x0002 value:  uuid: 00002a24-0000-1000-8000-00805f9b34fb
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:load_chrc() loading characteristic handle: 0x000a, value handle: 0x000b, properties 0x0012 value:  uuid: 00002a4d-0000-1000-8000-00805f9b34fb
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:load_desc() loading descriptor handle: 0x000c, value: 0x0000, value uuid: 00002902-0000-1000-8000-00805f9b34fb
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:load_desc() loading descriptor handle: 0x000d, value: 0x0000, value uuid: 00002908-0000-1000-8000-00805f9b34fb
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:load_chrc() loading characteristic handle: 0x000e, value handle: 0x000f, properties 0x0002 value:  uuid: 00002a4b-0000-1000-8000-00805f9b34fb
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:load_chrc() loading characteristic handle: 0x0010, value handle: 0x0011, properties 0x0002 value:  uuid: 00002a4a-0000-1000-8000-00805f9b34fb
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:load_chrc() loading characteristic handle: 0x0012, value handle: 0x0013, properties 0x0004 value:  uuid: 00002a4c-0000-1000-8000-00805f9b34fb
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:load_chrc() loading characteristic handle: 0x0015, value handle: 0x0016, properties 0x0012 value:  uuid: 00002a19-0000-1000-8000-00805f9b34fb
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:load_desc() loading descriptor handle: 0x0017, value: 0x0000, value uuid: 00002902-0000-1000-8000-00805f9b34fb
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:load_chrc() loading characteristic handle: 0x0019, value handle: 0x001a, properties 0x0020 value:  uuid: 00002a05-0000-1000-8000-00805f9b34fb
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:load_desc() loading descriptor handle: 0x001b, value: 0x0000, value uuid: 00002902-0000-1000-8000-00805f9b34fb
Oct  8 15:15:35 pop-os bluetoothd[50800]: profiles/battery/battery.c:batt_accept() BATT profile accept (AA:AA:AA:AA:AA:AA)
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/service.c:change_state() 0x55a0e9e9c9f0: device AA:AA:AA:AA:AA:AA profile batt-profile state changed: disconnected -> connected (0)
Oct  8 15:15:35 pop-os bluetoothd[50800]: profiles/deviceinfo/deviceinfo.c:deviceinfo_accept() deviceinfo profile accept (AA:AA:AA:AA:AA:AA)
Oct  8 15:15:35 pop-os bluetoothd[50800]: profiles/deviceinfo/deviceinfo.c:handle_characteristic() Unsupported characteristic: 00002a24-0000-1000-8000-00805f9b34fb
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/service.c:change_state() 0x55a0e9e9f950: device AA:AA:AA:AA:AA:AA profile deviceinfo state changed: disconnected -> connected (0)
Oct  8 15:15:35 pop-os bluetoothd[50800]: profiles/gap/gas.c:gap_accept() GAP profile accept (AA:AA:AA:AA:AA:AA)
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/service.c:change_state() 0x55a0e9e9ac80: device AA:AA:AA:AA:AA:AA profile gap-profile state changed: disconnected -> connected (0)
Oct  8 15:15:35 pop-os bluetoothd[50800]: profiles/input/hog.c:hog_device_accept() name=thumbmouse vendor=0x0, product=0x0, version=0x0
Oct  8 15:15:35 pop-os bluetoothd[50800]: input-hog profile accept failed for AA:AA:AA:AA:AA:AA
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/gatt-client.c:btd_gatt_client_connected() Device connected.
Oct  8 15:15:35 pop-os bluetoothd[50800]: src/device.c:gatt_debug() MTU exchange complete, with MTU: 256
Oct  8 15:15:35 pop-os bluetoothd[50800]: profiles/gap/gas.c:read_device_name_cb() GAP Device Name: thumbmouse
Oct  8 15:15:35 pop-os bluetoothd[50800]: profiles/gap/gas.c:read_appearance_cb() GAP Appearance: 0x03c0
Oct  8 15:15:36 pop-os bluetoothd[50800]: src/device.c:gatt_debug() Primary services found: 5
Oct  8 15:15:36 pop-os bluetoothd[50800]: src/device.c:gatt_debug() start: 0x0001, end: 0x0005, uuid: 00001800-0000-1000-8000-00805f9b34fb
Oct  8 15:15:36 pop-os bluetoothd[50800]: src/device.c:gatt_debug() start: 0x0006, end: 0x0008, uuid: 0000180a-0000-1000-8000-00805f9b34fb
Oct  8 15:15:36 pop-os bluetoothd[50800]: src/device.c:gatt_debug() start: 0x0009, end: 0x0013, uuid: 00001812-0000-1000-8000-00805f9b34fb
Oct  8 15:15:36 pop-os bluetoothd[50800]: src/device.c:gatt_debug() start: 0x0014, end: 0x0017, uuid: 0000180f-0000-1000-8000-00805f9b34fb
Oct  8 15:15:36 pop-os bluetoothd[50800]: src/device.c:gatt_debug() start: 0x0018, end: 0xffff, uuid: 00001801-0000-1000-8000-00805f9b34fb
Oct  8 15:15:36 pop-os bluetoothd[50800]: src/device.c:gatt_client_ready_cb() status: success, error: 0
Oct  8 15:15:36 pop-os bluetoothd[50800]: input-hog profile accept failed for AA:AA:AA:AA:AA:AA
Oct  8 15:15:36 pop-os bluetoothd[50800]: src/gatt-client.c:btd_gatt_client_ready() GATT client ready
Oct  8 15:15:36 pop-os bluetoothd[50800]: src/gatt-client.c:create_services() Exporting objects for GATT services: AA:AA:AA:AA:AA:AA
Oct  8 15:15:36 pop-os bluetoothd[50800]: src/gatt-client.c:service_create() Exported GATT service: /org/bluez/hci0/dev_AA_AA_AA_AA_AA_AA/service0006
Oct  8 15:15:36 pop-os bluetoothd[50800]: src/gatt-client.c:characteristic_create() Exported GATT characteristic: /org/bluez/hci0/dev_AA_AA_AA_AA_AA_AA/service0006/char0007
Oct  8 15:15:36 pop-os bluetoothd[50800]: src/gatt-client.c:service_create() Exported GATT service: /org/bluez/hci0/dev_AA_AA_AA_AA_AA_AA/service0018
Oct  8 15:15:36 pop-os bluetoothd[50800]: src/gatt-client.c:characteristic_create() Exported GATT characteristic: /org/bluez/hci0/dev_AA_AA_AA_AA_AA_AA/service0018/char0019
Oct  8 15:15:36 pop-os bluetoothd[50800]: src/gatt-client.c:descriptor_create() Exported GATT characteristic descriptor: /org/bluez/hci0/dev_AA_AA_AA_AA_AA_AA/service0018/char0019/desc001b
Oct  8 15:15:36 pop-os bluetoothd[50800]: src/device.c:device_svc_resolved() /org/bluez/hci0/dev_AA_AA_AA_AA_AA_AA err 0
Oct  8 15:15:36 pop-os bluetoothd[50800]: profiles/battery/battery.c:parse_battery_level() Battery Level updated: 15%
Oct  8 15:15:36 pop-os bluetoothd[50800]: profiles/battery/battery.c:batt_io_ccc_written_cb() Battery Level: notification enabled
Oct  8 15:15:36 pop-os bluetoothd[50800]: src/device.c:gatt_debug() Registered handler for "Service Changed": 1
Oct  8 15:22:34 pop-os bluetoothd[50800]: Terminating
Oct  8 15:22:34 pop-os bluetoothd[50800]: src/adapter.c:adapter_shutdown() 
Oct  8 15:22:34 pop-os bluetoothd[50800]: src/adapter.c:set_mode() sending set mode command for index 0
[... cut ...]
Oct  8 15:22:34 pop-os bluetoothd[50800]: src/service.c:change_state() 0x55a0e9e9f950: device AA:AA:AA:AA:AA:AA profile deviceinfo state changed: disconnecting -> disconnected (0)
Oct  8 15:22:34 pop-os bluetoothd[50800]: src/service.c:change_state() 0x55a0e9e9ac80: device AA:AA:AA:AA:AA:AA profile gap-profile state changed: connected -> disconnecting (0)
Oct  8 15:22:34 pop-os bluetoothd[50800]: src/service.c:change_state() 0x55a0e9e9ac80: device AA:AA:AA:AA:AA:AA profile gap-profile state changed: disconnecting -> disconnected (0)
Oct  8 15:22:34 pop-os bluetoothd[50800]: src/gatt-client.c:btd_gatt_client_disconnected() Device disconnected. Cleaning up.
```

v01d, 2020-10-08:
I'm using BlueZ 5.53 in PopOS 20.04, btw

sjanc, 2020-11-17:
afair security is mandatory for HoG

closed 2025-08-31 by github-actions[bot]