bluez/bluez #43: Question: How to override RequestPasskey (Python)

dhoomakethu, 2020-10-13:
Using Python example for [simple-agent](https://github.com/bluez/bluez/blob/master/test/simple-agent#L75), I am trying to run the agent with `DisplayOnly` capability and would like to return a fixed pass key always . But the passkey is being generated outside the scope of python script. Is there a way to override this behaviour?

Running bluez 5.48-1 snap on ubuntu core 16 machine with out keyboard and display.

```
$ sudo snap list bluez
Name   Version  Rev  Tracking  Publisher   Notes
bluez  5.48-1   229  20/edge   canonical✓  -
```

Vudentz, 2020-10-13:
You can always modify the simple-agent to not ask for user input if that what you mean.

dhoomakethu, 2020-10-14:
Yes, I have modified the code to return the passkey of my choice but the problem is `RequestPasskey` is being called only when I set the capability to `KeyboardOnly` and the passkey is generated by the initiator is expected to be keyed in. Where as I am trying to control the generation of passkey, which seems to be happening automatically when I set the capability to `DisplayOnly`. 

I am running a gatt-server with custom agent which runs as a system daemon on a headless server. There is no way the Passkey can be displayed or keyed in.

mikael-marvin, 2024-03-27:
Hi dhoomakethu, is there a way i can contact you? i would like to ask you some questions about making a custom agent in python using bluez and dbus...
im having a hard time finding resources about it and im rly new to all this

closed 2025-08-31 by github-actions[bot]