bluez/bluez #44: Using flags `authorize` and `encrypted-auth-read/write`

dhoomakethu, 2020-10-13:
Hello,

I am playing around with [example gatt server](https://github.com/bluez/bluez/blob/master/test/example-gatt-server) and was trying to use the flags `authorize` and `encrypt-auth-read` in the [TestEncryptCharacteristic](https://github.com/bluez/bluez/blob/master/test/example-gatt-server#L533) class. The server would fail to start with the below error
`ERROR Erroring during registration of application - org.bluez.Error.Failed: No valid service object found`

Any idea/examples on using these flags?

dhoomakethu, 2020-10-13:
Looks like `authorize` is not recognized in my version of bluez.
```
Oct 13 15:39:56 bluetoothd[27934]: Bluetooth management interface 1.10 initialized
Oct 13 18:38:07 bluetoothd[27934]: Invalid characteristic flag: authorize
Oct 13 18:38:07 bluetoothd[27934]: Failed to parse characteristic properties
Oct 13 18:38:07 bluetoothd[27934]: No valid external GATT objects found
```

Vudentz, 2020-10-13:
@dhoomakethu are you sure you are running with a version of bluetoothd that does support authorize flags?

You could try using bluetoothctl:

> menu gatt
> register-service..
> register-characteristic <UUID> <Flags=read,write,notify...> [handle]

Under flags set the flags you want. Btw, tipically authorize is only used for qualification purpose, since PTS want to test an specific error can be generated, usually apps can do the authorization implicitly during the operation.

dhoomakethu, 2020-10-14:
Thanks @Vudentz , the version of bluez was not supporting `authorize` so had to upgrade the bluez. I was trying to find the scenario in which this [part](https://github.com/bluez/bluez/blob/master/test/simple-agent#L59) of the code is hit, so was wild guessing using `encrypt-auth-read` or `read` should help me with it.

closed 2021-01-08 by Vudentz