bluez/bluez #45: segfault in media_endpoint_destroy

sileht, 2020-10-13:
Hi, since archlinux have upgraded bluez to 5.55 it got a segfault:

```
# /usr/lib/bluetooth/bluetoothd
free(): double free detected in tcache 2
Aborted (core dumped)
```

So I recompiled the package with symbol and get the following backtrace:

```
(gdb) bt
#0  0x00007ffff7b9e615 in raise () from /usr/lib/libc.so.6
#1  0x00007ffff7b87862 in abort () from /usr/lib/libc.so.6
#2  0x00007ffff7be05e8 in __libc_message () from /usr/lib/libc.so.6
#3  0x00007ffff7be827a in malloc_printerr () from /usr/lib/libc.so.6
#4  0x00007ffff7be9d4c in _int_free () from /usr/lib/libc.so.6
#5  0x000055555558735e in media_endpoint_destroy (endpoint=0x5555556c4ce0) at profiles/audio/media.c:180
#6  0x0000555555587b63 in media_endpoint_create (adapter=adapter@entry=0x5555556c3480, sender=sender@entry=0x5555556be868 ":1.2566", path=0x5555556c7534 "/MediaEndpoint/A2DPSink/sbc",
    uuid=0x5555556c7568 "0000110b-0000-1000-8000-00805f9b34fb", delay_reporting=0, codec=<optimized out>, capabilities=0x5555556c75bc "\377\377\002\065", size=4, err=0x7fffffffe340) at profiles/audio/media.c:823
#7  0x00005555555881e4 in register_endpoint (conn=<optimized out>, msg=0x5555556bd580, data=0x5555556c3480) at profiles/audio/media.c:926
#8  0x0000555555604519 in process_message (connection=0x5555556927f0, message=0x5555556bd580, method=0x555555676ba0 <media_methods>, iface_user_data=<optimized out>) at gdbus/object.c:259
#9  0x00007ffff7e357d6 in ?? () from /usr/lib/libdbus-1.so.3
#10 0x00007ffff7e253bd in dbus_connection_dispatch () from /usr/lib/libdbus-1.so.3
#11 0x0000555555600bb1 in message_dispatch (data=0x5555556927f0) at gdbus/mainloop.c:72
#12 0x00007ffff7eb5924 in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
#13 0x00007ffff7f09621 in ?? () from /usr/lib/libglib-2.0.so.0
#14 0x00007ffff7eb4e73 in g_main_loop_run () from /usr/lib/libglib-2.0.so.0
#15 0x0000555555619a06 in mainloop_run () at src/shared/mainloop-glib.c:79
#16 0x0000555555619e88 in mainloop_run_with_signal (func=<optimized out>, user_data=0x0) at src/shared/mainloop-notify.c:201
#17 0x0000555555574ac1 in main (argc=<optimized out>, argv=<optimized out>) at src/main.c:971
```

IAlso  notice it's related to `MultiProfile = multiple`, I don't have crash with `MultiProfile = off`

Vudentz, 2020-10-13:
I tried with MultiProfile=multiple but I don't see this error, do you have the daemon logs when the issue happens? Also does it manifest right when bluetoothd is loaded?

sileht, 2020-10-14:
It occurs when bluetoohd start.

My configuration change is exactly:
```
AutoEnable=true
MultiProfile = multiple
```

The complete log before the crash:

```
# /usr/lib/bluetooth/bluetoothd -n -d
bluetoothd[2209330]: Bluetooth daemon 5.55
bluetoothd[2209330]: src/main.c:parse_config() parsing /etc/bluetooth/main.conf
bluetoothd[2209330]: src/main.c:parse_config() Key file does not have key “DiscoverableTimeout” in group “General”
bluetoothd[2209330]: src/main.c:parse_config() Key file does not have key “AlwaysPairable” in group “General”
bluetoothd[2209330]: src/main.c:parse_config() Key file does not have key “PairableTimeout” in group “General”
bluetoothd[2209330]: src/main.c:parse_config() Key file does not have key “Privacy” in group “General”
bluetoothd[2209330]: src/main.c:parse_config() Key file does not have key “JustWorksRepairing” in group “General”
bluetoothd[2209330]: src/main.c:parse_config() Key file does not have key “TemporaryTimeout” in group “General”
bluetoothd[2209330]: src/main.c:parse_config() Key file does not have key “Name” in group “General”
bluetoothd[2209330]: src/main.c:parse_config() Key file does not have key “Class” in group “General”
bluetoothd[2209330]: src/main.c:parse_config() Key file does not have key “DeviceID” in group “General”
bluetoothd[2209330]: src/main.c:parse_config() Key file does not have key “ReverseServiceDiscovery” in group “General”
bluetoothd[2209330]: src/main.c:parse_config() MultiProfile=multiple
bluetoothd[2209330]: src/main.c:parse_config() Key file does not have key “Cache” in group “GATT”
bluetoothd[2209330]: src/main.c:parse_config() Key file does not have key “KeySize” in group “GATT”
bluetoothd[2209330]: src/main.c:parse_config() Key file does not have key “ExchangeMTU” in group “GATT”
bluetoothd[2209330]: src/main.c:parse_config() Key file does not have key “Channels” in group “GATT”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/main.c:parse_controller_config() Key file does not have group “Controller”
bluetoothd[2209330]: src/adapter.c:adapter_init() sending read version command
bluetoothd[2209330]: Starting SDP server
bluetoothd[2209330]: src/sdpd-service.c:register_device_id() Adding device id record for 0002:1d6b:0246:0537
bluetoothd[2209330]: src/sdpd-service.c:add_record_to_server() Adding record with handle 0x10001
bluetoothd[2209330]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00001002-0000-1000-8000-00805f9
bluetoothd[2209330]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 0000113a-0000-1000-8000-00805f9
bluetoothd[2209330]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 0000113b-0000-1000-8000-00805f9
bluetoothd[2209330]: src/plugin.c:plugin_init() Loading builtin plugins
bluetoothd[2209330]: src/plugin.c:add_plugin() Loading hostname plugin
[... cut ...]
bluetoothd[2209330]: profiles/audio/a2dp.c:a2dp_server_listen() setsockopt(L2CAP_OPTIONS): Invalid argument (22)
bluetoothd[2209330]: profiles/audio/media.c:release_endpoint() sender=:1.9229 path=/MediaEndpoint/A2DPSink/sbc
bluetoothd[2209330]: Endpoint unregistered: sender=:1.9229 path=/MediaEndpoint/A2DPSink/sbc
bluetoothd[2209330]: profiles/audio/media.c:media_endpoint_destroy() sender=:1.9229 path=/MediaEndpoint/A2DPSink/sbc
bluetoothd[2209330]: profiles/audio/avdtp.c:avdtp_unregister_sep() SEP 0x55b983e9b910 unregistered: type:1 codec:0 seid:1
```

belzebub40k, 2020-12-27:
@sileht I came across the same problem when setting `MultiProfile = multiple`. Installing `bluez-git` from AUR fixed it for me so I guess this problem is fixed in the master branch.

sileht, 2020-12-29:
I confirm this is fixed in master (1:5.55.r197.g8b9ec615b-1)

closed 2025-08-31 by github-actions[bot]