bluez/bluez #46: SHANWAN PS3 GamePad - vibration issue when connected wirelessly

Ahmed-E-86, 2020-10-14:
Kubuntu 20.04.1
Kernel 5.8.14
Bluez 5.53

I have 3rd party controllers for PS3 (SHANWAN PS3 GamePad), it works fine when it is connected by the USB cable, but I have an issue with the vibration when it is connected wirelessly, the vibration works at full speed for around 10 seconds instead of vibrating for a second or two.

When I connect the controller by the USB cable, I get these messages from dmesg:

```
[21397.801573] usb 3-2: new full-speed USB device number 5 using xhci_hcd
[21397.951098] usb 3-2: New USB device found, idVendor=054c, idProduct=0268, bcdDevice= 1.00
[21397.951103] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[21397.951106] usb 3-2: Product: PS3 GamePad
[21397.951108] usb 3-2: Manufacturer: SHANWAN
[21397.955927] input: SHANWAN PS3 GamePad Motion Sensors as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/0003:054C:0268.0003/input/input29
[21398.013884] input: SHANWAN PS3 GamePad as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/0003:054C:0268.0003/input/input28
[21398.014462] sony 0003:054C:0268.0003: input,hiddev0,hidraw1: USB HID v81.10 Joystick [SHANWAN PS3 GamePad] on usb-0000:00:14.0-2/input0
[21461.499738] usb 3-2: USB disconnect, device number 5
```

but when it is connected by Bluetooth, I get these messages from dmesg:

```
[21471.515216] sony 0005:054C:0268.0004: unknown main item tag 0x0
[21471.530748] input: Sony PLAYSTATION(R)3 Controller Motion Sensors as /devices/pci0000:00/0000:00:14.0/usb3/3-12/3-12:1.0/bluetooth/hci0/hci0:42/0005:054C:0268.0004/input/input31
[21471.531075] input: Sony PLAYSTATION(R)3 Controller as /devices/pci0000:00/0000:00:14.0/usb3/3-12/3-12:1.0/bluetooth/hci0/hci0:42/0005:054C:0268.0004/input/input30
[21471.531283] sony 0005:054C:0268.0004: input,hidraw1: BLUETOOTH HID v80.00 Joystick [Sony PLAYSTATION(R)3 Controller] on 1c:65:9d:82:b0:1c
```

I think if Bluez recognizes the controller as `SHANWAN PS3 GamePad` instead of `Sony PLAYSTATION(R)3 Controller`, the controller will behave correctly, but I do not know how to do that.

I searched the internet, and I found a similar issue with the same controller:
https://forum.batocera.org/d/4471-vibration-generic-ps3-controller

jangofett4, 2020-12-08:
Yep I have the exact same problem and not sure whats the reason.
dmesg:
```
[ 9975.458006] usb 1-2: new full-speed USB device number 13 using xhci_hcd
[ 9975.614589] usb 1-2: New USB device found, idVendor=054c, idProduct=0268, bcdDevice= 1.00
[ 9975.614595] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 9975.614598] usb 1-2: Product: PS3 GamePad
[ 9975.614600] usb 1-2: Manufacturer: SHANWAN
[ 9975.636635] input: SHANWAN PS3 GamePad Motion Sensors as /devices/pci0000:00/0000:00:08.1/0000:05:00.3/usb1/1-2/1-2:1.0/0003:054C:0268.0010/input/input57
[ 9975.691210] input: SHANWAN PS3 GamePad as /devices/pci0000:00/0000:00:08.1/0000:05:00.3/usb1/1-2/1-2:1.0/0003:054C:0268.0010/input/input56
[ 9975.691518] sony 0003:054C:0268.0010: input,hiddev1,hidraw3: USB HID v81.10 Joystick [SHANWAN PS3 GamePad] on usb-0000:05:00.3-2/input0
```

BlueZ thinks its a genuine Sony Dualshock 3.

jangofett4, 2021-01-21:
Bump? I'm way too inexperienced with Linux drivers to fix this issue, but vibration doesn't seem to be handled in `plugins/sixaxis.c`. If someone can tell me where to look for I'm willing to try to fix this.

mrfixit2001, 2021-02-10:
The linux kernel HID driver handles the shanwan driver differently. This line:
https://github.com/torvalds/linux/blob/master/drivers/hid/hid-sony.c#L2987
sets the driver quirk dependent on the device name. If the device name isn't "SHANWAN PS3 GamePad" then this quirk is not set and the driver doesn't do some things, including a fix for the non-stop rumbling, as well as send different output reports.

The question now is... how can bluez correctly identify a shanwan device? I suppose as long as it pairs over USB first you should be able to pull the name from the usb device and override the one you've manually set in your struct. Shouldn't be too difficult but I'm not familiar enough with bluez to figure out how to pull the device name from the usb device.

sjanc, 2021-02-10:
yeap, thats the plan :-)  getting name from udev and pass it instead of default one, just need to find spare evening

sjanc, 2021-02-11:
Please check https://github.com/bluez/bluez/pull/95

Note that this may not be final version as I'm considering to simply always pass name read from USB (regardless of device connected)

jangofett4, 2021-02-12:
I'm trying to test the #95:
```sh
$ git clone -b shanwan https://github.com/sjanc/bluez
$ cd bluez
$ ./bootstrap-configure
$ ./configure --prefix=/usr --mandir=/usr/share/man --sysconfdir=/etc --localstatedir=/var
$ make
$ sudo make install
```
Build & install is successful, however vibration problem still exist. I'm suspicious of driver not actually being installed. How can I be sure if I'm using the Bluez version I just built?

sjanc, 2021-02-12:
sixaxis plugin is not enabled by default so you need to add --enable-sixaxis to configure

jangofett4, 2021-02-12:
Ok I recompiled and installed again. Currently its running the driver I just built for certain. Gamepad works however all 4 lights keep flashing (every half a second or so, not pairing light). Also vibration is not working in bluetooth mode altogether.
dmesg:
```
[  408.651311] usb 1-2: new full-speed USB device number 10 using xhci_hcd
[  408.810858] usb 1-2: New USB device found, idVendor=054c, idProduct=0268, bcdDevice= 1.00
[  408.810861] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  408.810863] usb 1-2: Product: PS3 GamePad
[  408.810864] usb 1-2: Manufacturer: SHANWAN
[  408.825863] sony 0003:054C:0268.0009: controller with MAC address 07:84:ae:6a:63:25 already connected
[  408.826023] sony 0003:054C:0268.0009: hiddev97,hidraw4: USB HID v81.10 Joystick [SHANWAN PS3 GamePad] on usb-0000:05:00.3-2/input0
[  408.826025] sony 0003:054C:0268.0009: failed to claim input
[  409.612832] usb 1-2: USB disconnect, device number 10
[  410.007983] usb 1-2: new full-speed USB device number 11 using xhci_hcd
[  410.164864] usb 1-2: New USB device found, idVendor=054c, idProduct=0268, bcdDevice= 1.00
[  410.164867] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  410.164869] usb 1-2: Product: PS3 GamePad
[  410.164870] usb 1-2: Manufacturer: SHANWAN
[  410.185897] input: SHANWAN PS3 GamePad Motion Sensors as /devices/pci0000:00/0000:00:08.1/0000:05:00.3/usb1/1-2/1-2:1.0/0003:054C:0268.000A/input/input36
[  410.241367] input: SHANWAN PS3 GamePad as /devices/pci0000:00/0000:00:08.1/0000:05:00.3/usb1/1-2/1-2:1.0/0003:054C:0268.000A/input/input35
[  410.241553] sony 0003:054C:0268.000A: input,hiddev97,hidraw3: USB HID v81.10 Joystick [SHANWAN PS3 GamePad] on usb-0000:05:00.3-2/input0
[  416.356591] usb 1-2: USB disconnect, device number 11
[  423.608865] sony 0005:054C:0268.000B: unknown main item tag 0x0
[  423.632182] input: SHANWAN PS3 GamePad Motion Sensors as /devices/pci0000:00/0000:00:08.1/0000:05:00.4/usb3/3-2/3-2:1.0/bluetooth/hci0/hci0:3/0005:054C:0268.000B/input/input38
[  423.632549] input: SHANWAN PS3 GamePad as /devices/pci0000:00/0000:00:08.1/0000:05:00.4/usb3/3-2/3-2:1.0/bluetooth/hci0/hci0:3/0005:054C:0268.000B/input/input37
[  423.634943] sony 0005:054C:0268.000B: input,hidraw3: BLUETOOTH HID v80.00 Joystick [SHANWAN PS3 GamePad] on 10:63:c8:b0:44:5c
```
I noticed `failed to claim input` and `unkonwn main item tag 0x0` lines.

Ahmed-E-86, 2021-02-12:
When I connect it through USB, I get 4 lights, but when I unplug it and plug it again, I get only one light, but I do not get the message that tells me if I want to trust this device or not anymore. It might be an issue with KDE.

Bluez always asks me if I want to trust this device everytime I plug it whatever I use Cinnamon, Gnome, or KDE, but that is another issue with this controller. I do not know if it is a common issue with this controller.

jangofett4, 2021-02-12:
Had a similar issue, solution was simply to install `bluez-plugins` and `bluez-utils` in Arch Linux and reboot. I removed my bluez packages in order to use the driver I built, trust dialog does open up in KDE for me.

mrfixit2001, 2021-02-12:
@jangofett4 what kernel are you using? Just to confirm the HID driver has the proper code. Can you give us output from ```uname -a```

I've tested "udev_device_get_property_value(hid_parent, "HID_NAME")" on my patch sets with my older version of bluez (will be updating shortly) and it's properly passing the name of the controller thru to the kernel now.

Unfortunately - I would also point out for anyone else reading this... that if your controller doesn't register exactly as "SHANWAN PS3 GamePad" then the kernel driver will not apply the shanwan quirks in the driver. I've seen this in some PS3 clones that are actually shanwan. I've had to patch the kernel to check if SHANWAN is anywhere in the name in order for the quirks to get picked up.

jangofett4, 2021-02-12:
Output of `uname -a`
```sh
➜  ~ uname -a
Linux cezeri 5.10.15-arch1-1 #1 SMP PREEMPT Wed, 10 Feb 2021 18:32:40 +0000 x86_64 GNU/Linux
```
It's stable `linux` package from Arch repository, no custom kernel (yet).

> Unfortunately - I would also point out for anyone else reading this... that if your controller doesn't register exactly as "SHANWAN PS3 GamePad" then the kernel driver will not apply the shanwan [...]

I think mine is identified exactly like that string, at least dmesg reports like that. Also since vibration from USB works fine I would guess that quirk is successfully applied.

Ahmed-E-86, 2021-02-12:
> Had a similar issue, solution was simply to install `bluez-plugins` and `bluez-utils` in Arch Linux and reboot. I removed my bluez packages in order to use the driver I built, trust dialog does open up in KDE for me.

I use Kubuntu 20.10, and there is nothing with these names. I only found bluez-tools, and there is nothing for plugins.

Ahmed-E-86, 2021-02-13:
> Ok I recompiled and installed again. Currently its running the driver I just built for certain. Gamepad works however all 4 lights keep flashing (every half a second or so, not pairing light). Also vibration is not working in bluetooth mode altogether.

Yeah, vibration does not work with Bluetooth now, but I like that more than vibrating for a longer time than it should be.

mrfixit2001, 2021-02-14:
Vibration with bluez alone is up to the kernel to handle properly, which now depends on what it's seeing the name of. Shanwan controllers seem to be a bit all over the place. The clone I have identifies itself as ```ShanWan     PLAYSTATION(R)3 Controller```
So the kernel would handle this like it's a normal DS3 controller and not a shanwan. When I force the kernel driver to identify this as a shanwan device, the LEDs and rumble stop working. When I allow it to function as a normal DS3, both work but the rumble effect continues for a long time until it times out completely. I'm working on a fix for that, but in the meantime I have revived and updated the qtsixa application that you can run right along with bluez that will make everything work properly. As of right now I still need to fix the way the threading works, as every rumble causes emulators to freeze up for half a second. But its "good enough" until I figure out how to make the kernel handle these clones properly. I expect I need to add a quirk to force the kernel to close the socket for the ff effect for these clones only. We will see.
https://github.com/mrfixit2001/qtsixa

jangofett4, 2021-02-14:
Thanks for the information, I tried to use qtsixa before but it wasn't working, I'll check your fork of it. Is there any way to get name from udev?  
`lsusb` reports:
```
Bus 001 Device 009: ID 054c:0268 Sony Corp. Batoh Device / PlayStation 3 Controller
```
`dmesg` reports:
```
[  152.198284] input: SHANWAN PS3 GamePad Motion Sensors as /devices/pci0000:00/0000:00:08.1/0000:05:00.3/usb1/1-2/1-2:1.0/0003:054C:0268.0004/input/input26
[  152.254670] input: SHANWAN PS3 GamePad as /devices/pci0000:00/0000:00:08.1/0000:05:00.3/usb1/1-2/1-2:1.0/0003:054C:0268.0004/input/input25
[  152.254818] sony 0003:054C:0268.0004: input,hiddev97,hidraw3: USB HID v81.10 Joystick [SHANWAN PS3 GamePad] on usb-0000:05:00.3-2/input0
```
I'm guessing `SHANWAN PS3 GamePad` is the correct one, but I might be wrong.

mrfixit2001, 2021-02-18:
FYI - I have fixed shanwan clones in my linux kernel. My fork is dedicated to rockchip compatibility but my patch can be applied to any kernel. This patch directly applies to the issue being described. I do not have any other shanwan or sixaxis devices to test with, but it fixes the issue on mine. It's possible this patch should be adjusted to apply to all shanwan devices, not just the clones. I wrote the patch so that it doesn't interfere with the current shanwan codepath and uses my new clone quirk.
https://github.com/mrfixit2001/rockchip-kernel/commit/b719b937105f0f4693d8f3993daf6b1d68c18700

sjanc, 2021-02-18:
OK, so I'm sending it to bluez mailing list for merging

Ahmed-E-86, 2021-02-18:
@mrfixit2001 Thank you so much for your efforts, but building each new kernel with the patch, it is not something that I would like to do.

@sjanc Thank you so much for your patch. I hope it will be merged soon.

mrfixit2001, 2021-02-19:
I am continuing to do testing and tweaks to this, when I’m 100% satisfied with it, I’ll submit it to torvalds to hopefully make it into mainline support.

Ahmed-E-86, 2021-02-19:
> I am continuing to do testing and tweaks to this, when I’m 100% satisfied with it, I’ll submit it to torvalds to hopefully make it into mainline support.

Amazing! I hope it will make it to mainline soon. Thanks for your support.

mrfixit2001, 2021-02-19:
Here's the squashed commit with all the fixes in it. I'll look into upstreaming it.
https://github.com/mrfixit2001/rockchip-kernel/commit/eab66f57ca831f81c61d9d3e9e144f1cfcf6e169

mrfixit2001, 2021-02-19:
https://www.spinics.net/lists/linux-input/msg71953.html

sjanc, 2021-02-22:
FYI I've applied sixaxis plugin patch to bluez master

Ahmed-E-86, 2021-02-22:
> FYI I've applied sixaxis plugin patch to bluez master

I am glad that is mainlined.

My main issue with this controller is that a lot of time it just does not work using USB or Bluetooth, and I get this errors from dmesg:
```
[ 1718.694099] usb 3-2: device descriptor read/64, error -71
[ 1718.934115] usb 3-2: device descriptor read/64, error -71
[ 1719.042171] usb usb3-port2: attempt power cycle
[ 1719.698191] usb 3-2: new full-speed USB device number 14 using xhci_hcd
[ 1719.698330] usb 3-2: Device not responding to setup address.
[ 1719.906338] usb 3-2: Device not responding to setup address.
[ 1720.114205] usb 3-2: device not accepting address 14, error -71
[ 1720.242227] usb 3-2: new full-speed USB device number 15 using xhci_hcd
[ 1720.242363] usb 3-2: Device not responding to setup address.
[ 1720.450370] usb 3-2: Device not responding to setup address.
[ 1720.658242] usb 3-2: device not accepting address 15, error -71
[ 1720.658312] usb usb3-port2: unable to enumerate USB device
```

Ahmed-E-86, 2021-02-22:
After searching, I have found a solution to this issue:

`sudo nano /sys/module/usbcore/parameters/autosuspend`

I changed it to -1

`sudo nano /sys/module/usbcore/parameters/old_scheme_first`

I changed it to Y

I am not sure which one fixed this issue.

Ahmed-E-86, 2021-02-22:
The solution to this issue is to press the reset button inside the controller when it is connected by the USB cable.

Ahmed-E-86, 2021-07-10:
> https://www.spinics.net/lists/linux-input/msg71953.html

You said nothing about your patches. Is it mainlined yet?

DocMAX, 2021-07-20:
Somehow everything is messed up now. it's nice to see the SHANWAN PS3 Gamepad name now. But they are not acting as PS3 controllers anymore. Rumble does not work and all LED blink instead of showing 1, 2, 3, or 4 single LED. What happened?

mrfixit2001, 2021-07-20:
I replied via email but it didn’t show up. So I’m posting manually, but Sorry if this eventually shows up twice. 

That could be a few things. Check your dmesg output to confirm what HID driver is being used by your kernel. Mainline Sony driver will only use Shanwan quirks if the name is EXACTLY this, case sensitive with one space between words: SHANWAN PS3 GamePad

If you have any of the sixad sixaxis plugins installed then they will override the name that bluez is assigning and force something like SONY PLAYSTATION(R)3 controller (depending on the sixad fork)

DocMAX, 2021-07-20:
Thank you for your reply. This is my output. Isn't it normal?
```
[Di Jul 20 22:12:43 2021] input: SHANWAN PS3 GamePad Motion Sensors as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/usb1/1-8/1-8.1/1-8.1.1/1-8.1.1:1.0/bluetooth/hci0/hci0:40/0005:054C:0268.002D/input/input108
[Di Jul 20 22:12:43 2021] input: SHANWAN PS3 GamePad as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/usb1/1-8/1-8.1/1-8.1.1/1-8.1.1:1.0/bluetooth/hci0/hci0:40/0005:054C:0268.002D/input/input107
[Di Jul 20 22:12:43 2021] sony 0005:054C:0268.002D: input,hidraw6: BLUETOOTH HID v80.00 Joystick [SHANWAN PS3 GamePad] on 00:09:dd:10:6f:75

```

mrfixit2001, 2021-07-20:
That looks pretty standard, yes. What kernel are you using? If it’s not mainline, Any clue where the source for it is or if the Sony HID driver has been patched at all?

You may want to try removing the device from bluetoothctl and letting bluez re-trust it over usb with the sixaxis plugin enabled.

DocMAX, 2021-07-20:
It's the current standard gentoo kernel. No modifications.
`Linux game 5.12.17-gentoo-dist #1 SMP Fri Jul 16 08:02:40 CEST 2021 x86_64 AMD Ryzen 9 5950X 16-Core Processor AuthenticAMD GNU/Linux`

mrfixit2001, 2021-07-20:
These Shanwan clones are all so interesting, different, and unpredictable. It sounds like your Shanwan isn’t a true Shanwan (which is very common), meaning in your case it needs to be treated like a normal DS3 instead of a Shanwan device. The kernel driver says the udev name bluez has assigned matches the Shanwan devices so it’s output reports are handled differently - which can cause LED and rumble to not work properly.

It’s possible that you could override bluez’s name assigned to the Mac, but I haven’t found a good way to make that persist. It used to be as simple as editing the files in /var/Bluetooth but i think it’s become more difficult in the latest versions. If you’re able to get it to change then almost ANY other name than that specific Shanwan name should resolve your issue. 

If you can’t get the name changed, you could install the sixad driver, which will intercept the comm between bluez and the kernel and handle everything for you. But you may encounter a slight delay right after vibration due to the way it’s programmed.

closed 2025-08-31 by github-actions[bot]