bluez/bluez #47: It's not possible to scan a PHY CODEC encoded broadcast device

zjdliu, 2020-10-22:
hi,

I was scanning using bluetoothctl applications and found that it's not possible to scan a PHY CODEC encoded broadcast device

scan info:
[image]

PHY CODED device info:
[image]

It can be seen that Bluetoothctl is not scanning it

Vudentz, 2020-10-22:
@zjdliu Does you controller support coded PHY though? That would explain it not being able to discover.

zjdliu, 2020-10-23:
> @zjdliu您的控制器是否支持编码PHY？那将解释它无法发现。

This is a feature supported by the controller 

[image]

sjanc, 2020-11-17:
HI,

I guess controller supports coded PHY but doesn't support Extended Advertising (so you'll be able to switch to coded once connected over legacy advertising)

closed 2025-08-31 by github-actions[bot]