bluez/bluez #48: Segfault in bluetoothd during service discovery of an emulated BLE keyboard.

hdasch, 2020-10-24:
Testing a GATT service for emulating a keyboard on one machine, bluetoothd on the host machine crashes.  Linux kernel version is 5.9.1.  BlueZ built from master (a9bf6a2a6).

Here's the traceback:
```
Program received signal SIGSEGV, Segmentation fault.
0x00005555555bbbe8 in external_service_char_cb (status=0 '\000', chars=0x555555757d50 = {...}, user_data=0x55555574b0a0) at profiles/input/hog-lib.c:570
570                     end = (next ? next->handle - 1 : primary->range.end);
(gdb) bt
#0  0x00005555555bbbe8 in external_service_char_cb (status=0 '\000', chars=0x555555757d50 = {...}, user_data=0x55555574b0a0) at profiles/input/hog-lib.c:570
#1  0x00005555555d7a3a in char_discovered_cb (status=10 '\n', ipdu=0x555555758be0 "\001\b;", iplen=5, user_data=0x555555760c90) at attrib/gatt.c:671
#2  0x00005555555d961c in attrib_callback_result (opcode=1 '\001', pdu=0x55555575ac81, length=4, user_data=0x555555757e00) at attrib/gattrib.c:260
#3  0x000055555565e3f7 in handle_rsp (chan=0x55555575a720, opcode=1 '\001', pdu=0x55555575ac81 "\b;", pdu_len=4) at src/shared/att.c:814
#4  0x000055555565e9d2 in can_read_data (io=0x555555761050, user_data=0x55555575a720) at src/shared/att.c:1011
#5  0x000055555566fbaa in watch_callback (channel=0x5555557539f0, cond=G_IO_IN, user_data=0x55555575ebe0) at src/shared/io-glib.c:157
#6  0x00007ffff7e1d914 in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
#7  0x00007ffff7e717d1 in ?? () from /usr/lib/libglib-2.0.so.0
#8  0x00007ffff7e1ce63 in g_main_loop_run () from /usr/lib/libglib-2.0.so.0
#9  0x0000555555670101 in mainloop_run () at src/shared/mainloop-glib.c:66
#10 0x00005555556706e3 in mainloop_run_with_signal (func=0x5555555e223b <signal_callback>, user_data=0x0) at src/shared/mainloop-notify.c:188
#11 0x00005555555e2828 in main (argc=1, argv=0x7fffffffe4d8) at src/main.c:959
(gdb) p/x primary
$1 = 0x0
(gdb)
```

Debugging messages from console log attached: [bt-host.log](https://github.com/bluez/bluez/files/5433696/bt-host.log)

The crash is easy enough to fix.  But it suggests a different problem.  This occurs at the end of service discovery.  At this point `primary` should not be NULL.

Further instrumentation suggests that `discover_primary` has not been called because `hog->attr` is not NULL when `bt_hog_attach` is called.  The attr so referenced has a uuid of 1812.  See hoglib.c:1583:
```
	if (!hog->attr && !hog->primary) {
		discover_primary(hog, hog->attrib, NULL, primary_cb, hog);
		return true;
	}
```

Is this a race condition?  Or is it an issue in the peripheral implementation?

I've attached the output of btmon-logger, [hci.log](https://github.com/bluez/bluez/files/5433705/hci.log),  but it seems corrupt.  Let me know if there's a better way to document this failure.

Thanks,
Hugh

Vudentz, 2020-10-27:
Looks like the we are attempting to use the primary when in fact that is not set, looks like foreach_hog_external is trying to discover external characteristics when in fact this should already be in the database, anyway we will be removing the GAttrib at some point so we will need to rewrite a good part of hog-lib.

hdasch, 2020-10-28:
I appreciate your analysis.  But if fixing this requires a rewrite "at some point", it sounds like this may take months to years rather than days to weeks to resolve.  

I've seen at least one other failure mode that I haven't been able to adequately document.  Should I put in additional effort to do so?  Or should I put my project aside as unworkable for the near future?

Is such a rewrite already considered?  Should I expect a fix in the next six months?

Thanks,
Hugh

hdasch, 2020-11-01:
> I've seen at least one other failure mode that I haven't been able to
> adequately document.  Should I put in additional effort to do so?

Just to follow up on this point - that there are other race conditions
in service discovery.  With debugging turned on I see warnings of the form:

```
bluetoothd[6075]: profiles/input/hog-lib.c:external_report_reference_cb() Read External Report Reference descriptor failed: Request attribute has encountered an unlikely error
```

It's not always read external report reference.  Sometimes the host
system is not able to read the report map.

The "unlikely error" on the host side is a reflection of the
peripheral rejecting a read request because the server is busy.  See,
https://github.com/bluez/bluez/blob/master/src/shared/gatt-server.c#L957