bluez/bluez #49: Key file does not have group "Controller"

Darapsas, 2020-11-07:
Started getting this issue after upgrading linux from `5.8.*` to `5.9.*` 

[image]

Vudentz, 2020-11-09:
This logs don't indicate any problem, the Controller logs is just warning for missing setting which is not really an error (though we should probably change to only print them when debug is enabled), can you attach the btmon logs, does the controller even shows up, from the logs it does seem it appears as hci0.

Shackelford-Arden, 2020-11-16:
Seeing a similar message on the following:

Pop! OS 20.10
Kernel: 5.8.0-7630-generic

I see that message and then you'll notice I get the message of:

```bluetoothd: Loading LTKs timed out for hci0```

btmon logs:

```
Bluetooth monitor ver 5.55
= Note: Linux version 5.8.0-7630-generic (x86_64)                                                                    0.458806
= Note: Bluetooth subsystem version 2.22                                                                             0.458810
= New Index: A4:B1:C1:0C:BB:6E (Primary,USB,hci0)                                                             [hci0] 0.458811
@ MGMT Open: bluetoothd (privileged) version 1.17                                                           {0x0001} 0.458811
= bluetoothd: Terminating                                                                                            4.332684
= bluetoothd: Endpoint unregistered: sender=:1.77 path=/MediaEndpoint/A2DPSink/sbc                                   4.332805
= bluetoothd: Endpoint unregistered: sender=:1.77 path=/MediaEndpoint/A2DPSource/sbc                                 4.332832
@ MGMT Close: bluetoothd                                                                                    {0x0001} 4.338185
= bluetoothd: Stopping SDP server                                                                                    4.338245
= bluetoothd: Exit                                                                                                   4.338260
= bluetoothd: Bluetooth daemon 5.55                                                                                  4.357779
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.357980
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.357999
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358006
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358011
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358014
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358017
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358032
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358036
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358039
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358048
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358056
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358059
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358067
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358070
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358073
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358084
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358092
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358100
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358109
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358118
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358127
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358136
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358144
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358154
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358162
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358171
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358180
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                     4.358188
@ MGMT Open: bluetoothd (privileged) version 1.17                                                           {0x0001} 4.359764
= bluetoothd: Starting SDP server                                                                                    4.359869
@ MGMT Command: Read Management Version Information (0x0001) plen 0                                         {0x0001} 4.362576
[... cut ...]
        Status: Failed (0x03)
= bluetoothd: Failed to set mode: Failed (0x03)                                                              [hci0] 24.931957
@ MGMT Command: Set IO Capability (0x0018) plen 1                                                   {0x0001} [hci0] 24.931991
        Capability: KeyboardDisplay (0x04)
@ MGMT Event: Command Complete (0x0001) plen 3                                                      {0x0001} [hci0] 24.931995
      Set IO Capability (0x0018) plen 0
        Status: Success (0x00)
```

Darapsas, 2020-11-19:
Current Linux version: `5.9.8-arch1-1`
Current Bluetooth packages:
```
bluez 5.55-1
bluez-libs 5.55-1
bluez-utils 5.55-1
pulseaudio-bluetooth 13.99.3-1
```

@Vudentz sorry I made you wait so long, here are the logs from `sudo btmon` I get when I restart the Bluetooth service (`sudo systemctl restart bluetooth`), I hope this helps in any way:

```
darapsas@core ~ % sudo btmon
Bluetooth monitor ver 5.55
= Note: Linux version 5.9.8-arch1-1 (x86_64)                                                                                                                                                                                          0.633130
= Note: Bluetooth subsystem version 2.22                                                                                                                                                                                              0.633132
= New Index: 00:E0:45:93:A6:03 (Primary,USB,hci0)                                                                                                                                                                              [hci0] 0.633133
@ MGMT Open: bluetoothd (privileged) version 1.18                                                                                                                                                                            {0x0001} 0.633134
= bluetoothd: Terminating                                                                                                                                                                                                            34.863273
@ MGMT Close: bluetoothd                                                                                                                                                                                                    {0x0001} 34.864060
= bluetoothd: Stopping SDP server                                                                                                                                                                                                    34.864128
= bluetoothd: Exit                                                                                                                                                                                                                   34.864300
= bluetoothd: Bluetooth daemon 5.55                                                                                                                                                                                                  34.874762
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                                                                                                                                     34.874904
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                                                                                                                                     34.874919
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                                                                                                                                     34.874928
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                                                                                                                                     34.874935
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                                                                                                                                     34.874942
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                                                                                                                                     34.874950
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                                                                                                                                     34.874957
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                                                                                                                                     34.874964
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                                                                                                                                     34.874971
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                                                                                                                                     34.874978
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                                                                                                                                     34.874985
= bluetoothd: src/main.c:parse_controller_config() Key file does not have group “Controller”                                                                                                                                     34.874993
[... cut ...]
@ MGMT Command: Read Controller Index List (0x0003) plen 0                                                                                                                                                                  {0x0001} 34.879520
@ MGMT Event: Command Complete (0x0001) plen 5                                                                                                                                                                              {0x0001} 34.879521
      Read Controller Index List (0x0003) plen 2
        Status: Success (0x00)
        Controllers: 0
```

Darapsas, 2020-11-19:
@Shackelford-Arden well you at least get one controller, my situation is a little bit worse - I get zero, a big fat :zero: :cry:

Darapsas, 2020-12-04:
Good evening @Vudentz,
After the latest update, everything seems to work correctly.

Current software versions:
linux: `5.9.11-arch2-1`
bluez: `5.55-1`
bluez-libs: `5.55-1`
bluez-utils: `5.55-1`
pulseaudio-bluetooth: `14.0-1`

closed 2020-12-04 by Darapsas