bluez/bluez #53: segfault connecting to specific device

i0r14, 2020-11-16:
Hello,
I have 2 bluetooth devices: a pair of true wireless earbuds and a JBL Link 300. Both devices have been paired. 
I can connect and use the earbuds without any problems.
When I try to connect to the JBL, the **bluetooth service throws a segfault**.
I kept downgrading bluez, getting packages from here: [https://archive.archlinux.org/packages/b/bluez/](url).

What I found out is that the problem started from here:
**bluez-5.51-1-x86_64.pkg.tar.xz                     22-Sep-2019 20:46    676K**

Link 300 works fine with this one:
**bluez-5.50-7-x86_64.pkg.tar.xz                     06-Sep-2019 19:32    518K**

The issue seems similar to these:
[https://bugs.archlinux.org/task/48644](url)
[https://bugs.archlinux.org/task/53442](url)

System Info:
OS: `Arch with kernel 5.9.8-arch1-1`
bluez: `5.55`
dongle: `ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)`

This is the output of systemd when connecting to the JBL:
```
$ systemctl status bluetooth
bluetoothd[6079]: profiles/audio/a2dp.c:register_remote_sep() Could not register remote sep /org/bluez/hci0/dev_XX_XX_XX_XX_XX_XX/sep7
bluetoothd[6079]: profiles/audio/a2dp.c:register_remote_sep() Could not register remote sep /org/bluez/hci0/dev_XX_XX_XX_XX_XX_XX/sep8
bluetoothd[6079]: profiles/audio/a2dp.c:register_remote_sep() Could not register remote sep /org/bluez/hci0/dev_XX_XX_XX_XX_XX_XX/sep9
bluetooth.service: Main process exited, code=dumped, status=11/SEGV
bluetooth.service: Failed with result 'core-dump'.
```

dmesg:
```
$ dmesg
[ 6874.438813] bluetoothd[6079]: segfault at 3 ip 00005574cedbf517 sp 00007ffc58571f40 error 4 in bluetoothd[5574cedba000+a8000]
[ 6874.438818] Code: 48 8d 0d d7 35 0a 00 ba 04 00 00 00 4c 89 f7 44 0f b6 c0 be 01 00 00 00 31 c0 ff 15 db 58 10 00 48 8b 7d 10 67 e8 99 77 00 00 <44> 0f b6 6b 03 48 8b 7d 10 44 0f b6 f8 67 e8 66 77 00 00 48 83 ec
```

The coredump:
```
# coredumpctl info 6079
           PID: 6079 (bluetoothd)
           UID: 0 (root)
           GID: 0 (root)
           Signal: 11 (SEGV)
           Timestamp: Sat 2020-11-14 13:45:39 CET (20h ago)
           Command Line: /usr/lib/bluetooth/bluetoothd
           Executable: /usr/lib/bluetooth/bluetoothd
           Control Group: /system.slice/bluetooth.service
           Unit: bluetooth.service
           Slice: system.slice
           Boot ID: XXX
           Machine ID: XXX
           Hostname: XX
           Storage: /var/lib/systemd/coredump/core.bluetoothd.0.2c2cc8a8928c48d3b4080aebc2d5d6ad.6079.1605357939000000.zst
           Message: Process 6079 (bluetoothd) of user 0 dumped core.
                
                Stack trace of thread 6079:
                #0  0x00005574cedbf517 n/a (bluetoothd + 0x25517)
                #1  0x00005574cee4df6d n/a (bluetoothd + 0xb3f6d)
                #2  0x00005574cedbfe3e n/a (bluetoothd + 0x25e3e)
                #3  0x00005574cedc2b13 n/a (bluetoothd + 0x28b13)
                #4  0x00005574cedc4d5e n/a (bluetoothd + 0x2ad5e)
                #5  0x00005574cedca3a9 n/a (bluetoothd + 0x303a9)
                #6  0x00007f2ed307d914 g_main_context_dispatch (libglib-2.0.so.0 + 0x52914)
                #7  0x00007f2ed30d17d1 n/a (libglib-2.0.so.0 + 0xa67d1)
                #8  0x00007f2ed307ce63 g_main_loop_run (libglib-2.0.so.0 + 0x51e63)
                #9  0x00005574cee5fa06 n/a (bluetoothd + 0xc5a06)
                #10 0x00005574cee5fe88 n/a (bluetoothd + 0xc5e88)
                #11 0x00005574cedbaac1 n/a (bluetoothd + 0x20ac1)
                #12 0x00007f2ed2d50152 __libc_start_main (libc.so.6 + 0x28152)
                #13 0x00005574cedbb77e n/a (bluetoothd + 0x2177e)
```
Let me know if some data is missing and I'll provide it.
Thanks for your work.

Vudentz, 2020-11-16:
5.50 is quite old actually, could you try upgrading to 5.55 to check if you can still reproduce?

i0r14, 2020-11-16:
Sorry, my first post wasn't clear enough.

Bluez was on 5.55 when I found out about those segfaults. 
Actually, the 3 blocks of code from my first post show what happened when the system was on bluez 5.55.
Then, I kept downgrading bluez, version by version, every time reproducing the segfault until ver. 5.50, that seems to not be affected. 

Thanks for your help.

i0r14, 2020-11-18:
These are the outputs of btmon. 
The first is with 5.50 ( device works perfectly ), the second with 5.55 ( segfault as soon as the device is connected ).
As I said, any other versions in between of 5.50 and 5.55 are affected by the same segfault. 

5.50
```
Bluetooth monitor ver 5.50
= Note: Linux version 5.9.8-arch1-1 (x86_64)                                                                                  0.427306
= Note: Bluetooth subsystem version 2.22                                                                                      0.427308
= New Index: YY:YY:YY:YY:YY:YY (Primary,USB,hci0)                                                                      [hci0] 0.427309
= Open Index: YY:YY:YY:YY:YY:YY                                                                                        [hci0] 0.427309
= Index Info: YY:YY:YY:YY:YY:YY (Cambridge Silicon Radio)                                                              [hci0] 0.427309
@ MGMT Open: bluetoothd (privileged) version 1.18                                                                    {0x0001} 0.427310
< HCI Command: Create Connection (0x01|0x0005) plen 13                                                             #1 [hci0] 12.853776
        Address: XX:XX:XX:XX:XX:XX (AzureWave Technology Inc.)
        Packet type: 0xcc18
          DM1 may be used
          DH1 may be used
          DM3 may be used
          DH3 may be used
          DM5 may be used
          DH5 may be used
        Page scan repetition mode: R2 (0x02)
        Page scan mode: Mandatory (0x00)
        Clock offset: 0x0000
        Role switch: Allow slave (0x01)
> HCI Event: Command Status (0x0f) plen 4                                                                          #2 [hci0] 12.975244
      Create Connection (0x01|0x0005) ncmd 1
        Status: Success (0x00)
> HCI Event: Role Change (0x12) plen 8                                                                             #3 [hci0] 14.018495
        Status: Success (0x00)
        Address: XX:XX:XX:XX:XX:XX (AzureWave Technology Inc.)
        Role: Slave (0x01)
> HCI Event: Connect Complete (0x03) plen 11                                                                       #4 [hci0] 14.064508
        Status: Success (0x00)
        Handle: 70
        Address: XX:XX:XX:XX:XX:XX (AzureWave Technology Inc.)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
< HCI Command: Read Remote Supported Features (0x01|0x001b) plen 2                                                 #5 [hci0] 14.064561
        Handle: 70
> HCI Event: Page Scan Repetition Mode Change (0x20) plen 7                                                        #6 [hci0] 14.065534
        Address: XX:XX:XX:XX:XX:XX (AzureWave Technology Inc.)
        Page scan repetition mode: R1 (0x01)
> HCI Event: Command Status (0x0f) plen 4                                                                          #7 [hci0] 14.069506
      Read Remote Supported Features (0x01|0x001b) ncmd 0
        Status: Success (0x00)
> HCI Event: Max Slots Change (0x1b) plen 3                                                                        #8 [hci0] 14.167529
        Handle: 70
        Max slots: 5
> HCI Event: Command Status (0x0f) plen 4                                                                          #9 [hci0] 14.169528
      NOP (0x00|0x0000) ncmd 1
        Status: Success (0x00)
> HCI Event: Read Remote Supported Features (0x0b) plen 11                                                        #10 [hci0] 14.174534
        Status: Success (0x00)
        Handle: 70
        Features: 0xff 0xfe 0x8f 0xfe 0xdb 0xff 0x7b 0x87
          3 slot packets
          5 slot packets
          Encryption
          Slot offset
          Timing accuracy
          Role switch
          Hold mode
          Sniff mode
          Power control requests
          Channel quality driven data rate (CQDDR)
          SCO link
          HV2 packets
          HV3 packets
[... 578 lines cut ...]
        Destination CID: 64
        Source CID: 64
@ MGMT Close: bluetoothd                                                                                             {0x0001} 5.849925
> ACL Data RX: Handle 71 flags 0x02 dlen 12                                                                        #64 [hci0] 5.883195
      L2CAP: Disconnection Response (0x07) ident 5 len 4
        Destination CID: 64
        Source CID: 64
> HCI Event: Number of Completed Packets (0x13) plen 5                                                             #65 [hci0] 5.883603
        Num handles: 1
        Handle: 71
        Count: 1
< HCI Command: Disconnect (0x01|0x0006) plen 3                                                                     #66 [hci0] 7.981897
        Handle: 71
        Reason: Remote User Terminated Connection (0x13)
> HCI Event: Command Status (0x0f) plen 4                                                                          #67 [hci0] 8.093870
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4                                                                     #68 [hci0] 8.122874
        Status: Success (0x00)
        Handle: 71
        Reason: Connection Terminated By Local Host (0x16)

```

I hope this could be of any help.

Vudentz, 2020-11-19:
Looks there is a problem parsing the remote SEPs, there are 9 seps and at some point it seems the registration fails which is probably what is causing the crash.

i0r14, 2020-11-19:
@Vudentz, here is the valgrind output as you asked for.

```
# valgrind --leak-check=full /usr/lib/bluetooth/bluetoothd

==2089== Memcheck, a memory error detector
==2089== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==2089== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==2089== Command: /usr/lib/bluetooth/bluetoothd
==2089== 
==2089== Invalid read of size 1
==2089==    at 0x12D517: ??? (in /usr/lib/bluetooth/bluetoothd)
==2089==    by 0x1BBF6C: ??? (in /usr/lib/bluetooth/bluetoothd)
==2089==    by 0x12DE3D: ??? (in /usr/lib/bluetooth/bluetoothd)
==2089==    by 0x130B12: ??? (in /usr/lib/bluetooth/bluetoothd)
==2089==    by 0x132D5D: ??? (in /usr/lib/bluetooth/bluetoothd)
==2089==    by 0x1383A8: ??? (in /usr/lib/bluetooth/bluetoothd)
==2089==    by 0x48C0913: g_main_context_dispatch (in /usr/lib/libglib-2.0.so.0.6600.2)
==2089==    by 0x49147D0: ??? (in /usr/lib/libglib-2.0.so.0.6600.2)
==2089==    by 0x48BFE62: g_main_loop_run (in /usr/lib/libglib-2.0.so.0.6600.2)
==2089==    by 0x1CDA05: ??? (in /usr/lib/bluetooth/bluetoothd)
==2089==    by 0x1CDE87: ??? (in /usr/lib/bluetooth/bluetoothd)
==2089==    by 0x128AC0: ??? (in /usr/lib/bluetooth/bluetoothd)
==2089==  Address 0x3 is not stack'd, malloc'd or (recently) free'd
==2089== 
==2089== 
==2089== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==2089==  Access not within mapped region at address 0x3
==2089==    at 0x12D517: ??? (in /usr/lib/bluetooth/bluetoothd)
==2089==    by 0x1BBF6C: ??? (in /usr/lib/bluetooth/bluetoothd)
==2089==    by 0x12DE3D: ??? (in /usr/lib/bluetooth/bluetoothd)
==2089==    by 0x130B12: ??? (in /usr/lib/bluetooth/bluetoothd)
==2089==    by 0x132D5D: ??? (in /usr/lib/bluetooth/bluetoothd)
==2089==    by 0x1383A8: ??? (in /usr/lib/bluetooth/bluetoothd)
==2089==    by 0x48C0913: g_main_context_dispatch (in /usr/lib/libglib-2.0.so.0.6600.2)
==2089==    by 0x49147D0: ??? (in /usr/lib/libglib-2.0.so.0.6600.2)
==2089==    by 0x48BFE62: g_main_loop_run (in /usr/lib/libglib-2.0.so.0.6600.2)
==2089==    by 0x1CDA05: ??? (in /usr/lib/bluetooth/bluetoothd)
==2089==    by 0x1CDE87: ??? (in /usr/lib/bluetooth/bluetoothd)
==2089==    by 0x128AC0: ??? (in /usr/lib/bluetooth/bluetoothd)
==2089==  If you believe this happened as a result of a stack
==2089==  overflow in your program's main thread (unlikely but
==2089==  possible), you can try to increase the size of the
==2089==  main thread stack using the --main-stacksize= flag.
==2089==  The main thread stack size used in this run was 8388608.
==2089== 
==2089== HEAP SUMMARY:
==2089==     in use at exit: 126,400 bytes in 2,324 blocks
==2089==   total heap usage: 10,456 allocs, 8,132 frees, 934,392 bytes allocated
==2089== 
==2089== LEAK SUMMARY:
==2089==    definitely lost: 0 bytes in 0 blocks
==2089==    indirectly lost: 0 bytes in 0 blocks
==2089==      possibly lost: 0 bytes in 0 blocks
==2089==    still reachable: 126,400 bytes in 2,324 blocks
==2089==         suppressed: 0 bytes in 0 blocks
==2089== Reachable blocks (those to which a pointer was found) are not shown.
==2089== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==2089== 
==2089== For lists of detected and suppressed errors, rerun with: -s
==2089== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Segmentation fault (core dumped)
```

i0r14, 2020-11-23:
Hello,
#57 seemed like the same issue, so I tried to patch sources with [4bbfda6](https://github.com/tedd-an/bluez/commit/4bbfda674dbe4fd53fd8a341a4f69dd993bb0da9).
I can confirm that this commit solves my problem too.

Thanks for your work.

closed 2020-11-23 by i0r14

VictorQueiroz, 2021-10-16:
This problem is still happening on 5.62 (Arch Linux) whenever I connect my "Phillips TAT1235":

```
Oct 16 05:07:59 VQ-L380-Yoga systemd[1]: Starting Bluetooth service...
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Bluetooth daemon 5.62
Oct 16 05:07:59 VQ-L380-Yoga systemd[1]: Started Bluetooth service.
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Starting SDP server
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Bluetooth management interface 1.18 initialized
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Failed to set privacy: Rejected (0x0b)
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSource/ldac
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSink/aptx_hd
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSource/aptx_hd
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSink/aptx
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSource/aptx
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSource/aac
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSink/sbc
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSource/sbc
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSink/sbc_xq
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSource/sbc_xq
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSource/faststream
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSource/faststream_duplex
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSource/aptx_ll_1
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSource/aptx_ll_0
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
Oct 16 05:07:59 VQ-L380-Yoga bluetoothd[5516]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Oct 16 05:08:07 VQ-L380-Yoga bluetoothd[5516]: src/device.c:search_cb() 6A:56:3A:1F:D8:36: error updating services: Host is down (112)
Oct 16 05:08:38 VQ-L380-Yoga bluetoothd[5516]: double free or corruption (out)
Oct 16 05:08:39 VQ-L380-Yoga systemd[1]: bluetooth.service: Main process exited, code=dumped, status=6/ABRT
Oct 16 05:08:39 VQ-L380-Yoga systemd[1]: bluetooth.service: Failed with result 'core-dump'.
```