bluez/bluez #55: Device characteristics are trying to be read even though device isn't paired - likely causing pairing procedure fail

limpkin, 2020-11-18:
Hello,

We're developing a BLE device whose characteristics values can only be read when the device is paired to the host (https://github.com/mooltipass/minible/blob/master/source_code/aux_mcu/src/LOGIC/logic_bluetooth.c#L905).  

However, during pairing I get the following errors, likely causing a pairing fail:
>Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/deviceinfo/dis.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:report_map_read_cb() Report Map read failed: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:report_map_read_cb() Report Map read failed: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:proto_mode_read_cb() Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error

What I'm seeing using a Bluetooth RF analyzer is that the host never tries to query the device for the end of PIN entering.... therefore leading to a pairing fail (see the warning below).

[image]

At the end of said pairing, it however tries to pair again but much later, which fails due to a time-out on our device.
The captures can be seen here: https://github.com/mooltipass/minible/issues/205 and more can be taken if you'd like.

bluetoothctl --version gives 
> bluetoothctl: 5.55
uname -r gives 
> 5.9.8-arch1-1

limpkin, 2020-11-19:
same issue confirmed on logitec keyboard craft, on bluez 5.55 compiled from source on ubuntu.
I'll try on another machine, in case it comes from the HW

limpkin, 2020-11-22:
So it doesn't seem like the pairing issue is related to the impossibility to read characteristics. 
This issue was initially reported for a Linux OS running in a VM. When trying natively on a machine, the first pairing attempt does fail while the second goes through (somehow...).
The bluetooth dongle remains the same for both configurations..... odd.

Vudentz, 2021-01-04:
@limpkin Can you collect the HCI trace using btmon? It seems very weird that it fails to pair sometimes and works in others, maybe the pairing agent is not responding in time and it timeouts for some reason.

gearhead, 2021-02-07:
Just read on the mail list that a log is needed. Mine is attached
need to repeat the pair attempt many times until it pairs then I can trust and it works, the pairing is the painful part. 
[failed_keyboard_pair.txt](https://github.com/bluez/bluez/files/5940060/failed_keyboard_pair.txt)

limpkin, 2021-02-07:
@gearhead thanks a lot for this... sorry we never got to.

gearhead, 2021-02-23:
Just compiled and installed 5.56 on arch and get the same result. many many attempts to pair and maybe if I am lucky it pairs. can I manually pair it (editing text files) until this gets resolved. 
```
# pacman -Q | grep bluez
bluez 5.56-1
bluez-libs 5.56-1
bluez-tools 0.2.0-5
bluez-utils 5.56-1
```

Vudentz, 2021-02-24:
> Just read on the mail list that a log is needed. Mine is attached
> need to repeat the pair attempt many times until it pairs then I can trust and it works, the pairing is the painful part.
> [failed_keyboard_pair.txt](https://github.com/bluez/bluez/files/5940060/failed_keyboard_pair.txt)

This doesn't seems to be related to the original problem since this is actually paring over BR/EDR:

```
@ MGMT Event: Device Connected (0x000b) plen 47                                                                                                                          {0x0001} [hci0] 23.677411
        **BR/EDR** Address: 08:B7:38:11:A6:C2 (Lite-On Technogy Corp.)
        Flags: 0x00000000
        Data length: 34
        Name (complete): HP Bluetooth Keyboard K4000
        Class: 0x002540
          Major class: Peripheral (mouse, joystick, keyboards)
          Minor class: 0x10
          Limited Discoverable Mode
```

Or is the peripheral dual-mode or needs GATT over BR/EDR? Anyway for BR/EDR the pairing is actually handled by the link layer which is down in the controller firmware, perhaps you might want to try another controller to check if this is hardware specific error.

gearhead, 2021-02-24:
If I try pairing it to another controller, I loose this connection the need to spend another hour trying until it pairs? Seriously, this used to be easy. bluetoothctl, power on keyboard, put in pairing mode, default-agent, scan on, pair XXXXXXXX, trust XXXXXX done. What could have changed? Once paired, The /var/lib/bluetooth/xxx/xxx/info shows:
```
[General]
Name=HP Bluetooth Keyboard K4000
Class=0x000540
SupportedTechnologies=BR/EDR;
Trusted=true
Blocked=false
WakeAllowed=true
Services=00001000-0000-1000-8000-00805f9b34fb;00001124-0000-1000-8000-00805f>

[LinkKey]
Key=143AC49B423B2E0584298707A185D1C9
Type=5
PINLength=0

[DeviceID]
Source=2
Vendor=1226
Product=121
Version=1281
```

gearhead, 2021-02-27:
just tried it on an intel BT adapter on a computer running Arch Linux  and have the same result. Repeated failure to pair. It asks for the 6 digit code, I type it in and it sits a while then finally says failed to pair. over and over again.
```
[bluetooth]# pair 08:B7:38:11:A6:C2
Attempting to pair with 08:B7:38:11:A6:C2
[CHG] Device 08:B7:38:11:A6:C2 Connected: yes
[agent] Passkey: 614594
[agent] Passkey: 614594
[agent] Passkey: 614594
[agent] Passkey: 614594
[agent] Passkey: 614594
[agent] Passkey: 614594
[agent] Passkey: 614594
[HP Bluetooth Keyboard K4000]#
Failed to pair: org.bluez.Error.AuthenticationFailed
[CHG] Device 08:B7:38:11:A6:C2 Connected: no
```

gearhead, 2021-03-04:
Should I open a new issue on this? It takes repeated attempts then eventually it pairs. As you can see, it will connect to allow me to type the code in, but then it waits a while end eventually says failed to pair. try again and again and again and finally success then I can trust and it works every time. It used to be one time and done.

gearhead, 2021-03-13:
please?

limpkin, 2021-04-13:
so it does look this issue was fixed in one of these 3 commits as I was able to pair today, building from source:
https://github.com/bluez/bluez/commit/d971ca5ca318d8a48d4cfedfd36ff6a2641a12e2
https://github.com/bluez/bluez/commit/8f15296d8b3e6f2a42e7bc92fc2c9e6b218f360e
https://github.com/bluez/bluez/commit/b5b7e6203b9303b20751f4ef0ad2d6a01a671ae9

gearhead, 2021-04-15:
wish i could build it to test. I get this:
```
...
mesh-cfgtest.c:(.text+0x93c): undefined reference to `l_tester_get_stage'
/usr/bin/ld: tools/mesh-cfgtest.o:mesh-cfgtest.c:(.text+0x944): more undefined references to `l_tester_get_stage' follow
/usr/bin/ld: tools/mesh-cfgtest.o: in function `main':
mesh-cfgtest.c:(.text.startup+0x208): undefined reference to `l_tester_add_full'
/usr/bin/ld: mesh-cfgtest.c:(.text.startup+0x298): undefined reference to `l_tester_add_full'
collect2: error: ld returned 1 exit status
distcc[307515] ERROR: compile (null) on localhost failed
make[1]: *** [Makefile:5888: tools/mesh-cfgtest] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:4062: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```

gearhead, 2021-04-19:
Just tried this and still get the same result. I built and installed ell-git and bluez-git on my target device (RPi b3+) and tried, again, to pair my keyboard and get:
```
[bluetooth]# pair 08:B7:38:11:A6:C2
Attempting to pair with 08:B7:38:11:A6:C2
[CHG] Device 08:B7:38:11:A6:C2 Connected: yes
[agent] Passkey: 457425
[agent] Passkey: 457425
[agent] Passkey: 457425
[agent] Passkey: 457425
[agent] Passkey: 457425
[agent] Passkey: 457425
[agent] Passkey: 457425
Failed to pair: org.bluez.Error.AuthenticationFailed
[CHG] Device 08:B7:38:11:A6:C2 Connected: no
```
Multiple attempts result in the same thing. I can type in the pass code and it responds when I do it, but it never pairs.

gearhead, 2021-05-19:
Still does not want to pair. Takes repeated attempts until it *finally* connects. This is on an RPi 3b+ running aarch64 kernel 5.10.36 and bluez 5.58 built from git with HEAD at 6a9229047. If another scan is needed for information to fix this, please let me know. Very disappointing that I can connect a media device but not a simple keyboard.

macmpi, 2021-05-30:
have you tried if setting `JustWorksRepairing=always` in `/etc/bluetooth/main.conf`  is of any help?

gearhead, 2021-05-31:
I tried the `JustWorksRepairing=always` and get the same result. It will not pair a keyboard. After I type in the code, it waits a bit then responds with the Auth Failed. This is with current broadcom firmware and bluez 5.51 or 5.58.

With current firmware and bluez 5.58, I can pair and connect and play from a phone and to a set of bluetooth headphones. I was able to build and install bluez 5.51 (5.50 would not build due to an ell function issue) and have the same result. I can pair and connect headphones and a phone, but not a keyboard. If I try many many times I can eventually get it to accept the code and pair. Please advise if there is anything else I can test.

gearhead, 2021-06-16:
Just tried with 5.59 which recently landed on Arch Arm. Same thing. The btmon log is attached. 
```
[bluetooth]# pair  A4:9A:58:6C:BE:7A
Attempting to pair with A4:9A:58:6C:BE:7A
[CHG] Device A4:9A:58:6C:BE:7A Connected: yes
[agent] Passkey: 513160
[agent] Passkey: 513160
[agent] Passkey: 513160
[agent] Passkey: 513160
[agent] Passkey: 513160
[agent] Passkey: 513160
[agent] Passkey: 513160
[DEL] Device 08:62:66:E1:82:18 Nexus Player
[CHG] Controller 43:45:C0:00:1F:AC Discoverable: no
Failed to pair: org.bluez.Error.AuthenticationFailed
[CHG] Device A4:9A:58:6C:BE:7A Connected: no
```
[btmon_.5.59.log](https://github.com/bluez/bluez/files/6663224/btmon_.5.59.log)

Vudentz, 2021-06-23:
```
> HCI Event: User Passkey Notification (0x3b) plen 10       #24 [hci0] 7.453183
        Address: A4:9A:58:6C:BE:7A (Samsung Electronics Co.,Ltd)
        Passkey: 513160
@ MGMT Event: Passkey Notify (0x0017) plen 12          {0x0001} [hci0] 7.453238
        BR/EDR Address: A4:9A:58:6C:BE:7A (Samsung Electronics Co.,Ltd)
        Passkey: 0x0007d488
        Entered: 0
> HCI Event: Keypress Notification (0x3c) plen 7            #25 [hci0] 7.456309
        Address: A4:9A:58:6C:BE:7A (Samsung Electronics Co.,Ltd)
        Notification type: Passkey entry started (0x00)
> HCI Event: Keypress Notification (0x3c) plen 7           #26 [hci0] 11.727688
        Address: A4:9A:58:6C:BE:7A (Samsung Electronics Co.,Ltd)
        Notification type: Passkey digit entered (0x01)
@ MGMT Event: Passkey Notify (0x0017) plen 12         {0x0001} [hci0] 11.727743
        BR/EDR Address: A4:9A:58:6C:BE:7A (Samsung Electronics Co.,Ltd)
        Passkey: 0x0007d488
        Entered: 1
> HCI Event: Keypress Notification (0x3c) plen 7           #27 [hci0] 11.960225
        Address: A4:9A:58:6C:BE:7A (Samsung Electronics Co.,Ltd)
        Notification type: Passkey digit entered (0x01)
@ MGMT Event: Passkey Notify (0x0017) plen 12         {0x0001} [hci0] 11.960289
        BR/EDR Address: A4:9A:58:6C:BE:7A (Samsung Electronics Co.,Ltd)
        Passkey: 0x0007d488
        Entered: 2
> HCI Event: Keypress Notification (0x3c) plen 7           #28 [hci0] 12.242704
        Address: A4:9A:58:6C:BE:7A (Samsung Electronics Co.,Ltd)
        Notification type: Passkey digit entered (0x01)
@ MGMT Event: Passkey Notify (0x0017) plen 12         {0x0001} [hci0] 12.242755
        BR/EDR Address: A4:9A:58:6C:BE:7A (Samsung Electronics Co.,Ltd)
        Passkey: 0x0007d488
        Entered: 3
> HCI Event: Keypress Notification (0x3c) plen 7           #29 [hci0] 12.510204
        Address: A4:9A:58:6C:BE:7A (Samsung Electronics Co.,Ltd)
        Notification type: Passkey digit entered (0x01)
@ MGMT Event: Passkey Notify (0x0017) plen 12         {0x0001} [hci0] 12.510262
        BR/EDR Address: A4:9A:58:6C:BE:7A (Samsung Electronics Co.,Ltd)
        Passkey: 0x0007d488
        Entered: 4
> HCI Event: Keypress Notification (0x3c) plen 7           #30 [hci0] 13.307712
        Address: A4:9A:58:6C:BE:7A (Samsung Electronics Co.,Ltd)
        Notification type: Passkey digit entered (0x01)
@ MGMT Event: Passkey Notify (0x0017) plen 12         {0x0001} [hci0] 13.307772
        BR/EDR Address: A4:9A:58:6C:BE:7A (Samsung Electronics Co.,Ltd)
        Passkey: 0x0007d488
        Entered: 5
> HCI Event: Keypress Notification (0x3c) plen 7           #31 [hci0] 13.490239
        Address: A4:9A:58:6C:BE:7A (Samsung Electronics Co.,Ltd)
        Notification type: Passkey digit entered (0x01)
@ MGMT Event: Passkey Notify (0x0017) plen 12         {0x0001} [hci0] 13.490335
        BR/EDR Address: A4:9A:58:6C:BE:7A (Samsung Electronics Co.,Ltd)
        Passkey: 0x0007d488
        Entered: 6
```

It looks like there are 7 keys entered, not sure if the last one was enter key but it does seem to cause another key press, not sure if that is actually causing the problem or not.

Vudentz, 2021-06-23:
@gearhead This is not even the same problem as the original one is a LE(HoG) and this is using classic HID, please create another issue, also paring with classic devices is down to the link layer which is part of the controller firmware, so there is probably nothing we can in the host.

closed 2021-06-23 by Vudentz

gearhead, 2021-06-25:
I finally figured out what is going on here. When pairing it asks for the 6 digits but does not ask for a Return. I was not typing a 'Return' so it waited then timed out before checking that I had typed the 6 digits correctly. When I type a return (My Windows machine prompted me for that but Linux does not) after the 6 digits, it pairs immediately.

Vudentz, 2021-06-25:
> I finally figured out what is going on here. When pairing it asks for the 6 digits but does not ask for a Return. I was not typing a 'Return' so it waited then timed out before checking that I had [...]

Interesting, so the windows does track down how many keys has been entered and then wroks the user to push enter? Perhaps this would be a good idea to be added to bluetoothctl, gnome, etc.