bluez/bluez #57: bluetoothd segfaults after connecting to loudspeaker

fikovnik, 2020-11-20:
I have a raspberry pi 3 setup as a loudspeaker:
```
[bluetoothctl]# info 00:1A:7D:DA:71:0F
Device 00:1A:7D:DA:71:0F (public)
Name: radio1
Alias: radio1
Class: 0x00040414
Icon: audio-card
Paired: yes
Trusted: no
Blocked: no
Connected: no
LegacyPairing: no
UUID: Audio Sink (0000110b-0000-1000-8000-00805f9b34fb)
UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
UUID: A/V Remote Control (0000110e-0000-1000-8000-00805f9b34fb)
UUID: PnP Information (00001200-0000-1000-8000-00805f9b34fb)
Modalias: usb:v1D6Bp0246d0532
RSSI: -69
```

I can connect to it using my android phone and a window box.

When I try on arch (5.9.4-arch1-1, bluez 5.55-1, pulseaudio 13.99.3-1, pulseaudio-modules-bt 1.4-3), the bluetoothd crashes:

```
Nov 06 13:25:03 kathmandu pulseaudio[1934]: Found duplicated D-Bus path for sep endpoint /org/bluez/hci0/dev_00_1A_7D_DA_71_0F/sep1
Nov 06 13:25:03 kathmandu pulseaudio[1934]: Found duplicated D-Bus path for sep endpoint /org/bluez/hci0/dev_00_1A_7D_DA_71_0F/sep2
Nov 06 13:25:03 kathmandu pulseaudio[1934]: Found duplicated D-Bus path for sep endpoint /org/bluez/hci0/dev_00_1A_7D_DA_71_0F/sep3
Nov 06 13:25:03 kathmandu bluetoothd[11096]: profiles/audio/a2dp.c:register_remote_sep() Could not register remote sep /org/bluez/hci0/dev_00_1A_7D_DA_71_0F/sep2
Nov 06 13:25:03 kathmandu audit[11096]: ANOM_ABEND auid=4294967295 uid=0 gid=0 ses=4294967295 pid=11096 comm="bluetoothd" exe="/usr/lib/bluetooth/bluetoothd" sig=11 res=1
Nov 06 13:25:03 kathmandu bluetoothd[11096]: profiles/audio/a2dp.c:register_remote_sep() Could not register remote sep /org/bluez/hci0/dev_00_1A_7D_DA_71_0F/sep3
Nov 06 13:25:03 kathmandu kernel: bluetoothd[11096]: segfault at 3 ip 00005602dce85517 sp 00007fffefc38050 error 4 in bluetoothd[5602dce80000+a8000]
Nov 06 13:25:03 kathmandu kernel: Code: 48 8d 0d d7 35 0a 00 ba 04 00 00 00 4c 89 f7 44 0f b6 c0 be 01 00 00 00 31 c0 ff 15 db 58 10 00 48 8b 7d 10 67 e8 99 77 00 00 <44> 0f b6 6b 03 48 8b 7d 10 44 0f b6 f8 67 e8 66 77 00 00 48 83 ec
Nov 06 13:25:03 kathmandu kernel: audit: type=1701 audit(1604665503.844:244): auid=4294967295 uid=0 gid=0 ses=4294967295 pid=11096 comm="bluetoothd" exe="/usr/lib/bluetooth/bluetoothd" sig=11 res=1
Nov 06 13:25:03 kathmandu audit: BPF prog-id=42 op=LOAD
Nov 06 13:25:03 kathmandu audit: BPF prog-id=43 op=LOAD
Nov 06 13:25:03 kathmandu kernel: audit: type=1334 audit(1604665503.891:245): prog-id=42 op=LOAD
Nov 06 13:25:03 kathmandu kernel: audit: type=1334 audit(1604665503.891:246): prog-id=43 op=LOAD
Nov 06 13:25:03 kathmandu systemd[1]: Started Process Core Dump (PID 11445/UID 0).
Nov 06 13:25:03 kathmandu audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@6-11445-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Nov 06 13:25:03 kathmandu kernel: audit: type=1130 audit(1604665503.894:247): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@6-11445-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Nov 06 13:25:04 kathmandu systemd[1]: bluetooth.service: Main process exited, code=dumped, status=11/SEGV
Nov 06 13:25:04 kathmandu systemd[1]: bluetooth.service: Failed with result 'core-dump'.
Nov 06 13:25:04 kathmandu audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=bluetooth comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
Nov 06 13:25:04 kathmandu kernel: audit: type=1131 audit(1604665504.181:248): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=bluetooth comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
Nov 06 13:25:04 kathmandu systemd-coredump[11446]: [?] Process 11096 (bluetoothd) of user 0 dumped core.

Stack trace of thread 2526285:
#0 0x000055ac5132c517 store_remote_sep (bluetoothd + 0x25517)
#1 0x000055ac513baf6d queue_foreach (bluetoothd + 0xb3f6d)
#2 0x000055ac5132ce3e store_remote_seps (bluetoothd + 0x25e3e)
#3 0x000055ac5132fb13 discover_cb (bluetoothd + 0x28b13)
#4 0x000055ac51331d5e finalize_discovery (bluetoothd + 0x2ad5e)
#5 0x000055ac513373a9 avdtp_parse_resp (bluetoothd + 0x303a9)
#6 0x00007f55bd639914 g_main_context_dispatch (libglib-2.0.so.0 + 0x52914)
#7 0x00007f55bd68d7d1 n/a (libglib-2.0.so.0 + 0xa67d1)
#8 0x00007f55bd638e63 g_main_loop_run (libglib-2.0.so.0 + 0x51e63)
#9 0x000055ac513cca06 mainloop_run (bluetoothd + 0xc5a06)
#10 0x000055ac513cce88 mainloop_run_with_signal (bluetoothd + 0xc5e88)
#11 0x000055ac51327ac1 main (bluetoothd + 0x20ac1)
#12 0x00007f55bd30c152 __libc_start_main (libc.so.6 + 0x28152)
#13 0x000055ac5132877e _start (bluetoothd + 0x2177e)Nov 06 13:25:04 kathmandu systemd[1]: systemd-coredump%406-11445-0.service: Succeeded.
Nov 06 13:25:04 kathmandu kernel: audit: type=1131 audit(1604665504.201:249): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@6-11445-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Nov 06 13:25:04 kathmandu audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@6-11445-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Nov 06 13:25:04 kathmandu audit: BPF prog-id=43 op=UNLOAD
Nov 06 13:25:04 kathmandu audit: BPF prog-id=42 op=UNLOAD
```

The segfault happens in `store_remote_sep` (`profiles/audio/a2dp.c`) after dereferencing NULL pointer:

```
2657 offset = sprintf(value, "%02hhx:%02hhx:%02hhx:",
2658   avdtp_get_type(sep->sep), codec->media_codec_type,
2659   avdtp_get_delay_reporting(sep->sep));
```
The `codec` is `NULL` When I add a simple null check and return early from the function, it works. I guess the problem is somewhere else, but I do not understand the bluez stack.

Connecting other loudspeaker (Bose mini 2 soundlink) works fine.
One last thing, if I install and start ofono service, bluetoothd does not crash, but after connecting, it immediately disconnects.

Vudentz, 2020-11-22:
Can you attach the output from btmon when connecting, it looks like the service capability might be 0 length when we discover which would explain why that would be NULL, but it is very wierd that it would even expose a SEP if there is no codec type but perhaps it depend on HFP for some reason, anyway we should fix even if the device is misbehaving there should cause a crash.

mortbauer, 2020-11-23:
It seems I'm having the same issue (also using a raspberry setup to act as bluetooth sound sink via bluealsa), below is the output of btmon when trying to connect:
```
Bluetooth monitor ver 5.55
= Note: Linux version 5.9.8-arch1-1 (x86_64)                           0.267552
= Note: Bluetooth subsystem version 2.22                               0.267554
= New Index: E0:06:E6:B6:89:46 (Primary,USB,hci0)               [hci0] 0.267556
= Open Index: E0:06:E6:B6:89:46                                 [hci0] 0.267556
= Index Info: E0:06:E6:B6:89:46 (Broadcom Corporation)          [hci0] 0.267557
@ MGMT Open: bluetoothd (privileged) version 1.18             {0x0001} 0.267559
< HCI Command: Create Connection (0x01|0x0005) plen 13       #1 [hci0] 2.787873
        Address: B8:27:EB:75:97:AA (Raspberry Pi Foundation)
        Packet type: 0xcc18
          DM1 may be used
          DH1 may be used
          DM3 may be used
          DH3 may be used
          DM5 may be used
          DH5 may be used
        Page scan repetition mode: R2 (0x02)
        Page scan mode: Mandatory (0x00)
        Clock offset: 0x0000
        Role switch: Allow slave (0x01)
> HCI Event: Command Status (0x0f) plen 4                    #2 [hci0] 2.916606
      Create Connection (0x01|0x0005) ncmd 1
        Status: Success (0x00)
> ACL Data RX: Handle 12 flags 0x02 dlen 10                  #3 [hci0] 6.345421
      L2CAP: Information Request (0x0a) ident 1 len 2
        Type: Extended features supported (0x0002)
> HCI Event: Role Change (0x12) plen 8                       #4 [hci0] 6.346422
        Status: Success (0x00)
        Address: B8:27:EB:75:97:AA (Raspberry Pi Foundation)
        Role: Slave (0x01)
> HCI Event: Connect Complete (0x03) plen 11                 #5 [hci0] 6.347419
        Status: Success (0x00)
        Handle: 12
        Address: B8:27:EB:75:97:AA (Raspberry Pi Foundation)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
< HCI Command: Read Remote Supported.. (0x01|0x001b) plen 2  #6 [hci0] 6.347535
        Handle: 12
> HCI Event: Max Slots Change (0x1b) plen 3                  #7 [hci0] 6.348417
        Handle: 12
        Max slots: 5
> HCI Event: Command Status (0x0f) plen 4                    #8 [hci0] 6.349410
      Read Remote Supported Features (0x01|0x001b) ncmd 1
        Status: Success (0x00)
> HCI Event: Read Remote Supported Features (0x0b) plen 11   #9 [hci0] 6.391448
        Status: Success (0x00)
        Handle: 12
        Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
          3 slot packets
          5 slot packets
          Encryption
          Slot offset
          Timing accuracy
          Role switch
          Sniff mode
          Power control requests
          Channel quality driven data rate (CQDDR)
          SCO link
          HV2 packets
          HV3 packets
          u-law log synchronous data
          A-law log synchronous data
          CVSD synchronous data
          Paging parameter negotiation
          Power control
          Transparent synchronous data
          Flow control lag (most significant bit)
          Broadcast Encryption
[... 270 lines cut ...]
        Count: 1
< ACL Data TX: Handle 12 flags 0x00 dlen 12                 #52 [hci0] 7.125814
      L2CAP: Disconnection Request (0x06) ident 5 len 4
        Destination CID: 64
        Source CID: 64
@ MGMT Close: bluetoothd                                      {0x0001} 7.125899
> ACL Data RX: Handle 12 flags 0x02 dlen 12                 #53 [hci0] 7.156783
      L2CAP: Disconnection Response (0x07) ident 5 len 4
        Destination CID: 64
        Source CID: 64
> HCI Event: Number of Completed Packets (0x13) plen 5      #54 [hci0] 7.353459
        Num handles: 1
        Handle: 12
        Count: 1
< HCI Command: Disconnect (0x01|0x0006) plen 3              #55 [hci0] 9.221308
        Handle: 12
        Reason: Remote User Terminated Connection (0x13)
> HCI Event: Command Status (0x0f) plen 4                   #56 [hci0] 9.222511
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4              #57 [hci0] 9.316452
        Status: Success (0x00)
        Handle: 12
        Reason: Connection Terminated By Local Host (0x16)
```

Vudentz, 2020-11-23:
Ive just send 2 patches to the mailing list, Im still not able to reproduce it but it does look like the SEID 2 has not been properly loaded/initiliazed which causes sep->codec to be NULL.

fikovnik, 2020-11-23:
Not sure if it will help, but here is what I did to configure the speaker on raspberry pi:
```
linux: 5.4.72-v7+
bluealsa: 0.13
bluez: 5.50-1.2~deb10u1+rpt2
```

- `/etc/bluetooth/main.conf` (showing non commented lines)
```
[General]
Class = 0x240414
DiscoverableTimeout = 0

[Policy]
AutoEnable=true
```

- `/etc/systemd/system/aplay.service`
```
[Unit]
Description=BlueALSA aplay service
After=bluetooth.service
Requires=bluetooth.service

[Service]
ExecStart=/usr/bin/bluealsa-aplay  --profile-a2dp -vv 00:00:00:00:00:00

[Install]
WantedBy=multi-user.target
```

- `/lib/systemd/system/bluealsa.service`
```
#+begin_src text
[Unit]
Description=BluezALSA proxy
Requires=bluetooth.service
After=bluetooth.service

[Service]
Type=simple
User=root
ExecStart=/usr/bin/bluealsa -p a2dp-sink

```

- `/lib/systemd/system/bluetooth.service`
```
[Unit]
Description=Bluetooth service
Documentation=man:bluetoothd(8)
ConditionPathIsDirectory=/sys/class/bluetooth

[Service]
Type=dbus
BusName=org.bluez
ExecStart=/usr/lib/bluetooth/bluetoothd --noplugin=sap
NotifyAccess=main
#WatchdogSec=10
#Restart=on-failure
CapabilityBoundingSet=CAP_NET_ADMIN CAP_NET_BIND_SERVICE
LimitNPROC=1
ProtectHome=true
ProtectSystem=full

[Install]
WantedBy=bluetooth.target
Alias=dbus-org.bluez.service
```

closed 2021-01-04 by Vudentz