bluez/bluez #60: bluetoothd crashes when change a2dp codec (profile) in pulseaudio

kapiwko, 2020-11-29:
I've started bluetoothd:

```
==202219== Memcheck, a memory error detector
==202219== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==202219== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==202219== Command: ./src/bluetoothd -n -d -f ./src/main.conf
==202219== 
bluetoothd[202219]: Bluetooth daemon 5.55
bluetoothd[202219]: src/main.c:parse_config() parsing ./src/main.conf
bluetoothd[202219]: src/main.c:parse_config() Key file does not have key “DiscoverableTimeout” in group “General”
bluetoothd[202219]: src/main.c:parse_config() Key file does not have key “AlwaysPairable” in group “General”
bluetoothd[202219]: src/main.c:parse_config() Key file does not have key “PairableTimeout” in group “General”
bluetoothd[202219]: src/main.c:parse_config() Key file does not have key “Privacy” in group “General”
bluetoothd[202219]: src/main.c:parse_config() Key file does not have key “JustWorksRepairing” in group “General”
bluetoothd[202219]: src/main.c:parse_config() Key file does not have key “TemporaryTimeout” in group “General”
bluetoothd[202219]: src/main.c:parse_config() Key file does not have key “Name” in group “General”
bluetoothd[202219]: src/main.c:parse_config() Key file does not have key “Class” in group “General”
bluetoothd[202219]: src/main.c:parse_config() Key file does not have key “DeviceID” in group “General”
bluetoothd[202219]: src/main.c:parse_config() Key file does not have key “ReverseServiceDiscovery” in group “General”
bluetoothd[202219]: src/main.c:parse_config() Key file does not have key “Cache” in group “GATT”
bluetoothd[202219]: src/main.c:parse_config() Key file does not have key “KeySize” in group “GATT”
bluetoothd[202219]: src/main.c:parse_config() Key file does not have key “ExchangeMTU” in group “GATT”
bluetoothd[202219]: src/main.c:parse_config() Key file does not have key “Channels” in group “GATT”
bluetoothd[202219]: src/main.c:parse_config() Key file does not have key “SessionMode” in group “AVDTP”
bluetoothd[202219]: src/main.c:parse_config() Key file does not have key “StreamMode” in group “AVDTP”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “MinAdvertisementInterval” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “MaxAdvertisementInterval” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “MultiAdvertisementRotationInterval” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “ScanIntervalAutoConnect” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “ScanWindowAutoConnect” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “ScanIntervalSuspend” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “ScanWindowSuspend” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “ScanIntervalDiscovery” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “ScanWindowDiscovery” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “ScanIntervalAdvMonitor” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “ScanWindowAdvMonitor” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “ScanIntervalConnect” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “ScanWindowConnect” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “MinConnectionInterval” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “MaxConnectionInterval” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “ConnectionLatency” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “ConnectionSupervisionTimeout” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “Autoconnecttimeout” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “AdvMonAllowlistScanDuration” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “AdvMonNoFilterScanDuration” in group “LE”
bluetoothd[202219]: src/main.c:parse_mode_config() Key file does not have key “EnableAdvMonInterleaveScan” in group “LE”
bluetoothd[202219]: src/adapter.c:adapter_init() sending read version command
bluetoothd[202219]: Starting SDP server
bluetoothd[202219]: src/sdpd-service.c:register_device_id() Adding device id record for 0002:1d6b:0246:0537
bluetoothd[202219]: src/plugin.c:plugin_init() Loading builtin plugins
bluetoothd[202219]: src/plugin.c:add_plugin() Loading hostname plugin
bluetoothd[202219]: src/plugin.c:add_plugin() Loading wiimote plugin
bluetoothd[202219]: src/plugin.c:add_plugin() Loading autopair plugin
bluetoothd[202219]: src/plugin.c:add_plugin() Loading policy plugin
bluetoothd[202219]: src/plugin.c:add_plugin() Loading neard plugin
bluetoothd[202219]: src/plugin.c:add_plugin() Loading sap plugin
bluetoothd[202219]: src/plugin.c:add_plugin() Loading a2dp plugin
bluetoothd[202219]: src/plugin.c:add_plugin() Loading avrcp plugin
bluetoothd[202219]: src/plugin.c:add_plugin() Loading network plugin
[... 2405 lines cut ...]
bluetoothd[202219]: profiles/network/connection.c:connection_unregister() /org/bluez/hci0/dev_E4_0E_EE_7B_CB_62 id 4374
bluetoothd[202219]: src/service.c:btd_service_unref() 0x5ec8990: ref=1
bluetoothd[202219]: profiles/network/connection.c:path_unregister() Unregistered interface org.bluez.Network1 on path /org/bluez/hci0/dev_E4_0E_EE_7B_CB_62
bluetoothd[202219]: src/service.c:btd_service_unref() 0x5ec8990: ref=0
bluetoothd[202219]: src/service.c:change_state() 0x5ed1f00: device E4:0E:EE:7B:CB:62 profile audio-avrcp-target state changed: disconnected -> unavailable (0)
bluetoothd[202219]: profiles/audio/control.c:path_unregister() Unregistered interface org.bluez.MediaControl1 on path /org/bluez/hci0/dev_E4_0E_EE_7B_CB_62
bluetoothd[202219]: src/service.c:btd_service_unref() 0x5ed1f00: ref=1
bluetoothd[202219]: src/service.c:btd_service_unref() 0x5ed1f00: ref=0
bluetoothd[202219]: src/service.c:change_state() 0x5edbbe0: device E4:0E:EE:7B:CB:62 profile a2dp-source state changed: disconnected -> unavailable (0)
bluetoothd[202219]: profiles/audio/source.c:source_unregister() /org/bluez/hci0/dev_E4_0E_EE_7B_CB_62
bluetoothd[202219]: src/service.c:btd_service_unref() 0x5edbbe0: ref=1
bluetoothd[202219]: src/service.c:btd_service_unref() 0x5edbbe0: ref=0
bluetoothd[202219]: src/device.c:btd_device_unref() Freeing device /org/bluez/hci0/dev_E4_0E_EE_7B_CB_62
bluetoothd[202219]: src/device.c:device_free() 0x55e6ca0
bluetoothd[202219]: src/device.c:device_remove() Removing device /org/bluez/hci0/dev_C0_7A_A5_00_9E_4A
bluetoothd[202219]: src/service.c:change_state() 0x5ee76e0: device C0:7A:A5:00:9E:4A profile avrcp-controller state changed: disconnected -> unavailable (0)
bluetoothd[202219]: profiles/audio/control.c:path_unregister() Unregistered interface org.bluez.MediaControl1 on path /org/bluez/hci0/dev_C0_7A_A5_00_9E_4A
bluetoothd[202219]: src/service.c:btd_service_unref() 0x5ef13d0: ref=1
bluetoothd[202219]: src/service.c:btd_service_unref() 0x5ee76e0: ref=1
bluetoothd[202219]: src/service.c:btd_service_unref() 0x5ee76e0: ref=0
bluetoothd[202219]: src/service.c:change_state() 0x5ef13d0: device C0:7A:A5:00:9E:4A profile audio-avrcp-target state changed: disconnected -> unavailable (0)
bluetoothd[202219]: src/service.c:btd_service_unref() 0x5ef13d0: ref=0
bluetoothd[202219]: src/service.c:change_state() 0x5ef83e0: device C0:7A:A5:00:9E:4A profile a2dp-sink state changed: disconnected -> unavailable (0)
bluetoothd[202219]: profiles/audio/sink.c:sink_unregister() /org/bluez/hci0/dev_C0_7A_A5_00_9E_4A
bluetoothd[202219]: src/service.c:btd_servic

Vudentz, 2020-11-30:
@kapiwko Can you try with the above patch?

kapiwko, 2020-11-30:
@Vudentz I've applied thit patch and it doesn't crash anymore!

Unfortunately only one A2DP profile works.

```
bluetoothd[276117]: Bluetooth daemon 5.55
bluetoothd[276117]: src/main.c:parse_config() parsing ./src/main.conf
bluetoothd[276117]: src/main.c:parse_config() Key file does not have key “DiscoverableTimeout” in group “General”
bluetoothd[276117]: src/main.c:parse_config() Key file does not have key “AlwaysPairable” in group “General”
bluetoothd[276117]: src/main.c:parse_config() Key file does not have key “PairableTimeout” in group “General”
bluetoothd[276117]: src/main.c:parse_config() Key file does not have key “Privacy” in group “General”
bluetoothd[276117]: src/main.c:parse_config() Key file does not have key “JustWorksRepairing” in group “General”
bluetoothd[276117]: src/main.c:parse_config() Key file does not have key “TemporaryTimeout” in group “General”
bluetoothd[276117]: src/main.c:parse_config() Key file does not have key “Name” in group “General”
bluetoothd[276117]: src/main.c:parse_config() Key file does not have key “Class” in group “General”
bluetoothd[276117]: src/main.c:parse_config() Key file does not have key “DeviceID” in group “General”
bluetoothd[276117]: src/main.c:parse_config() Key file does not have key “ReverseServiceDiscovery” in group “General”
bluetoothd[276117]: src/main.c:parse_config() Key file does not have key “Cache” in group “GATT”
bluetoothd[276117]: src/main.c:parse_config() Key file does not have key “KeySize” in group “GATT”
bluetoothd[276117]: src/main.c:parse_config() Key file does not have key “ExchangeMTU” in group “GATT”
bluetoothd[276117]: src/main.c:parse_config() Key file does not have key “Channels” in group “GATT”
bluetoothd[276117]: src/main.c:parse_config() Key file does not have key “SessionMode” in group “AVDTP”
bluetoothd[276117]: src/main.c:parse_config() Key file does not have key “StreamMode” in group “AVDTP”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have group “BREDR”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “MinAdvertisementInterval” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “MaxAdvertisementInterval” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “MultiAdvertisementRotationInterval” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “ScanIntervalAutoConnect” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “ScanWindowAutoConnect” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “ScanIntervalSuspend” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “ScanWindowSuspend” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “ScanIntervalDiscovery” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “ScanWindowDiscovery” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “ScanIntervalAdvMonitor” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “ScanWindowAdvMonitor” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “ScanIntervalConnect” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “ScanWindowConnect” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “MinConnectionInterval” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “MaxConnectionInterval” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “ConnectionLatency” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “ConnectionSupervisionTimeout” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “Autoconnecttimeout” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “AdvMonAllowlistScanDuration” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “AdvMonNoFilterScanDuration” in group “LE”
bluetoothd[276117]: src/main.c:parse_mode_config() Key file does not have key “EnableAdvMonInterleaveScan” in group “LE”
bluetoothd[276117]: src/adapter.c:adapter_init() sending read version command
bluetoothd[276117]: Starting SDP server
bluetoothd[276117]: src/sdpd-service.c:register_device_id() Adding device id record for 0002:1d6b:0246:0537
[... cut ...]
bluetoothd[276117]: profiles/audio/avdtp.c:avdtp_unref() 0x55b992fd2e00: ref=1
bluetoothd[276117]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_1C_52_16_4E_B9_9F: SINK_STATE_PLAYING -> SINK_STATE_DISCONNECTED
bluetoothd[276117]: profiles/audio/a2dp.c:channel_remove() chan 0x55b992fc0a60
bluetoothd[276117]: profiles/audio/avdtp.c:avdtp_unref() 0x55b992fd2e00: ref=0
bluetoothd[276117]: profiles/audio/avdtp.c:avdtp_free() 0x55b992fd2e00
```

Should I create new issue for that?

Vudentz, 2020-12-01:
Yes ,please have a separate issue for different codec settings, we also need the HCI traces (btmon).

closed 2020-12-01 by Vudentz