bluez/bluez #61: Crackling or choppy sound when scanning for a device.

842Mono, 2020-12-01:
Hi.

I'm using KDE neon (like Kubuntu with a different way updates are served). I use bluetooth headphones. The sound coming through the headphones is crackling or choppy when scanning is on.

I play music in the background, then I do `$ bluetoothctl` then `# scan on`. The sound immediately chops or crackles.

I filed a bug report at bugs.kde.org (here: https://bugs.kde.org/show_bug.cgi?id=429830), however they directed me to here. Also they couldn't reproduce the issue on their end. It might be hardware specific or something.

Also note that I tried making a screen recording, but the issue didn't appear in the screen recording.

Am I in the right place? Is this where I should report the bug?

Vudentz, 2020-12-01:
@842Mono Im afraid we can't really do much if the controller scheduler can't keep up with the audio data while scanning.

842Mono, 2020-12-01:
@Vudentz thank you for replying.

I'm surprised you understood what was going on so quickly. How do you know it's an issue with the controller scheduler?

Anyhow, so you mean this is, like, not a software issue?

Vudentz, 2020-12-01:
> @Vudentz thank you for replying.
> 
> I'm surprised you understood what was going on so quickly. How do you know it's an issue with the controller scheduler?
> 
> Anyhow, so you mean this is, like, not a software issue?

Yep, it is a recuring problem with some controllers it seems, if you attempt to scan it would prioritize the scanning over the audio data, there may be ways around it depending on the paremeter used for scanning but afaik we don't use very aggresive intervals for it.

jonas2515, 2023-10-28:
Does your device have a Broadcomm bluetooth chip? For these you have to send a vendor-specific HCI command to make audio streams work reliably, see eg. here in android: https://github.com/LineageOS/android_system_bt/blob/lineage-19.1/stack/l2cap/l2c_utils.cc#L2246

closed 2025-08-30 by github-actions[bot]