bluez/bluez #63: No agent available for request type 2

domenkozar, 2020-12-13:
```
Dec 13 20:19:58 guava bluetoothd[1620]: No agent available for request type 2
Dec 13 20:19:58 guava bluetoothd[1620]: device_confirm_passkey: Operation not permitted
Dec 13 20:19:58 guava bluetoothd[1620]: connect error: Permission denied (13)
Dec 13 20:20:01 guava bluetoothd[1620]: connect error: Too many levels of symbolic links (40)
```

I get a bunch of errors, not really clear if they are related or not.

domenkozar, 2020-12-13:
This happens if I use bluetooth manager, but if I use bluetoothctl then it works.

Vudentz, 2021-01-04:
@domenkozar Not sure what are you refering to as bluetooth manager, anyway it does look like it is a upper layer problem or lack of pairing agent.

closed 2021-01-04 by Vudentz