bluez/bluez #65: Wrong GATT MTU after exchange

mityada, 2020-12-13:
If GATT client requests MTU exchange 2 times, with first MTU larger then second MTU, BlueZ continues to use the first (larger) value. If client then reads a long characteristic it may get an error, as the value might be longer then (MTU - 1).

Here is a minimal reproducer with a Python GATT server and Android app as a client: https://github.com/mityada/bluez-mtu

Android app requests MTU of 200 bytes and then lowers it to 100 bytes:
```
12-14 01:33:27.111 28796 28809 D BluezMtu: Connected
12-14 01:33:27.111 28796 28809 D BluezMtu: Requesting MTU 200
12-14 01:33:27.112 28796 28809 D BluetoothGatt: configureMTU() - device: DE:AD:BE:EF:CA:FE mtu: 200
12-14 01:33:28.006 28796 28808 D BluetoothGatt: onConfigureMTU() - Device=DE:AD:BE:EF:CA:FE mtu=200 status=0
12-14 01:33:28.006 28796 28808 D BluezMtu: onMtuChanged 200 0
12-14 01:33:28.006 28796 28808 D BluezMtu: Requesting MTU 100
12-14 01:33:28.006 28796 28808 D BluetoothGatt: configureMTU() - device: DE:AD:BE:EF:CA:FE mtu: 100
12-14 01:33:28.021 28796 28809 D BluetoothGatt: onConfigureMTU() - Device=DE:AD:BE:EF:CA:FE mtu=100 status=0
12-14 01:33:28.021 28796 28809 D BluezMtu: onMtuChanged 100 0
```
```
Dec 14 01:35:09 laptop bluetoothd[41379]: src/device.c:gatt_debug() MTU exchange complete, with MTU: 200
Dec 14 01:35:09 laptop bluetoothd[41379]: src/device.c:gatt_debug() MTU exchange complete, with MTU: 100
```
However BlueZ continues to use MTU 200 and the read fails:
```
TestCharacteristic Read
mtu 200
```
```
12-14 01:33:28.064 28232 28259 E bt_att  : invalid response/indicate pkt size: 200, PDU size: 100
12-14 01:33:28.065 28796 28809 W BluetoothGatt: onCharacteristicRead() - Device=DE:AD:BE:EF:CA:FE handle=45 Status=133
12-14 01:33:28.065 28796 28809 D BluezMtu: onCharacteristicRead 133
12-14 01:33:28.066 28796 28809 E BluezMtu: Read failed
```

Looks like the problem is either in the code that sets the MTU after exchange
https://github.com/bluez/bluez/blob/98f5da02ce148a61efdfb2557152448f04bab51f/src/shared/gatt-server.c#L1536-L1538
https://github.com/bluez/bluez/blob/98f5da02ce148a61efdfb2557152448f04bab51f/src/shared/att.c#L1367-L1368
or in the code that uses it when reading characteristics
https://github.com/bluez/bluez/blob/98f5da02ce148a61efdfb2557152448f04bab51f/src/gatt-database.c#L2206-L2208

Vudentz, 2020-12-18:
Actually you are not suppose to change the MTU once it has been negotiated.

mityada, 2020-12-27:
> Actually you are not suppose to change the MTU once it has been negotiated.

Thanks, now I see it. However the behaviour is still weird. The second exchange completes without any errors on both sides, also there is no issue if the second MTU is larger than the first. I don't see why this case couldn't be handled in a more "expected" way, even though it's not strictly in line with the specification.

closed 2025-08-30 by github-actions[bot]