bluez/bluez #66: Can't connect new Xbox controller

Vladius25, 2020-12-14:
When i try to connect Xbox Series X controller, it keeps connecting and disconnecting.
```
[CHG] Device 98:7A:14:EC:DB:F5 Connected: no
[CHG] Device 98:7A:14:EC:DB:F5 Connected: yes
[CHG] Device 98:7A:14:EC:DB:F5 Connected: no
[CHG] Device 98:7A:14:EC:DB:F5 Connected: yes
[CHG] Device 98:7A:14:EC:DB:F5 Connected: no
```
I have ERTM disabled.

Errors in logs:
```
bluetoothd[800]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
bluetoothd[800]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
bluetoothd[800]: profiles/input/hog-lib.c:report_map_read_cb() Report Map read failed: Request attribute has encountered an unlikely error
bluetoothd[800]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
bluetoothd[800]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
```

I have Fedora 33 (5.9.13-200.fc33.x86_64), Bluez 5.55, and bluetooth chipset from Intel.

Vladius25, 2020-12-17:
Seems to be gamepad problem
https://www.reddit.com/r/linux_gaming/comments/js0trh/anyone_get_the_xbox_series_x_controller_to/gddwyjk?utm_source=share&utm_medium=web2x&context=3

closed 2020-12-17 by Vladius25

sombriks, 2022-03-16:
i had similar issue and what worked for me was this: https://www.reddit.com/r/linux_gaming/comments/js0trh/anyone_get_the_xbox_series_x_controller_to/gddwyjk/?utm_source=share&utm_medium=web2x&context=3

kakra, 2022-03-16:
This is more likely an issue with older bluez releases - also, since kernel 5.12, ERTM should not be disabled because otherwise rumble-related firmware crashes can occur and 5.12 has the needed L2CAP patch. I do not have any connection issue since around bluez 5.60 or 5.62 - even without `Privacy=device` (it doesn't seem to be needed and probably just works around another issue some people are having). If there has been a faulty pairing before, all cached data needs to be removed from bluez.

The latest firmware updates improve Bluetooth stability, especially the older models switch to BLE, and the XBXS model also gets a firmware update which should improve things. The update can be installed via an Xbox or the Windows Xbox Accessories app.

Here's a list of issues found with possible solutions: https://github.com/atar-axis/xpadneo/projects/11