bluez/bluez #67: ios 14.2 app disconnect with a peer ble device using BlueZ 5.50 sometimes

lirengen10, 2020-12-17:
Dear Sir,

I am using BlueZ on my ble device(Linux 4.4), recently, after I upgrade BlueZ to 5.50 from 5.47, then ios app would disconnect with my ble device after paring ble device sometimes, but it works fine when my ble device used BlueZ 5.47, and I saw the difference is that ios would show "request pairing" popup ui when connect to device with BlueZ 5.50, but there is not "request pairing" UI if connect to ble device with BlueZ5.47 . 

From the log, looks like it is related to long term key secure connection, how to disable the "request pairing" popup UI of ios at BlueZ5.50 ? just like disappear "request paring" UI at BlueZ5.47. Thanks in advance.

bluetoothd[13467]: src/adapter.c:connected_callback() hci0 device 6F:B0:49:C7:39:E3 connected eir_len 0
bluetoothd[13467]: src/device.c:device_create() dst 6F:B0:49:C7:39:E3
bluetoothd[13467]: src/device.c:device_new() address 6F:B0:49:C7:39:E3
bluetoothd[13467]: src/device.c:device_new() Creating device /org/bluez/hci0/dev_6F_B0_49_C7_39_E3
[btsetup] Central Device(6F:B0:49:C7:39:E3) Connected
btsetup service
btsetup: setup::presetup
lldpd event setup::presetup true received. 
2020-12-11T03:19:15 [WARN/lldpctl] unable to watch for neighbors. End of file reached
bluetoothd[13467]: src/gatt-database.c:connect_cb() New incoming LE ATT connection
bluetoothd[13467]: attrib/gattrib.c:g_attrib_ref() 0x18ab818: g_attrib_ref=1 
bluetoothd[13467]: src/device.c:load_gatt_db() Restoring 6F:B0:49:C7:39:E3 gatt database from file
bluetoothd[13467]: No cache for 6F:B0:49:C7:39:E3
bluetoothd[13467]: src/gatt-client.c:btd_gatt_client_connected() Device connected.
bluetoothd[13467]: src/device.c:gatt_debug() MTU exchange complete, with MTU: 185
bluetoothd[13467]: src/device.c:gatt_debug() Primary services found: 9
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x0001, end: 0x0005, uuid: 00001800-0000-1000-8000-00805f9b34fb
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x0006, end: 0x0009, uuid: 00001801-0000-1000-8000-00805f9b34fb
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x000a, end: 0x000e, uuid: d0611e78-bbb4-4591-a5f8-487910ae4366
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x000f, end: 0x0013, uuid: 9fa480e0-4967-4542-9390-d343dc5d04ae
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x0014, end: 0x0017, uuid: 0000180f-0000-1000-8000-00805f9b34fb
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x0018, end: 0x001d, uuid: 00001805-0000-1000-8000-00805f9b34fb
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x001e, end: 0x0022, uuid: 0000180a-0000-1000-8000-00805f9b34fb
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x0023, end: 0x002c, uuid: 7905f431-b5ce-4e99-a40f-4b1e122d00d0
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x002d, end: 0x0038, uuid: 89d3502b-0f36-433a-8ef4-c502ad55f8dc
bluetoothd[13467]: src/device.c:gatt_debug() Read By Grp Type - start: 0x0001 end: 0xffff
bluetoothd[13467]: src/device.c:gatt_debug() Read By Grp Type - start: 0x000a end: 0xffff
bluetoothd[13467]: src/device.c:gatt_debug() Read By Grp Type - start: 0x0010 end: 0xffff
bluetoothd[13467]: src/device.c:gatt_debug() Read By Type - start: 0x0006 end: 0x0009
bluetoothd[13467]: src/device.c:gatt_debug() Characteristics found: 16
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x0002, end: 0x0003, value: 0x0003, props: 0x02, uuid: 00002a00-0000-1
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x0004, end: 0x0006, value: 0x0005, props: 0x02, uuid: 00002a01-0000-1
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x0007, end: 0x000a, value: 0x0008, props: 0x20, uuid: 00002a05-0000-1
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x000b, end: 0x000f, value: 0x000c, props: 0x98, uuid: 8667556c-9a37-4
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x0010, end: 0x0014, value: 0x0011, props: 0x98, uuid: af0badb1-5b99-4
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x0015, end: 0x0018, value: 0x0016, props: 0x12, uuid: 00002a19-0000-1
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x0019, end: 0x001b, value: 0x001a, props: 0x12, uuid: 00002a2b-0000-1
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x001c, end: 0x001e, value: 0x001d, props: 0x02, uuid: 00002a0f-0000-1
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x001f, end: 0x0020, value: 0x0020, props: 0x02, uuid: 00002a29-0000-1
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x0021, end: 0x0023, value: 0x0022, props: 0x02, uuid: 00002a24-0000-1
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x0024, end: 0x0026, value: 0x0025, props: 0x88, uuid: 69d1d8f3-45e1-4
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x0027, end: 0x0029, value: 0x0028, props: 0x10, uuid: 9fbf120d-6301-4
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x002a, end: 0x002d, value: 0x002b, props: 0x10, uuid: 22eac6e9-24d6-4
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x002e, end: 0x0031, value: 0x002f, props: 0x98, uuid: 9b3c81d8-57b1-4
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x0032, end: 0x0035, value: 0x0033, props: 0x98, uuid: 2f7cabce-808d-4
bluetoothd[13467]: src/device.c:gatt_debug() start: 0x0036, end: 0x0038, value: 0x0037, props: 0x8a, uuid: c6b2f38c-23ab-4
bluetoothd[13467]: src/device.c:gatt_debug() Find Info - start: 0x0009 end: 0x0009
bluetoothd[13467]: src/device.c:gatt_debug() Descriptors found: 1
bluetoothd[13467]: src/device.c:gatt_debug() handle: 0x0009, uuid: 00002902-0000-1000-8000-00805f9b34fb
bluetoothd[13467]: src/device.c:gatt_debug() Write Req - handle: 0x0009
bluetoothd[13467]: src/gatt-database.c:gatt_ccc_write_cb() CCC write called for handle: 0x0009
bluetoothd[13467]: src/device.c:gatt_debug() Descriptors found: 2
bluetoothd[13467]: src/device.c:gatt_debug() handle: 0x000d, uuid: 00002900-0000-1000-8000-00805f9b34fb
bluetoothd[13467]: src/device.c:gatt_debug() handle: 0x000e, uuid: 00002902-0000-1000-8000-00805f9b34fb
bluetoothd[13467]: src/device.c:gatt_debug() Read By Type - start: 0x000a end: 0x000f
bluetoothd[13467]: src/device.c:gatt_debug() Ext. prop value: 0x0001
bluetoothd[13467]: src/device.c:gatt_debug() Value set status: 0
bluetoothd[13467]: src/device.c:gatt_debug() Find Info - start: 0x000f end: 0x000f
bluetoothd[13467]: src/device.c:gatt_debug() Descriptors found: 2
bluetoothd[13467]: src/device.c:gatt_debug() handle: 0x0012, uuid: 00002900-0000-1000-8000-00805f9b34fb
bluetoothd[13467]: src/device.c:gatt_debug() handle: 0x0013, uuid: 00002902-0000-1000-8000-00805f9b34fb
bluetoothd[13467]: src/device.c:gatt_debug() Write Req - handle: 0x000f
bluetoothd[13467]: src/gatt-database.c:gatt_ccc_write_cb() CCC write called for handle: 0x000f
bluetoothd[13467]: src/gatt-database.c:ccc_write_cb() External CCC write received with value: 0x0001
[... 126 lines cut ...]
bluetoothd[13467]: profiles/battery/battery.c:parse_battery_level() Battery Level updated: 52%
bluetoothd[13467]: profiles/battery/battery.c:batt_io_ccc_written_cb() Battery Level: notification enabled

~ # bluetoothd[13467]: src/adapter.c:dev_disconnected() Device 88:64:40:6F:F8:35 disconnected, reason 3
bluetoothd[13467]: src/adapter.c:adapter_remove_connection() 
bluetoothd[13467]: plugins/policy.c:disconnect_cb() reason 3
bluetoothd[13467]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 88:64:40:6F:F8:35 type 1 status 0xe
bluetoothd[13467]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
bluetoothd[13467]: src/device.c:device_bonding_failed() status 14
bluetoothd[13467]: src/adapter.c:resume_discovery() 
bluetoothd[13467]: src/device.c:att_disconnected_cb() 
bluetoothd[13467]: src/device.c:att_disconnected_cb() Connection reset by peer (104)
bluetoothd[13467]: src/service.c:change_state() 0x18ae718: device 88:64:40:6F:F8:35 profile gap-profile state changed: connected -> disconnecting (0)
bluetoothd[13467]: src/service.c:change_state() 0x18ae718: device 88:64:40:6F:F8:35 profile gap-profile state changed: disconnecting -> disconnected (0)
bluetoothd[13467]: src/service.c:change_state() 0x18affe8: device 88:64:40:6F:F8:35 profile batt-profile state changed: connected -> disconnecting (0)
bluetoothd[13467]: src/service.c:change_state() 0x18affe8: device 88:64:40:6F:F8:35 profile batt-profile state changed: disconnecting -> disconnected (0)
bluetoothd[13467]: src/service.c:change_state() 0x18b04f8: device 88:64:40:6F:F8:35 profile deviceinfo state changed: connected -> disconnecting (0)
bluetoothd[13467]: src/service.c:change_state() 0x18b04f8: device 88:64:40:6F:F8:35 profile deviceinfo state changed: disconnecting -> disconnected (0)
bluetoothd[13467]: src/gatt-client.c:btd_gatt_client_disconnected() Device disconnected. Cleaning up.
bluetoothd[13467]: src/device.c:att_disconnected_cb() Automatic connection disabled
bluetoothd[13467]: attrib/gattrib.c:g_attrib_unref() 0x18ab818: g_attrib_unref=0 
bluetoothd[13467]: src/gatt-database.c:att_disconnected() 

Best Regards,
Rengen Li

Star152, 2020-12-18:
I have the same issues，Whenever I connect RPi  with the  app on my ios phone, I get the following error：

> [Callback]centralManager(central,didDisconnectPeripheral:peripheral,error:Failed to encrypt the connection,the connection has timed out unexpectedly.)

Vudentz, 2020-12-21:
I wonder if this is related to having battery profile which apparently requires pairing in case of iOS, that said battey does seems to be able to be read and we were able to enable notification, in fact there is a new LTK:

`bluetoothd[13467]: src/adapter.c:new_long_term_key_callback() hci0 new LTK for 88:64:40:6F:F8:35 type 2 enc_size 16`

lirengen10, 2020-12-22:
@Vudentz ,
Thanks for your reply, can we disable battery notification from bluez5.50 ? how to disable it ?

Path-A, 2021-03-25:
Yes, this is a problem with the battery profile. This should be disabled by default. [People have had success recompiling Bluez with the battery modules removed](https://stackoverflow.com/q/66690927/10671392). Any word on fixing this issue?

Path-A, 2021-03-25:
I was able to get this to work [two different ways without recompiling](https://stackoverflow.com/a/66807679/10671392):

The simplest way is to prevent the battery plugin from loading on boot by modifying your bluetooth service file with "bluetoothd -P battery". You can read more about it in the link above.

lirengen10, 2021-04-25:
Hi Path-A,
It works, thank a lot.

closed 2021-04-25 by lirengen10