bluez/bluez #71: HOG does not register for BT_ATT_OP_HANDLE_NFY_MULT

zxtwonder, 2021-01-04:
gatt-client currently reports the client supports `BT_GATT_CHRC_CLI_FEAT_NFY_MULTI`
https://github.com/bluez/bluez/blob/c96d5ec7aafc8fd988f6f19b896115853b951d91/src/shared/gatt-client.c#L1981

However, hog-lib still only subscribes to `ATT_OP_HANDLE_NOTIFY` and ignores input notification sent as `Handle Multiple Value Notification (0x23)`.
https://github.com/bluez/bluez/blob/8b9ec615b3e174b02ff2df93e1980a786e7611a3/profiles/input/hog-lib.c#L339-L342

It looks like hog is implemented using a different codebase - gattrib. Is it deprecated? As a result, it was not able to take advantage of the update to `gatt-client.c:notify_cb`:
https://github.com/bluez/bluez/blob/c96d5ec7aafc8fd988f6f19b896115853b951d91/src/shared/gatt-client.c#L2166-L2182

Vudentz, 2021-01-04:
Yep, we need to convert HoG to use bt_gatt_client directly otherwise it won't be ablet to take advantage of such feature.

@zxtwonder do you actually have a device that does make use of of notify multiple already?

zxtwonder, 2021-01-04:
I am working on a bluez 5.55 to bluez 5.55 project and encountered this issue of inconsistent support of notify multiple.

Just curious - is the plan to deprecate attrib directory? It feels like an early implementation and there seems to be some issue handling `ACL_CONT` in gatttool when I was playing with it the other day.

Vudentz, 2021-01-04:
> I am working on a bluez 5.55 to bluez 5.55 project and encountered this issue of inconsistent support of notify multiple.
>
> Just curious - is the plan to deprecate attrib directory? It feels like an [...]

Yes, attrib is an early glib based implementation of ATT/GATT protocol which has already been deprecated, gattool is already deprecated as well:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=b1eb2c4cd057624312e0412f6c4be000f7fc3617

endrift, 2022-12-29:
I'm bumping up against this right now. How hard would this be to update?

Vudentz, 2023-01-05:
@endrift @zxtwonder please verify with the following set:

https://patchwork.kernel.org/project/bluetooth/list/?series=709312

endrift, 2023-01-06:
Yes, works fine on top of latest git, thanks.

closed 2023-01-06 by BluezTestBot