bluez/bluez #73: How does GATT server indicate to the client

veelenga, 2021-01-05:
Hey there.

I would like to understand how the GATT server can "indicate" to the client. I'm working on a "Cycling Speed and Cadence" service and a part of its procedure is to set a value by the client and get the indication back (meaning the server notifies the client using a blocking request)

> When the Set Cumulative Value Op Code is written to the SC Control Point and if the Wheel Revolution Data feature is supported by the Server, the Server shall set the Cumulative Wheel Revolutions value to the same value as the UINT32 parameter which accompanies the op code. For example, a parameter of 0x00000000 will set the Cumulative Wheel Revolutions value to 0. The response shall be indicated when the Cumulative Wheel Revolutions value is applied using the Response Code Op Code, the Request Op Code along with “Success” or other appropriate Response Value.

I read the [GATT API](https://github.com/bluez/bluez/blob/c96d5ec7aafc8fd988f6f19b896115853b951d91/doc/gatt-api.txt#L62-L282) and the characteristic doesn't seem to have any specific method to indicate to the client.

zxtwonder, 2021-01-08:
For the general `indicate` flow:
1. Server sends indication:
https://github.com/bluez/bluez/blob/a23fa59b09bbda8a35596ebccb78d352b6f5561d/test/example-gatt-server#L400-L402
2. Server receives confirmation:
https://github.com/bluez/bluez/blob/ca01d74f38946cce61c05dbcd5ac59a7624ebbcd/src/gatt-database.c#L1283

Is this what you are looking for?

Vudentz, 2021-01-08:
@zxtwonder thanks

Emiting PropertiesChanged with Value will trigger bluetoothd to notifify/indicate if there are peers connected that have subscribed.

closed 2021-01-08 by Vudentz

inonoob, 2021-02-17:
Hey @veelenga, 

I don't know if it is still relevant but I had the same kind of issue with the implementation of the Fitness equipment profile. The App request via Bluetooth to take Controle over the fitness machine and therefore sends back op-code to tell the app success or fail. I had to write additional code in the gatt-server python file. 

This is the part of my code. it might help you with your code: 

https://github.com/inonoob/pirowflo/blob/a7292914df2a1c86450eeb56b3bfaf6611da2b2f/src/waterrowerble.py#L349 

The thing is if you client ( iOS/Android device) writes to the specific Characteristic, you must have a callback function which writes the wanted op-code to the Characteristics and then call to ` ``PropertiesChanged``` in order to notify your client that the value of the characteristic has changed. And that value is your op-code. 

```self.PropertiesChanged(GATT_CHRC_IFACE, {'Value': value}, [])```