bluez/bluez #75: EATT channel triggers false failure from gatt-server pending operation check

zxtwonder, 2021-01-06:
I am doing a BLE GATT test between two machines running bluez 5.55. Sometimes, especially after reconnection, some read operations fail with `BT_ATT_ERROR_UNLIKELY`.

The error is returned from the check on pending read operation in `handle_read_req` and the existing pending operation points to an EATT channel.
https://github.com/bluez/bluez/blob/5a180f2ec9edfacafd95e5fed20d36fe8e077f07/src/shared/gatt-server.c#L970-L973

`pending_read_op` and `pending_write_op` seem to exist only for validation purpose based on my initial investigation. They do not look right any more with EATT. Should we record pending operations for each channel and check on a per-channel basis now?

The following are gdb outputs on the GATT server side:

gdb stack trace:
```c
(gdb) bt
#0  handle_read_req (chan=0x55f8e4c88a60, server=0x55f8e4c91ea0, opcode=10 '\n', handle=29, offset=<optimized out>) at src/shared/gatt-server.c:991
#1  0x000055f8e4a1ff09 in handle_notify (pdu_len=3, pdu=0x55f8e4c87a20 "\n\035", chan=0x55f8e4c88a60) at src/shared/att.c:972
#2  can_read_data (io=<optimized out>, user_data=0x55f8e4c88a60) at src/shared/att.c:1063
#3  0x000055f8e4a2b679 in watch_callback (channel=<optimized out>, cond=<optimized out>, user_data=<optimized out>) at src/shared/io-glib.c:170
#4  0x00007faf7cdbe3df in g_main_context_dispatch () from target:/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007faf7cdbe788 in ?? () from target:/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#6  0x00007faf7cdbeaa3 in g_main_loop_run () from target:/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#7  0x000055f8e4a2bcc9 in mainloop_run () at src/shared/mainloop-glib.c:79
#8  0x000055f8e4a2c130 in mainloop_run_with_signal (func=<optimized out>, user_data=0x0) at src/shared/mainloop-notify.c:201
#9  0x000055f8e4989779 in main (argc=<optimized out>, argv=<optimized out>) at src/main.c:971
```

gdb output on the channels (note the different type):
```c
(gdb) print *chan
$17 = {att = 0x55f8e4c6a0a0, fd = 48, io = 0x55f8e4c8cd80, type = 1 '\001', sec_level = 0, 
  queue = 0x55f8e4c8cea0, pending_req = 0x55f8e4c9e8b0, pending_ind = 0x0, writer_active = false, 
  in_req = true, buf = 0x55f8e4c87a20 "\n\035", mtu = 517}

(gdb) print *server->pending_read_op->chan
$18 = {att = 0x55f8e4c6a0a0, fd = 52, io = 0x55f8e4c920b0, type = 2 '\002', sec_level = 0, 
  queue = 0x55f8e4c9bfe0, pending_req = 0x0, pending_ind = 0x0, writer_active = false, in_req = true, 
  buf = 0x55f8e4ca0730 "\n\031", mtu = 517}
```

Vudentz, 2021-01-06:
That seems indeed a problem, not sure what was the intention of these pending_*_op to begin with since the att.c itself should be able to detect if there is already an ongoing request at the channel so we might just get rid of them.

zxtwonder, 2021-01-07:
Since there are two of them -`pending_read_op` and `pending_write_op`, it makes me really curious: Do ATT and GATT allow separate pending read and write operations? It is kind of a weird feature/limitation if true. Otherwise, it is definitely nice to have ATT layer handle them all.

Vudentz, 2021-01-07:
> Since there are two of them -`pending_read_op` and `pending_write_op`, it makes me really curious: Do ATT and GATT allow separate pending read and write operations? It is kind of a weird feature/limitation if true. Otherwise, it is definitely nice to have ATT layer handle them all.

Actually the read/write distintion was wrong as well, ATT allows only one outstanding request per channel, only other types of PDU's can be sent while a request is outstanding e.g. commands, notifications and indications.

closed 2021-01-08 by Vudentz