bluez/bluez #76: Can't get HIDP connection info

Maradonna90, 2021-01-07:
I had my BT Keyboard working yesterday (Keychron K2), used it today with a different arch device successfully. Than switched back to my regular and it wouldn't connect anymore

systemctl status bluetooth
```
Jan 07 20:55:46 marco bluetoothd[388]: src/main.c:parse_controller_config() Key file does not have key “LEMinConnectionInterval” in group “Controller”
Jan 07 20:55:46 marco bluetoothd[388]: src/main.c:parse_controller_config() Key file does not have key “LEMaxConnectionInterval” in group “Controller”
Jan 07 20:55:46 marco bluetoothd[388]: src/main.c:parse_controller_config() Key file does not have key “LEConnectionLatency” in group “Controller”
Jan 07 20:55:46 marco bluetoothd[388]: src/main.c:parse_controller_config() Key file does not have key “LEConnectionSupervisionTimeout” in group “Controller”
Jan 07 20:55:46 marco bluetoothd[388]: src/main.c:parse_controller_config() Key file does not have key “LEAutoconnecttimeout” in group “Controller”
Jan 07 20:55:46 marco systemd[1]: Started Bluetooth service.
Jan 07 20:55:46 marco bluetoothd[388]: Starting SDP server
Jan 07 20:55:46 marco bluetoothd[388]: Bluetooth management interface 1.14 initialized
Jan 07 20:58:35 marco bluetoothd[388]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
Jan 07 20:58:40 marco bluetoothd[388]: profiles/input/device.c:control_connect_cb() connect to DC:2C:26:F3:6D:65: Host is down (112)
```

Maradonna90, 2021-01-08:
ok. so on my other arch machine its still working. MAC is obv the same. it just work. 
Config under `/etc/bluetooth/main.conf` is the same as well

akana, 2021-11-13:
I have the same problem with Keychron K2.

```
# journal -f
Nov 13 09:00:26 127.0.0.1localhost bluetoothd[40166]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
Nov 13 09:00:31 127.0.0.1localhost bluetoothd[40166]: profiles/input/device.c:control_connect_cb() connect to DC:2C:26:06:15:95: Host is down (112)

```

- OS : Arch Linux x86_64
- Kernel :  5.10.78-1-lts
- bluez : 5.62
- bluez-libs : 5.62
- bluez-utils : 5.62

tzawlxih, 2022-01-31:
Same issue with Epomaker Keyboard.

```
Jan 31 11:39:48 local bluetoothd[26588]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
Jan 31 11:39:54 local bluetoothd[26588]: profiles/input/device.c:control_connect_cb() connect to DC:2C:26:FD:E1:10: Host is down (112)
```

- Arch Linux x86_64
- 5.16.4-arch1-1
- bluez 5.63-1

naiame-neltheri, 2022-04-21:
Same problem with Keychron K3. It was working fine until a week ago, maybe something related with the latest update?
```
Apr 21 23:29:38 local bluetoothd[444]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
Apr 21 23:29:43 local bluetoothd[444]: profiles/input/device.c:control_connect_cb() connect to DC:2C:26:09:C4:2A: Host is down (112)
```

macmpi, 2022-05-13:
referring https://github.com/bluez/bluez/issues/319 with further tests scenarii

closed 2025-08-30 by github-actions[bot]