bluez/bluez #77: Android device disconnected

Star152, 2021-01-08:
I am using the bluez5.55 example. I run program example-advertisement and example-gatt-server, I get the following error when I use the "NRF connect" app on my Android phone to connect to Bluetooth：

> Error8(0x8):GATT INSUF AUTHORIZATION
Error8(0x8):GATT CONN TIMEOUT

This is the settings on my raspberry pi:

> pi@raspberrypi:~ $ sudo btmon
Bluetooth monitor ver 5.55
= Note: Linux version 4.19.118-v7+ (armv7l)                                           0.590540
= Note: Bluetooth subsystem version 2.22                                              0.590547 
= New Index: B8:27:EB:44:EB:EF (Primary,UART,hci0)                             [hci0] 0.590549 
= Open Index: B8:27:EB:44:EB:EF                                                [hci0] 0.590551 
= Index Info: B8:27:EB:44:EB:EF (Cypress Semiconductor)                        [hci0] 0.590552
@ MGMT Open: bluetoothd (privileged) version 1.14                            {0x0001} 0.590556 
@ MGMT Open: btmgmt (privileged) version 1.14                                {0x0002} 0.590559 
@ MGMT Event: Command Complete (0x0001) plen 283                    {0x0003} [hci0] 292.687197
      Read Controller Information (0x0004) plen 280
        Status: Success (0x00)
        Address: B8:27:EB:44:EB:EF (Raspberry Pi Foundation)
        Version: Bluetooth 5.0 (0x09)
        Manufacturer: Cypress Semiconductor (305)
        Supported settings: 0x0001bfff
          Powered
          Connectable
          Fast Connectable
          Discoverable
          Bondable
          Link Security
          Secure Simple Pairing
          BR/EDR
          High Speed
          Low Energy
          Advertising
          Secure Connections
          Debug Keys
          Privacy
          Static Address
          PHY Configuration
        Current settings: 0x00000ac1
          Powered
          Secure Simple Pairing
          BR/EDR
          Low Energy
          Secure Connections
        Class: 0x000000
          Major class: Miscellaneous
          Minor class: 0x00
        Name: raspberrypi
        Short name:

When APP connect to Bluetooth,here is the message:

> > HCI Event: LE Meta Event (0x3e) plen 19                                #93 [hci0] 564.428906
      LE Connection Complete (0x01)
        Status: Success (0x00)
        Handle: 65
        Role: Slave (0x01)
        Peer address type: Random (0x01)
        Peer address: 42:64:D9:2C:A3:9D (Resolvable)
        Connection interval: 45.00 msec (0x0024)
        Connection latency: 0 (0x0000)
        Supervision timeout: 5000 msec (0x01f4)
        Master clock accuracy: 0x00
@ MGMT Event: Device Connected (0x000b) plen 13                     {0x0003} [hci0] 564.428963 
        LE Address: 42:64:D9:2C:A3:9D (Resolvable)
        Flags: 0x00000000
        Data length: 0
@ MGMT Event: Device Connected (0x000b) plen 13                     {0x0001} [hci0] 564.428963 
        LE Address: 42:64:D9:2C:A3:9D (Resolvable)
        Flags: 0x00000000
        Data length: 0
@ MGMT Event: Device Connected (0x000b) plen 13                     {0x0002} [hci0] 564.428963 
        LE Address: 42:64:D9:2C:A3:9D (Resolvable)
        Flags: 0x00000000
        Data length: 0
> HCI Event: Vendor (0xff) plen 5                                        #94 [hci0] 564.428915 
        55 00 00 41 00                                   U..A.
< HCI Command: LE Read Remote Used Features (0x08|0x0016) plen 2         #95 [hci0] 564.429154 
        Handle: 65
> HCI Event: Command Status (0x0f) plen 4                                #96 [hci0] 564.429829 
      LE Read Remote Used Features (0x08|0x0016) ncmd 1
        Status: Success (0x00)
> HCI Event: Command Complete (0x0e) plen 14                             #97 [hci0] 564.429845 
      LE Read Remote Used Features (0x08|0x0016) ncmd 1
        Status: Success (0x00)
        00 00 00 00 00 00 00 00 00 00                    ..........
> HCI Event: LE Meta Event (0x3e) plen 12                                #98 [hci0] 564.500950 
      LE Read Remote Used Features (0x04)
        Status: Success (0x00)
        Handle: 65
        Features: 0x3f 0x00 0x00 0x00 0x00 0x00 0x00 0x00
          LE Encryption
          Connection Parameter Request Procedure
          Extended Reject Indication
          Slave-initiated Features Exchange
          LE Ping
          LE Data Packet Length Extension
= bluetoothd: src/device.c:load_gatt_db() No cache for 42:64:D9:2C:A3:9D            564.520707 
< ACL Data TX: Handle 65 flags 0x00 dlen 7                               #99 [hci0] 564.518344 
      ATT: Exchange MTU Request (0x02) len 2
        Client RX MTU: 517
> HCI Event: LE Meta Event (0x3e) plen 11                               #100 [hci0] 564.635676
      LE Data Length Change (0x07)
        Handle: 65
        Max TX octets: 251
        Max TX time: 2120
        Max RX octets: 27
        Max RX time: 328
< HCI Command: LE Set Advertising Parameters (0x08|0x0006) plen 15      #101 [hci0] 566.278946
        Min advertising interval: 1280.000 msec (0x0800)
        Max advertising interval: 1280.000 msec (0x0800)
        Type: Connectable undirected - ADV_IND (0x00)
        Own address type: Public (0x00)
        Direct address type: Public (0x00)
        Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
        Channel map: 37, 38, 39 (0x07)
        Filter policy: Allow Scan Request from Any, Allow Connect Request from Any (0x00)      
> HCI Event: Command Complete (0x0e) plen 4                             #102 [hci0] 566.279477 
      LE Set Advertising Parameters (0x08|0x0006) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1             #103 [hci0] 566.279548
        Advertising: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                             #104 [hci0] 566.280171 
      LE Set Advertise Enable (0x08|0x000a) ncmd 1
        Status: Success (0x00)
> HCI Event: Number of Completed Packets (0x13) plen 5                  #105 [hci0] 569.946271
        Num handles: 1
        Handle: 65
        Count: 1
> HCI Event: Disconnect Complete (0x05) plen 4                          #106 [hci0] 569.946331 
        Status: Success (0x00)
        Handle: 65
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8                   {0x0003} [hci0] 569.946362 
        LE Address: 42:64:D9:2C:A3:9D (Resolvable)
        Reason: Connection timeout (0x01)
@ MGMT Event: Device Disconnected (0x000c) plen 8                   {0x0001} [hci0] 569.946362 
        LE Address: 42:64:D9:2C:A3:9D (Resolvable)
        Reason: Connection timeout (0x01)
@ MGMT Event: Device Disconnected (0x000c) plen 8                   {0x0002} [hci0] 569.946362
        LE Address: 42:64:D9:2C:A3:9D (Resolvable)
        Reason: Connection timeout (0x01)

I wish you can give me some helps,thanks

Vudentz, 2021-01-08:
This sounds like a link layer problem which out of control from the host stack, have you tried with a different controller?

Star152, 2021-01-14:
Thanks，I'm using the USB Bluetooth module on a computer with Ubuntu and it's communicating fine, I'll keep trying on the Raspberry Pi.

closed 2025-08-30 by github-actions[bot]