bluez/bluez #78: OnePlus Buds Z having quick switch issues

ForceGT, 2021-01-13:
I got a OnePlus Buds Z and with version 5.50 of BlueZ it fails to quick switch
Quick Switch happens only once that is I can only switch from my Arch Linux machine to my OnePlus 6 device when I try to switch back it keeps reconnecting in bluetoothctl

Also no error is shown

[image]

Vudentz, 2021-01-13:
Interesting, is quick switch something like connecting to another address? Also it does looks like there is another address connecting and something the OnePlus Buds Z also appears to be using a different address, or is it because it is the other piece being connected?

ForceGT, 2021-01-14:
Quick Switch is when it can pair to the previously connected device automatically
If the device doesn't respond or no connection due to some reason they connect back to the device they were already connected before the attempt to connect to the previous device

Also OnePlus Buds Z uses the same address starting with EA something

ForceGT, 2021-01-14:
Also there is another issue
On opening the case of the buds they're supposed to connect automatically to the device they were paired with
Many times the same issues happens then too

What I do right now is try through the gnome-settings
And after toggling the GUI toggle it somehow happens after two-three attempts

anandrajaram21, 2021-04-23:
> Also there is another issue
> On opening the case of the buds they're supposed to connect automatically to the device they were paired with
> Many times the same issues happens then too
>
> What I do right [...]

YES, I have the exact same problem too. It never autoconnected. Always connected, disconnected, connected, disconnected, and I had to connect to it manually to get it to connect. Many have had this problem across the internet.

ForceGT, 2021-04-23:
> YES, I have the exact same problem too. It never autoconnected. Always connected, disconnected, connected, disconnected, and I had to connect to it manually to get it to connect. Many have had this [...]

I think it's some issue with the firmware only, because OnePlus has been super lazy with updates and they won't officially support Linux also so I don't know what to do next

ForceGT, 2021-08-30:
Quick Update: I think the bluez version 5.61 has fixed it for me and now everything works seamlessly as expected. I am not sure what is the reason behind this but I'm happy

closed 2021-08-30 by ForceGT