bluez/bluez #80: BLUEZ. Not found adapter

vincenzogianfelice, 2021-01-13:
Hi. I have installed Bluez 5.54 from official page, and when run bluetoothctl i display this:

`No default controller available`

and command incorporate **gatt** (`Invalid command in menu main: gatt`)

WHY!?!?!?!
Thanks

Vudentz, 2021-01-13:
Do you have any kernel logs? Does the controller shows up in dmesg or hciconfig?

vincenzogianfelice, 2021-01-13:
Hi. The command hciconfig not even install it...and kernel message, I don't think anything has changed from how it was before.

vincenzogianfelice, 2021-01-13:
Installation:

```
$ ./configure --prefix=/usr --mandir=/usr/share/man --sysconfdir=/etc --localstatedir=/var --disable-cups --enable-library
$ make -j2
$ make install
```
after installation, when digit `service bluetooth start`, the service not start!! I tried many times, but nothing.
when I managed to get it started, i have launched command **bluetoothctl** (other command: "**hcitools**, ecc" not are installed!)  i display this problem:

`No default controller available`

Ps. I Have **Debian 9** on Raspberry PI 0 W.
Mah, have not words. 
I don't know if they complicate people's lives ... Mah !!
I hope that someone can shed "light" on this problem which apparently has persisted for a long time already !.

zxtwonder, 2021-01-14:
@vincenzogianfelice I am not particularly familiar with *the* Debian 9 on rpi 0w, but what is the output of `ps ax | grep attach` on your rpi? If there is no process like btattach or hciattach listed, its bluetooth device has not been attached yet. Has bluetooth ever worked on this OS before?

If your Debian 9 does not have rpi-specific packages, you will probably need:
1. Download and install its bluetooth firmware ([github](https://github.com/LibreELEC/brcmfmac_sdio-firmware-rpi) )
2. Check your `config.txt` and make sure it does not contain `miniuart-bt` for simplicity ([reference](https://www.raspberrypi.org/documentation/configuration/uart.md))
3. Execute `sudo btattach -B /dev/ttyAMA0 -P bcm &`

Ideally, bluetooth service should be automatically started by systemd by now. (I haven't touched a 0w in a while and never on Debian. Some more tweaks might be needed...)

vincenzogianfelice, 2021-01-14:
Yes, and this is result when try to scan:

`Failed to set discoverable on: org.bluez.Error.Failed`

also I find almost nothing. hciconfig, what happened to it????

zxtwonder, 2021-01-14:
Some of those (hciconfig, hciattach, btattach, etc) are from standard debian bluez package, some (bt-network, bt-*) are from related bluez-tools package, but the other (btuart) are scripts from Raspberry PI OS. I am having a hard time to grasp what OS image you are running. Where was the OS image from?

Could you collect the following outputs?
1. `cat /etc/os-release`
2. `uname -a`
3. `ps ax | grep attach`

vincenzogianfelice, 2021-01-15:
Yes. I have raspbian **stretch 9.13** (_`Linux home 4.14.98+ #1200 Tue Feb 12 20:11:02 GMT 2019 armv6l GNU/Linux`_)

**os-release**:
```
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

zxtwonder, 2021-01-17:
For bluetooth to stop working on raspbian, there must be something wrong with the OS...

I guess it might be easier to start from a new raspbian instead. Otherwise, the following steps should get bluetooth working again on a raspberry pi. All other utils and bluetoothd are built from bluez.

> @vincenzogianfelice I am not particularly familiar with _the_ Debian 9 on rpi 0w, but what is the output of `ps ax | grep attach` on your rpi? If there is no process like btattach or hciattach [...]

inonoob, 2021-02-11:
Hey, 

do you have a second Bluetooth dongle connected via USB ? if you start the following script ```/usr/bin/btuart``` does it then start the internal Bluetooth ? 

If you have a bluetooth dongle then the internal one is never called to be activate. 

```
pi@raspberrypi:~ $ cat /usr/bin/btuart
#!/bin/sh

HCIATTACH=/usr/bin/hciattach
if grep -q "Pi 4" /proc/device-tree/model; then
  BDADDR=
else
  SERIAL=`cat /proc/device-tree/serial-number | cut -c9-`
  B1=`echo $SERIAL | cut -c3-4`
  B2=`echo $SERIAL | cut -c5-6`
  B3=`echo $SERIAL | cut -c7-8`
  BDADDR=`printf b8:27:eb:%02x:%02x:%02x $((0x$B1 ^ 0xaa)) $((0x$B2 ^ 0xaa)) $((0x$B3 ^ 0xaa))`
fi

if [ -e /sys/class/bluetooth/hci0 ]; then
  # Bluetooth is already enabled
  exit 0
fi

uart0="`cat /proc/device-tree/aliases/uart0`"
serial1="`cat /proc/device-tree/aliases/serial1`"

if [ "$uart0" = "$serial1" ] ; then
	uart0_pins="`wc -c /proc/device-tree/soc/gpio@7e200000/uart0_pins/brcm\,pins | cut -f 1 -d ' '`"
	if [ "$uart0_pins" = "16" ] ; then
		$HCIATTACH /dev/serial1 bcm43xx 3000000 flow - $BDADDR
	else
		$HCIATTACH /dev/serial1 bcm43xx 921600 noflow - $BDADDR
	fi
else
	$HCIATTACH /dev/serial1 bcm43xx 460800 noflow - $BDADDR
fi
```

so if  ```if [ -e /sys/class/bluetooth/hci0 ]; then``` sees the hci0 from the USB bluetooth dongle then it would never activate the internal one. 

maybe that helps you ?