bluez/bluez #81: [Bluez 5.45] Failed to setup connection with BLE profile

Leesphone, 2021-01-15:
We tried porting bluez 5.45 to our device (Linux kernel 3.10), we tried install LightBlue on both Android/iOS, we found:
   If connect with BLE profile, it will be failed. (Refer to attachment)
   Sometimes we observed success case but we found the profile is L2CAP but not BLE profile. (Refer to attachment)

From the fail log, it show client disconnect before connection setup successful. Any thought? Someone mentioned this may related to specific kernel version, is there any way to fix it base on kernel 3.10?

HCI sniffer - Bluetooth packet analyzer ver 5.45
device: hci0 snap_len: 1500 filter: 0xffffffff
2021-01-12 18:44:49.737475 < HCI Command: LE Set Advertising Parameters (0x08|0x0006) plen 15
    min 1280.000ms, max 1280.000ms
    type 0x00 (ADV_IND - Connectable undirected advertising) ownbdaddr 0x00 (Public)
    directbdaddr 0x00 (Public) 00:00:00:00:00:00
    channelmap 0x07 filterpolicy 0x00 (Allow scan from any, connection from any)
2021-01-12 18:44:49.738888 > HCI Event: Command Complete (0x0e) plen 4
    **LE Set Advertising Parameters (0x08|0x0006) ncmd 2
    status 0x0c
    Error: Command Disallowed**
2021-01-12 18:44:49.740771 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
2021-01-12 18:44:49.741881 > HCI Event: Command Complete (0x0e) plen 4
    **LE Set Advertise Enable (0x08|0x000a) ncmd 2
    status 0x0c
    Error: Command Disallowed**
2021-01-12 18:45:02.561020 > ACL data: handle 24 flags 0x02 dlen 7
    ATT: MTU req (0x02)
      client rx mtu 527
2021-01-12 18:45:16.451987 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 24 reason 0x13
    Reason: Remote User Terminated Connection

[L2CAP_Profile_OK.txt](https://github.com/bluez/bluez/files/5818186/L2CAP_Profile_OK.txt)
[LE_Profile_Fail.txt](https://github.com/bluez/bluez/files/5818187/LE_Profile_Fail.txt)
[LE_Profile_failed_hcidump.log](https://github.com/bluez/bluez/files/5818193/LE_Profile_failed_hcidump.log)

zxtwonder, 2021-01-17:
This is what I found in the management doc:
https://github.com/bluez/bluez/blob/ca01d74f38946cce61c05dbcd5ac59a7624ebbcd/doc/mgmt-api.txt#L20-L21
https://github.com/bluez/bluez/blob/ca01d74f38946cce61c05dbcd5ac59a7624ebbcd/doc/mgmt-api.txt#L41-L46

And https://core.docs.ubuntu.com/en/stacks/bluetooth/bluez/docs/reference/enablement/supported-kernels

It seems that bluetooth kernel modules will need to be backported at least...

Leesphone, 2021-01-18:
@zxtwonder Thanks for your update, is there any patch for Linux 3.10 kernel to support it? or any quick fix? We are using BLE for pairing only while it's hard to migrate kernel due to platform limitation.

zxtwonder, 2021-01-18:
@Leesphone Sorry, I am not familiar with the kernel side of bluetooth, as well as whether it is sufficient to only replace those bluetooth-related ko files.

Some of references that may be helpful as a starting point:
- https://forum.xda-developers.com/t/guide-how-to-port-bluez-to-kitkat-lollipop-marshmallow.2640723/
- https://bugs.chromium.org/p/chromium/issues/detail?id=356411 (No actual changelist listed there though. Will need to dig deeper)