bluez/bluez #82: SIGSEGV in att_disconnected when accessing already released device_state

zxtwonder, 2021-01-15:
#### Version:
BlueZ 5.55

#### Background:
I encountered several instances of this segmentation fault during an experiment involving `example-gatt-server`, `example-advertisement` and `simple-agent` on a PC and an iPhone as the client. I haven't been able to find out the exact reliable repro steps (it mixes with another bonding issue). It involves reconnecting, deleting the trusted device, restarting `example-gatt-server` while connected and `example-gatt-server` sending notifications.

#### Possible cause:
`bt_att_unregister_disconnect` is not called from `device_state_free`. As a result, after `device_state` is released from `send_notification_to_device`, att still has a reference to the released `device_state` and can trigger disconnect callback with it.

1. Is it possible to add the following to `device_state_free` and set `state->disc_id = 0` afterwards?
https://github.com/bluez/bluez/blob/5a180f2ec9edfacafd95e5fed20d36fe8e077f07/src/gatt-database.c#L3769-L3770
2. Why does `send_notification_to_device` uses `btd_adapter_get_device` instead of `btd_adapter_find_device`, which is used by `att_disconnected`? `btd_adapter_get_device` will even create the device if it is not found.

#### Possible event flow:
1. New device is connected and bonded. `gatt-database` adds a `device_state` for the device.
2. Delete the device. When dbus gatt server signals `property_changed_cb`, `send_notification_to_device` finds the `device_state` for the deleted device and calls `device_state_free`.
3. When the disconnect callback of att channel is finally invoked, `att_disconnected` is called with the released `device_state`.

#### gdb output:
The following happened in the same order.
1. Setup breakpoints
```c
(gdb) break btd_gatt_database_att_disconnected
Breakpoint 2 at 0x5555555bed50: file src/gatt-database.c, line 3753.
(gdb) break device_state_free
Breakpoint 3 at 0x5555555b8330: device_state_free. (3 locations)
(gdb) c
Continuing.
```
2. `device_state` is released (pointer: **0x555555697be0**): 
```c
Breakpoint 3, device_state_free (data=0x555555697be0) at src/gatt-database.c:1400
1400			device_state_free(device_state);
(gdb) bt
#0  device_state_free (data=0x555555697be0) at src/gatt-database.c:1400
#1  send_notification_to_device (data=0x555555697be0, user_data=0x7fffffffdd20) at src/gatt-database.c:1400
#2  0x0000555555606f8d in queue_foreach (queue=0x555555695620, function=function@entry=0x5555555bdde0 <send_notification_to_device>, 
    user_data=user_data@entry=0x7fffffffdd20) at src/shared/queue.c:220
#3  0x00005555555ba7e8 in send_notification_to_devices (user_data=0x5555556c0d00, conf=0x0, ccc_handle=28, len=8, 
    value=0x5555556c826c "", handle=<optimized out>, database=<optimized out>) at src/gatt-database.c:1422
#4  property_changed_cb (proxy=0x5555556c0d00, name=<optimized out>, iter=<optimized out>, user_data=0x5555556b9b00)
    at src/gatt-database.c:2721
#5  0x0000555555604692 in add_property (proxy=proxy@entry=0x5555556c0d00, name=0x5555556c825c "Value", 
    iter=iter@entry=0x7fffffffded0, send_changed=send_changed@entry=1) at gdbus/client.c:340
#6  0x0000555555604ae0 in update_properties (proxy=proxy@entry=0x5555556c0d00, iter=iter@entry=0x7fffffffdf60, 
    send_changed=send_changed@entry=1) at gdbus/client.c:372
#7  0x0000555555604ec0 in update_properties (send_changed=1, iter=0x7fffffffdf60, proxy=0x5555556c0d00) at gdbus/client.c:519
#8  properties_changed (conn=<optimized out>, msg=<optimized out>, user_data=0x5555556c0d00) at gdbus/client.c:510
#9  0x0000555555600ac8 in signal_filter (connection=0x5555556917f0, message=0x555555699220, user_data=0x5555556995e0)
    at gdbus/watch.c:407
#10 0x0000555555600cc3 in message_filter (connection=0x5555556917f0, message=0x555555699220, user_data=<optimized out>)
    at gdbus/watch.c:557
#11 0x00007ffff7e3cd61 in dbus_connection_dispatch () from /lib/x86_64-linux-gnu/libdbus-1.so.3
#12 0x00005555555ffc68 in message_dispatch (data=0x5555556917f0) at gdbus/mainloop.c:72
#13 0x00007ffff7ecd3df in g_main_context_dispatch () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#14 0x00007ffff7ecd788 in ?? () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#15 0x00007ffff7ecdaa3 in g_main_loop_run () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#16 0x0000555555618cc9 in mainloop_run () at src/shared/mainloop-glib.c:79
#17 0x0000555555619130 in mainloop_run_with_signal (func=<optimized out>, user_data=0x0) at src/shared/mainloop-notify.c:201
#18 0x0000555555576779 in main (argc=<optimized out>, argv=<optimized out>) at src/main.c:971
(gdb) info args 
data = 0x555555697be0
(gdb) info locals 
state = 0x555555697be0
state = <optimized out>
(gdb) c
Continuing.
```
3. Potential call to unregister att disconnect callback, but `device_state` has been released already without unregistering
```c
Breakpoint 2, btd_gatt_database_att_disconnected (database=0x555555690b20, device=device@entry=0x5555556b0020)
    at src/gatt-database.c:3753
3753	{
(gdb) bt
#0  btd_gatt_database_att_disconnected (database=0x555555690b20, device=device@entry=0x5555556b0020) at src/gatt-database.c:3753
#1  0x00005555555e57d6 in gatt_server_cleanup (device=0x5555556b0020) at src/device.c:611
#2  gatt_server_cleanup (device=0x5555556b0020) at src/device.c:606
#3  attio_cleanup (device=0x5555556b0020) at src/device.c:631
#4  0x0000555555606f8d in queue_foreach (queue=0x5555556c1b80, function=function@entry=0x55555560b140 <disconn_handler>, 
    user_data=0x68) at src/shared/queue.c:220
#5  0x000055555560d4aa in disconnect_cb (io=<optimized out>, user_data=0x5555556c2320) at src/shared/att.c:664
#6  0x0000555555618679 in watch_callback (channel=<optimized out>, cond=<optimized out>, user_data=<optimized out>)
    at src/shared/io-glib.c:170
#7  0x00007ffff7ecd3df in g_main_context_dispatch () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#8  0x00007ffff7ecd788 in ?? () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#9  0x00007ffff7ecdaa3 in g_main_loop_run () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#10 0x0000555555618cc9 in mainloop_run () at src/shared/mainloop-glib.c:79
#11 0x0000555555619130 in mainloop_run_with_signal (func=<optimized out>, user_data=0x0) at src/shared/mainloop-notify.c:201
#12 0x0000555555576779 in main (argc=<optimized out>, argv=<optimized out>) at src/main.c:971
(gdb) info args 
database = 0x555555690b20
device = 0x5555556b0020
(gdb) info locals 
server = <optimized out>
att = <optimized out>
state = <optimized out>
addr = <optimized out>
type = <optimized out>
__func__ = "btd_gatt_database_att_disconnected"
(gdb) c
Continuing.
```
4. SIGSEGV when disconnect callback is called (pointer: **0x555555697be0**)
```c
Program received signal SIGSEGV, Segmentation fault.
att_disconnected (err=<optimized out>, user_data=0x555555697be0) at src/gatt-database.c:331
331		device = btd_adapter_find_device(state->db->adapter, &state->bdaddr,
(gdb) bt
#0  att_disconnected (err=<optimized out>, user_data=0x555555697be0) at src/gatt-database.c:331
#1  0x0000555555606f8d in queue_foreach (queue=0x5555556c1b80, function=function@entry=0x55555560b140 <disconn_handler>, 
    user_data=0x68) at src/shared/queue.c:220
#2  0x000055555560d4aa in disconnect_cb (io=<optimized out>, user_data=0x5555556c2320) at src/shared/att.c:664
#3  0x0000555555618679 in watch_callback (channel=<optimized out>, cond=<optimized out>, user_data=<optimized out>)
    at src/shared/io-glib.c:170
#4  0x00007ffff7ecd3df in g_main_context_dispatch () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff7ecd788 in ?? () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#6  0x00007ffff7ecdaa3 in g_main_loop_run () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#7  0x0000555555618cc9 in mainloop_run () at src/shared/mainloop-glib.c:79
#8  0x0000555555619130 in mainloop_run_with_signal (func=<optimized out>, user_data=0x0) at src/shared/mainloop-notify.c:201
#9  0x0000555555576779 in main (argc=<optimized out>, argv=<optimized out>) at src/main.c:971
(gdb) info args 
err = <optimized out>
user_data = 0x555555697be0
(gdb) info locals 
state = 0x555555697be0
device = <optimized out>
__func__ = "att_disconnected"
(gdb) print *state
$1 = {db = 0x2f69612f0000001b, bdaddr = {b = "conduc"}, bdaddr_type = 116 't', disc_id = 0, cli_feat = "e", change_aware = 122, 
  out_of_sync = false, ccc_states = 0x30656369767265, pending = 0x0}

```
Note: `*state` contains corrupted data probably because it has been released

Vudentz, 2021-01-19:
@zxtwonder can you give the above patch a try?

zxtwonder, 2021-01-20:
@Vudentz Thanks for the fix. I looked at the change and it should fix the issue. I will need to revert some changes to test it out.

On the other hand, why don't we clean up disconnect callback when device_state is released like below? The current approach releases device_state-related resources in separate places and makes it harder to maintain correctness.
```c
static void device_state_free(void *data)
{
	struct device_state *state = data;

+	if (state->disc_id)
+		bt_att_unregister_disconnect(att, state->disc_id);

	queue_destroy(state->ccc_states, free);

	if (state->pending) {
		free(state->pending->value);
		free(state->pending);
	}

	free(state);
}
```

Vudentz, 2021-01-20:
> @Vudentz Thanks for the fix. I looked at the change and it should fix the issue. I will need to revert some changes to test it out.
>
> On the other hand, why don't we clean up disconnect callback when [...]

We don't have access to the att instance here and it would probably be more complicated to add an att instance to device_state because then we need to make sure it is cleanup properly when we disconnect, etc.

> 	queue_destroy(state->ccc_states, free);
> 
> 	if (state->pending) {
> 		free(state->pending->value);
> 		free(state->pending);
> 	}
> 
> 	free(state);
> }
> ```

zxtwonder, 2021-01-20:
Fair enough and sorry for not realizing that there is no att instance there...

closed 2021-01-21 by Vudentz