bluez/bluez #83: Blackberry 10 can't connect as media source

WTFThatPerson, 2021-01-16:
I try to use Raspberry Pi as bluetooth media sink and while Android connects fine, BB 10 device can't connect it's AVDTP part of connection: there are 3 tries to connect, media play of menu BB 10 device blinks for a moment "connected" then returns to "disconnected" state; AVCTP part of connection work fine though, so I end up with BB 10 as AV control device only
`uname -r
5.4.83-4-ARCH
journalctl -f -u bluetooth
bluetoothd[199376]: src/adapter.c:connected_callback() hci0 device 94:EB:CD:0E:F5:91 connected eir_len 22
bluetoothd[199376]: profiles/audio/a2dp.c:confirm_cb() AVDTP: incoming connect from 94:EB:CD:0E:F5:91
bluetoothd[199376]: src/agent.c:agent_ref() 0x557a72c850: ref=2
bluetoothd[199376]: src/agent.c:agent_call_authorize_service() authorize service request was sent for /org/bluez/hci0/dev_94_EB_CD_0E_F5_91
bluetoothd[199376]: src/agent.c:agent_ref() 0x557a72c850: ref=3
bluetoothd[199376]: src/agent.c:agent_unref() 0x557a72c850: ref=2
bluetoothd[199376]: src/agent.c:agent_unref() 0x557a72c850: ref=1
bluetoothd[199376]: profiles/audio/source.c:source_set_state() State changed /org/bluez/hci0/dev_94_EB_CD_0E_F5_91: SOURCE_STATE_DISCONNECTED -> SOURCE_STATE_CONNECTING
bluetoothd[199376]: profiles/audio/avdtp.c:avdtp_connect_cb() AVDTP: connected signaling channel to 94:EB:CD:0E:F5:91
bluetoothd[199376]: profiles/audio/avdtp.c:avdtp_connect_cb() AVDTP imtu=672, omtu=1691
bluetoothd[199376]: profiles/audio/avdtp.c:avdtp_ref() 0x557a746be0: ref=1
bluetoothd[199376]: profiles/audio/avdtp.c:set_disconnect_timer() timeout 1
bluetoothd[199376]: profiles/audio/avdtp.c:session_cb()
bluetoothd[199376]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received DISCOVER_CMD
bluetoothd[199376]: src/adapter.c:dev_disconnected() Device 94:EB:CD:0E:F5:91 disconnected, reason 2
bluetoothd[199376]: src/adapter.c:adapter_remove_connection()
bluetoothd[199376]: plugins/policy.c:disconnect_cb() reason 2
bluetoothd[199376]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 94:EB:CD:0E:F5:91 type 0 status 0xe
bluetoothd[199376]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
bluetoothd[199376]: src/device.c:device_bonding_failed() status 14
bluetoothd[199376]: src/adapter.c:resume_discovery()
bluetoothd[199376]: profiles/audio/avdtp.c:session_cb()
bluetoothd[199376]: profiles/audio/avdtp.c:avdtp_ref() 0x557a746be0: ref=2
bluetoothd[199376]: profiles/audio/avdtp.c:avdtp_unref() 0x557a746be0: ref=1
bluetoothd[199376]: profiles/audio/avdtp.c:connection_lost() Disconnected from 94:EB:CD:0E:F5:91
bluetoothd[199376]: profiles/audio/source.c:source_set_state() State changed /org/bluez/hci0/dev_94_EB_CD_0E_F5_91: SOURCE_STATE_CONNECTING -> SOURCE_STATE_DISCONNECTED
bluetoothd[199376]: profiles/audio/a2dp.c:channel_remove() chan 0x557a747d60
bluetoothd[199376]: profiles/audio/avdtp.c:avdtp_unref() 0x557a746be0: ref=0
bluetoothd[199376]: profiles/audio/avdtp.c:avdtp_free() 0x557a746be0
bluetoothd[199376]: src/adapter.c:connected_callback() hci0 device 94:EB:CD:0E:F5:91 connected eir_len 22
bluetoothd[199376]: profiles/audio/avctp.c:avctp_confirm_cb() AVCTP: incoming connect from 94:EB:CD:0E:F5:91
bluetoothd[199376]: profiles/audio/avctp.c:avctp_set_state() AVCTP Connecting
bluetoothd[199376]: src/agent.c:agent_ref() 0x557a72c850: ref=2
bluetoothd[199376]: src/agent.c:agent_call_authorize_service() authorize service request was sent for /org/bluez/hci0/dev_94_EB_CD_0E_F5_91
bluetoothd[199376]: src/agent.c:agent_ref() 0x557a72c850: ref=3
bluetoothd[199376]: src/agent.c:agent_unref() 0x557a72c850: ref=2
bluetoothd[199376]: src/agent.c:agent_unref() 0x557a72c850: ref=1
bluetoothd[199376]: profiles/audio/avctp.c:avctp_connect_cb() AVCTP: connected to 94:EB:CD:0E:F5:91
bluetoothd[199376]: profiles/audio/avctp.c:init_uinput() AVRCP: uinput initialized for BLACKBERRY-9D1E
bluetoothd[199376]: profiles/audio/avrcp.c:controller_init() 0x557a745220 version 0x0103
bluetoothd[199376]: src/service.c:change_state() 0x557a71ee90: device 94:EB:CD:0E:F5:91 profile audio-avrcp-target state changed: disconnected -> connected (0)
bluetoothd[199376]: profiles/audio/player.c:media_player_controller_create() /org/bluez/hci0/dev_94_EB_CD_0E_F5_91/player0
bluetoothd[199376]: profiles/audio/avrcp.c:target_init() 0x557a7409a0 version 0x0103
bluetoothd[199376]: src/service.c:change_state() 0x557a722910: device 94:EB:CD:0E:F5:91 profile avrcp-controller state changed: disconnected -> connected (0)
bluetoothd[199376]: profiles/audio/avctp.c:avctp_set_state() AVCTP Connected
bluetoothd[199376]: profiles/audio/player.c:media_player_set_position() 4294967295
bluetoothd[199376]: profiles/audio/avrcp.c:handle_vendordep_pdu() AVRCP PDU 0x10, company 0x001958 len 0x0001
bluetoothd[199376]: profiles/audio/avrcp.c:avrcp_handle_get_capabilities() id=3
bluetoothd[199376]: profiles/audio/player.c:media_player_set_status() stopped
bluetoothd[199376]: profiles/audio/player.c:media_player_set_playlist_item() 0
bluetoothd[199376]: profiles/audio/player.c:media_player_set_metadata() Title:
bluetoothd[199376]: profiles/audio/player.c:media_player_set_metadata() Artist:
bluetoothd[199376]: profiles/audio/player.c:media_player_set_metadata() Album:
bluetoothd[199376]: profiles/audio/player.c:media_player_set_metadata() Duration:
bluetoothd[199376]: profiles/audio/player.c:media_player_set_duration() 4294967295
bluetoothd[199376]: profiles/audio/player.c:media_player_set_position() 4294967295
bluetoothd[199376]: profiles/audio/player.c:media_player_set_status() stopped
bluetoothd[199376]: profiles/audio/player.c:media_player_set_duration() 4294967295
bluetoothd[199376]: profiles/audio/player.c:media_player_set_position() 4294967295
bluetoothd[199376]: profiles/audio/player.c:media_player_set_status() stopped
sudo btmon
Bluetooth monitor ver 5.55
= Note: Linux version 5.4.83-4-ARCH (aarch64)                          0.501117
= Note: Bluetooth subsystem version 2.22                               0.501124
= New Index: DC:A6:32:36:7A:8D (Primary,UART,hci0)              [hci0] 0.501126
= Open Index: DC:A6:32:36:7A:8D                                 [hci0] 0.501127
[... 934 lines cut ...]
          AVRCP: GetPlayStatus pt Single len 0x0000
> HCI Event: Number of Completed Packets (0x13) plen 5   #340 [hci0] 101.131478
        Num handles: 1
        Handle: 12
        Count: 2
> ACL Data RX: Handle 12 flags 0x02 dlen 26              #341 [hci0] 101.135136
      Channel: 64 len 22 [PSM 23 mode Basic (0x00)] {chan 2}
      AVCTP Control: Response: type 0x00 label 6 PID 0x110e
        AV/C: Stable: address 0x48 opcode 0x00
          Subunit: Panel
          Opcode: Vendor Dependent
          Company ID: 0x001958
          AVRCP: GetPlayStatus pt Single len 0x0009
            SongLength: 0xffffffff (4294967295 miliseconds)
            SongPosition: 0xffffffff (4294967295 miliseconds)
            PlayStatus: 0x00 (STOPPED)
> HCI Event: Mode Change (0x14) plen 6                   #342 [hci0] 102.416537
        Status: Success (0x00)
        Handle: 12
        Mode: Sniff (0x02)
        Interval: 400.000 msec (0x0280)
`
Since, as mentioned above, there are 3 identical tries to connect, I cutted two of them.
When I try to compare btmon log of successful Android connection, everything look more or less similar until start authentication pending: strange things begin around #257 in btmon log above: while Android tries to get AVPTP capabilities, BB 10 in turn, for some uknown reason, tries to disable encryption and read clock offset then Raspberry ends AVDTP connection with "Authentication Failure" message.
Can anyone suggest how to repair those things?

WTFThatPerson, 2021-01-16:
I try to use Raspberry Pi as bluetooth media sink and while Android connects fine, BB 10 device can't connect it's AVDTP part of connection: there are 3 tries to connect, media play of menu BB 10 device blinks for a moment "connected" then returns to "disconnected" state; AVCTP part of connection work fine though, so I end up with BB 10 as AV control device only

```
uname -r
5.4.83-4-ARCH
journalctl -f -u bluetooth
bluetoothd[199376]: src/adapter.c:connected_callback() hci0 device 94:EB:CD:0E:F5:91 connected eir_len 22
bluetoothd[199376]: profiles/audio/a2dp.c:confirm_cb() AVDTP: incoming connect from 94:EB:CD:0E:F5:91
bluetoothd[199376]: src/agent.c:agent_ref() 0x557a72c850: ref=2
bluetoothd[199376]: src/agent.c:agent_call_authorize_service() authorize service request was sent for /org/bluez/hci0/dev_94_EB_CD_0E_F5_91
bluetoothd[199376]: src/agent.c:agent_ref() 0x557a72c850: ref=3
bluetoothd[199376]: src/agent.c:agent_unref() 0x557a72c850: ref=2
bluetoothd[199376]: src/agent.c:agent_unref() 0x557a72c850: ref=1
bluetoothd[199376]: profiles/audio/source.c:source_set_state() State changed /org/bluez/hci0/dev_94_EB_CD_0E_F5_91: SOURCE_STATE_DISCONNECTED -> SOURCE_STATE_CONNECTING
bluetoothd[199376]: profiles/audio/avdtp.c:avdtp_connect_cb() AVDTP: connected signaling channel to 94:EB:CD:0E:F5:91
bluetoothd[199376]: profiles/audio/avdtp.c:avdtp_connect_cb() AVDTP imtu=672, omtu=1691
bluetoothd[199376]: profiles/audio/avdtp.c:avdtp_ref() 0x557a746be0: ref=1
bluetoothd[199376]: profiles/audio/avdtp.c:set_disconnect_timer() timeout 1
bluetoothd[199376]: profiles/audio/avdtp.c:session_cb()
bluetoothd[199376]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received DISCOVER_CMD
bluetoothd[199376]: src/adapter.c:dev_disconnected() Device 94:EB:CD:0E:F5:91 disconnected, reason 2
bluetoothd[199376]: src/adapter.c:adapter_remove_connection()
bluetoothd[199376]: plugins/policy.c:disconnect_cb() reason 2
bluetoothd[199376]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 94:EB:CD:0E:F5:91 type 0 status 0xe
bluetoothd[199376]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
bluetoothd[199376]: src/device.c:device_bonding_failed() status 14
bluetoothd[199376]: src/adapter.c:resume_discovery()
bluetoothd[199376]: profiles/audio/avdtp.c:session_cb()
bluetoothd[199376]: profiles/audio/avdtp.c:avdtp_ref() 0x557a746be0: ref=2
bluetoothd[199376]: profiles/audio/avdtp.c:avdtp_unref() 0x557a746be0: ref=1
bluetoothd[199376]: profiles/audio/avdtp.c:connection_lost() Disconnected from 94:EB:CD:0E:F5:91
bluetoothd[199376]: profiles/audio/source.c:source_set_state() State changed /org/bluez/hci0/dev_94_EB_CD_0E_F5_91: SOURCE_STATE_CONNECTING -> SOURCE_STATE_DISCONNECTED
bluetoothd[199376]: profiles/audio/a2dp.c:channel_remove() chan 0x557a747d60
bluetoothd[199376]: profiles/audio/avdtp.c:avdtp_unref() 0x557a746be0: ref=0
bluetoothd[199376]: profiles/audio/avdtp.c:avdtp_free() 0x557a746be0
bluetoothd[199376]: src/adapter.c:connected_callback() hci0 device 94:EB:CD:0E:F5:91 connected eir_len 22
bluetoothd[199376]: profiles/audio/avctp.c:avctp_confirm_cb() AVCTP: incoming connect from 94:EB:CD:0E:F5:91
bluetoothd[199376]: profiles/audio/avctp.c:avctp_set_state() AVCTP Connecting
bluetoothd[199376]: src/agent.c:agent_ref() 0x557a72c850: ref=2
bluetoothd[199376]: src/agent.c:agent_call_authorize_service() authorize service request was sent for /org/bluez/hci0/dev_94_EB_CD_0E_F5_91
bluetoothd[199376]: src/agent.c:agent_ref() 0x557a72c850: ref=3
bluetoothd[199376]: src/agent.c:agent_unref() 0x557a72c850: ref=2
bluetoothd[199376]: src/agent.c:agent_unref() 0x557a72c850: ref=1
bluetoothd[199376]: profiles/audio/avctp.c:avctp_connect_cb() AVCTP: connected to 94:EB:CD:0E:F5:91
bluetoothd[199376]: profiles/audio/avctp.c:init_uinput() AVRCP: uinput initialized for BLACKBERRY-9D1E
bluetoothd[199376]: profiles/audio/avrcp.c:controller_init() 0x557a745220 version 0x0103
bluetoothd[199376]: src/service.c:change_state() 0x557a71ee90: device 94:EB:CD:0E:F5:91 profile audio-avrcp-target state changed: disconnected -> connected (0)
bluetoothd[199376]: profiles/audio/player.c:media_player_controller_create() /org/bluez/hci0/dev_94_EB_CD_0E_F5_91/player0
bluetoothd[199376]: profiles/audio/avrcp.c:target_init() 0x557a7409a0 version 0x0103
bluetoothd[199376]: src/service.c:change_state() 0x557a722910: device 94:EB:CD:0E:F5:91 profile avrcp-controller state changed: disconnected -> connected (0)
bluetoothd[199376]: profiles/audio/avctp.c:avctp_set_state() AVCTP Connected
bluetoothd[199376]: profiles/audio/player.c:media_player_set_position() 4294967295
bluetoothd[199376]: profiles/audio/avrcp.c:handle_vendordep_pdu() AVRCP PDU 0x10, company 0x001958 len 0x0001
bluetoothd[199376]: profiles/audio/avrcp.c:avrcp_handle_get_capabilities() id=3
bluetoothd[199376]: profiles/audio/player.c:media_player_set_status() stopped
bluetoothd[199376]: profiles/audio/player.c:media_player_set_playlist_item() 0
bluetoothd[199376]: profiles/audio/player.c:media_player_set_metadata() Title:
bluetoothd[199376]: profiles/audio/player.c:media_player_set_metadata() Artist:
bluetoothd[199376]: profiles/audio/player.c:media_player_set_metadata() Album:
bluetoothd[199376]: profiles/audio/player.c:media_player_set_metadata() Duration:
bluetoothd[199376]: profiles/audio/player.c:media_player_set_duration() 4294967295
bluetoothd[199376]: profiles/audio/player.c:media_player_set_position() 4294967295
[... cut ...]
    Since, as mentioned above, there are 3 identical tries to connect, I cutted two of them.
    When I try to compare btmon log of successful Android connection, everything look more or less similar until start authentication pending: strange things begin around #257 in btmon log above: while Android tries to get AVPTP capabilities, BB 10 in turn, for some uknown reason, tries to disable encryption and read clock offset then Raspberry ends AVDTP connection with "Authentication Failure" message.
    Can anyone suggest how to repair those things?

Vudentz, 2021-01-18:
bluetoothd[199376]: src/agent.c:agent_call_authorize_service() authorize service request was sent for /org/bluez/hci0/dev_94_EB_CD_0E_F5_91

It looks like the agent never responds to this request, alternatively you can mark the device as trusted to bypass authorization.

WTFThatPerson, 2021-01-20:
Things are not so easy unfortunately, I'm afraid. Trusting will do nothing, because problem, as I said, isn't as simple as it seems. BB 10 doesn't refuse to connect at all, it refuses connect as media source only.  So, if I'll trust it, I do it for media controller. Once more, it starts connect, I have request for authorize A2DP service with UUID 0000110d.., prompt of bluetoothctl changes to BB 10 device, but after my answer "yes" connection resets, prompt goes back to "bluetooth" and everything goes cycling 3 times until media controller request with  UUID 0000110e.., which one is processed successfully.

WTFThatPerson, 2021-01-22:
Well, I marked device as trusted - still no joy, as I suspected: all the same three unsuccessful tries for mediastream followed by success AV controller connect are here, but without advertize part, of course. I put now full log from start of bluetooth service.
Timestamps on bluetooth log are:
19:35:51 bluetooth service start
19:36:02 bluealsa service start
19:41:48 unsuccsessful tries to start media stream service
19:41:53 successful start of media controller service
journalctl -f -u bluetooth
```
Jan 22 19:35:51 Mediaserver systemd[1]: Starting Bluetooth service...
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: Bluetooth daemon 5.55
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_config() parsing /etc/bluetooth/main.conf
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_config() Key file does not have key вЂњDiscoverableTimeoutвЂќ in group вЂњGeneralвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_config() Key file does not have key вЂњAlwaysPairableвЂќ in group вЂњGeneralвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_config() Key file does not have key вЂњPairableTimeoutвЂќ in group вЂњGeneralвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_config() Key file does not have key вЂњPrivacyвЂќ in group вЂњGeneralвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_config() Key file does not have key вЂњJustWorksRepairingвЂќ in group вЂњGeneralвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_config() Key file does not have key вЂњTemporaryTimeoutвЂќ in group вЂњGeneralвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_config() Key file does not have key вЂњNameвЂќ in group вЂњGeneralвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_config() class=0x240408
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_config() Key file does not have key вЂњDeviceIDвЂќ in group вЂњGeneralвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_config() Key file does not have key вЂњReverseServiceDiscoveryвЂќ in group вЂњGeneralвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_config() Key file does not have key вЂњCacheвЂќ in group вЂњGATTвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_config() Key file does not have key вЂњKeySizeвЂќ in group вЂњGATTвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_config() Key file does not have key вЂњExchangeMTUвЂќ in group вЂњGATTвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_config() Key file does not have key вЂњChannelsвЂќ in group вЂњGATTвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_controller_config() Key file does not have key вЂњBRPageScanTypeвЂќ in group вЂњControllerвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_controller_config() Key file does not have key вЂњBRPageScanIntervalвЂќ in group вЂњControllerвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_controller_config() Key file does not have key вЂњBRPageScanWindowвЂќ in group вЂњControllerвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_controller_config() Key file does not have key вЂњBRInquiryScanTypeвЂќ in group вЂњControllerвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_controller_config() Key file does not have key вЂњBRInquiryScanIntervalвЂќ in group вЂњControllerвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_controller_config() Key file does not have key вЂњBRInquiryScanWindowвЂќ in group вЂњControllerвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_controller_config() Key file does not have key вЂњBRLinkSupervisionTimeoutвЂќ in group вЂњControllerвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_controller_config() Key file does not have key вЂњBRPageTimeoutвЂќ in group вЂњControllerвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_controller_config() Key file does not have key вЂњBRMinSniffIntervalвЂќ in group вЂњControllerвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_controller_config() Key file does not have key вЂњBRMaxSniffIntervalвЂќ in group вЂњControllerвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_controller_config() Key file does not have key вЂњLEMinAdvertisementIntervalвЂќ in group вЂњControllerвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_controller_config() Key file does not have key вЂњLEMaxAdvertisementIntervalвЂќ in group вЂњControllerвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_controller_config() Key file does not have key вЂњLEMultiAdvertisementRotationIntervalвЂќ in group вЂњControllerвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_controller_config() Key file does not have key вЂњLEScanIntervalAutoConnectвЂќ in group вЂњControllerвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_controller_config() Key file does not have key вЂњLEScanWindowAutoConnectвЂќ in group вЂњControllerвЂќ
Jan 22 19:35:51 Mediaserver bluetoothd[164609]: src/main.c:parse_controller_config() Key file does not have key вЂњLEScanIntervalSuspendвЂќ in group вЂњControllerвЂќ
[... cut ...]
            SongLength: 0xffffffff (4294967295 miliseconds)
            SongPosition: 0xffffffff (4294967295 miliseconds)
            PlayStatus: 0x00 (STOPPED)
> HCI Event: Number of Completed Packets (0x13) plen 5   #222 [hci0] 265.793838
        Num handles: 1
        Handle: 12
        Count: 1
> HCI Event: Mode Change (0x14) plen 6                   #223 [hci0] 267.090182
        Status: Success (0x00)
        Handle: 12
        Mode: Sniff (0x02)
        Interval: 400.000 msec (0x0280)
```
Any suggestions, please?

WTFThatPerson, 2021-01-23:
I tried to look closer to differences between Blackberry and Android communications with Raspberry Pi: everything looking very similar until "Authorization pending" status and "Connection successful - No further information available" then (#35 in btmon log from my **first** post, not from previous). Problems begin after L2CAP configure request with ident 3 (#36):

Android pauses, as per my understanding, appropriate response, and at next step asks for MTU size 672, Raspberry acknowledges that MTU size,  restores paused request with ident 3 and sends responce "successful" for it. I see HCI event  "QoS setup complete" after that, then Android asks for ACP SEIDs, receives answer and at next step tries get AVDTP capabilities.

Blackberry, in turn, pauses that response with ident 3, asks for MTU with different size of 1691, Raspberry acknowledges that MTU size,  restores paused request with ident 3 and sends response "successful" for it, but it includes for some unknown reason new MTU size 672 in that response (## 38-40 in above mentioned log). There isn't Qos HCI event in log, but Blackberry skips to ACP SEIDs request, receives answer (#42) and stops any activity then until timeout and break coonection.

Since a lack of knowledge, maybe I messed something, but that's a way I see all those things.

WTFThatPerson, 2021-01-28:
Another one difference between successful connection to Android device and unsuccessful to Blackberry one are CIDs during establishing connection: Android use different source CID 65 and destination CID 64 while Blackberry uses the same CID 64 for both. Does is matters?

WTFThatPerson, 2021-02-28:
So, problem lays somewhere inside SSP - for some uknown reason it generates two authentication requests while connects to Blackberry: the first one I can see and answer, and the second one directed no nowhere, so things end up with auth error for AVDTP profile, meanwhile the first one is accepted and lets smartphone connect using AVCTP profile. I tried a whole bunch of different agents which have a whole bunch of different weirdnesses with the very same result: Blackberry won't connect to AVDTP profile and connects to AVCTP one no matter what agent and what pairing mode is used.  And no, trusting devices doesn't help, it seems work for one, I mean AVCTP part, but does nothing for AVDTP profile. Only way to overcome that problem is turn off SSP mode; after input 
sudo btmgmt ssp off
second request appears and Blackberry successfully connects using AVDTP profile. I'm not too happy with such workaround, keeping in mind I should input all that staff after every power cycle of adapter
So does anyone have different solution?

WTFThatPerson, 2021-03-28:
At the end, I believe it's a matter of Blackberry connect protocol and problems of storing session parameters inside 
`avdtp_parse_cmd`
loop. 
As per my understanding, remote device initiates only audio streaming part of connect; **avrcp** part is proceeding by plugin.
Blackberry in turn sends **avctp** request explicitly and it appears in timing terms inside above mentioned loop. It looks like proceeding that request breaks loop and whole **avdtp** connection then. At least, after inserting pause inside
`avctp_confirm_cb`
things start improve. One way or another I don't think that time dependency of extra connect is right thing.