bluez/bluez #84: [Bluez 5.55]: AVRCP events not coming-in

macmpi, 2021-01-18:
Using Bluez 5.55 on Alpine 3.13 (linux 5.10.7 on PiZeroW)
Headset connected to Pi receives audio fine (bluez-alsa) but AVRCP commands are not received in _uinput_.
_btmon_ shows them flowing-in, but _evtest_ stays silent.

Doesn´t that line ring a bell:
`Input device ID: bus 0x5 vendor 0x0 product 0x0 version 0x0`
Wouldn´t it be that device is erroneously identified/filtered-out? Tried with other headsets with same result.

Same setup was working fine on Alpine 3.11 with bluez 5.52

Thanks for any hint.

```
~$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Mitsumi Electric Apple USB Keyboard
/dev/input/event1:	Logitech M4848
/dev/input/event2:	MacALLY USB Two Button Mini Mouse
/dev/input/event3:	Vivan  BT600 (AVRCP)
Select the device event number [0-3]: 3
Input driver version is 1.0.1
Input device ID: bus 0x5 vendor 0x0 product 0x0 version 0x0
Input device name: "Vivan  BT600 (AVRCP)"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 2 (KEY_1)
    Event code 3 (KEY_2)
    Event code 4 (KEY_3)
    Event code 5 (KEY_4)
    Event code 6 (KEY_5)
    Event code 7 (KEY_6)
    Event code 8 (KEY_7)
    Event code 9 (KEY_8)
    Event code 10 (KEY_9)
    Event code 11 (KEY_0)
    Event code 28 (KEY_ENTER)
    Event code 52 (KEY_DOT)
    Event code 59 (KEY_F1)
    Event code 60 (KEY_F2)
    Event code 61 (KEY_F3)
    Event code 62 (KEY_F4)
    Event code 63 (KEY_F5)
    Event code 64 (KEY_F6)
    Event code 65 (KEY_F7)
    Event code 66 (KEY_F8)
    Event code 67 (KEY_F9)
    Event code 103 (KEY_UP)
    Event code 105 (KEY_LEFT)
    Event code 106 (KEY_RIGHT)
    Event code 108 (KEY_DOWN)
    Event code 113 (KEY_MUTE)
    Event code 114 (KEY_VOLUMEDOWN)
    Event code 115 (KEY_VOLUMEUP)
    Event code 138 (KEY_HELP)
    Event code 139 (KEY_MENU)
    Event code 163 (KEY_NEXTSONG)
    Event code 165 (KEY_PREVIOUSSONG)
    Event code 166 (KEY_STOPCD)
    Event code 167 (KEY_RECORD)
    Event code 168 (KEY_REWIND)
    Event code 171 (KEY_CONFIG)
    Event code 174 (KEY_EXIT)
    Event code 200 (KEY_PLAYCD)
    Event code 201 (KEY_PAUSECD)
    Event code 208 (KEY_FASTFORWARD)
    Event code 353 (KEY_SELECT)
    Event code 356 (KEY_POWER2)
    Event code 358 (KEY_INFO)
    Event code 362 (KEY_PROGRAM)
    Event code 364 (KEY_FAVORITES)
    Event code 395 (KEY_LIST)
    Event code 398 (KEY_RED)
    Event code 399 (KEY_GREEN)
    Event code 400 (KEY_YELLOW)
    Event code 401 (KEY_BLUE)
    Event code 402 (KEY_CHANNELUP)
    Event code 403 (KEY_CHANNELDOWN)
    Event code 405 (KEY_LAST)
  Event type 2 (EV_REL)
Key repeat handling:
  Repeat type 20 (EV_REP)
    Repeat code 0 (REP_DELAY)
      Value    250
    Repeat code 1 (REP_PERIOD)
      Value     33
Properties:
Testing ... (interrupt to exit)

```

Vudentz, 2021-01-18:
Are there any errors on the bluetoothd logs?  If we are able to send it over uinput then it is probably kernel related.

macmpi, 2021-01-18:
Thanks for feedback.
I did not see any bluetooth error in var/log/messages.
Not sure it indeed reaches _uinput_ as _evtest_ remains silent: how can this be checked?

Here is what I get at keypress from btmon:
```
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                   #197 [hci0] 78.000000
        Num handles: 1
        Handle: 12
        Count: 2
> ACL Data RX: Handle 12 flags 0x02 dlen 12                                                                                              #198 [hci0] 88.000000
      Channel: 67 len 8 [PSM 23 mode Basic (0x00)] {chan 3}
      AVCTP Control: Command: type 0x00 label 0 PID 0x110e
        AV/C: Control: address 0x48 opcode 0x7c
          Subunit: Panel
          Opcode: Passthrough
          Operation: 0x44 (PLAY Pressed)
          Length: 0x00
< ACL Data TX: Handle 12 flags 0x00 dlen 12                                                                                              #199 [hci0] 88.000000
      Channel: 68 len 8 [PSM 23 mode Basic (0x00)] {chan 3}
      AVCTP Control: Response: type 0x00 label 0 PID 0x110e
        AV/C: Accepted: address 0x48 opcode 0x7c
          Subunit: Panel
          Opcode: Passthrough
          Operation: 0x44 (PLAY Pressed)
          Length: 0x00
> ACL Data RX: Handle 12 flags 0x02 dlen 12                                                                                              #200 [hci0] 88.000000
      Channel: 67 len 8 [PSM 23 mode Basic (0x00)] {chan 3}
      AVCTP Control: Command: type 0x00 label 1 PID 0x110e
        AV/C: Control: address 0x48 opcode 0x7c
          Subunit: Panel
          Opcode: Passthrough
          Operation: 0xc4 (PLAY Released)
          Length: 0x00
< ACL Data TX: Handle 12 flags 0x00 dlen 12                                                                                              #201 [hci0] 88.000000
      Channel: 68 len 8 [PSM 23 mode Basic (0x00)] {chan 3}
      AVCTP Control: Response: type 0x00 label 1 PID 0x110e
        AV/C: Accepted: address 0x48 opcode 0x7c
          Subunit: Panel
          Opcode: Passthrough
          Operation: 0xc4 (PLAY Released)
          Length: 0x00
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                   #202 [hci0] 88.000000
        Num handles: 1
        Handle: 12
        Count: 2
> HCI Event: Mode Change (0x14) plen 6                                                                                                   #203 [hci0] 98.000000
        Status: Success (0x00)
        Handle: 12
        Mode: Sniff (0x02)
        Interval: 250.000 msec (0x0190)
```

Vudentz, 2021-01-20:
You need to be running btmon before connecting the device, otherwise btmon cannot resolve the PSM of the L2CAP connection which is why you just see the hexdump.

Looks at the changes the only thing we change was the bitmask of AVC/AVRCP:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=a078a364768cf62b92c4ffaa89647dd49282981a

Vudentz, 2021-01-20:
I tried with a sony headset I have here and it seems to be working fine:

`Event: time 1611178682.185211, type 1 (EV_KEY), code 200 (KEY_PLAYCD), value 1
Event: time 1611178682.185211, -------------- SYN_REPORT ------------
Event: time 1611178682.194691, type 1 (EV_KEY), code 200 (KEY_PLAYCD), value 0
Event: time 1611178682.194691, -------------- SYN_REPORT ------------
Event: time 1611178685.676321, type 1 (EV_KEY), code 165 (KEY_PREVIOUSSONG), value 1
Event: time 1611178685.676321, -------------- SYN_REPORT ------------
Event: time 1611178685.697508, type 1 (EV_KEY), code 165 (KEY_PREVIOUSSONG), value 0
Event: time 1611178685.697508, -------------- SYN_REPORT ------------
Event: time 1611178687.992797, type 1 (EV_KEY), code 163 (KEY_NEXTSONG), value 1
Event: time 1611178687.992797, -------------- SYN_REPORT ------------
Event: time 1611178688.013834, type 1 (EV_KEY), code 163 (KEY_NEXTSONG), value 0
Event: time 1611178688.013834, -------------- SYN_REPORT ------------
Event: time 1611178689.434883, type 1 (EV_KEY), code 201 (KEY_PAUSECD), value 1
Event: time 1611178689.434883, -------------- SYN_REPORT ------------
Event: time 1611178689.456384, type 1 (EV_KEY), code 201 (KEY_PAUSECD), value 0
Event: time 1611178689.456384, -------------- SYN_REPORT ------------
Event: time 1611178696.913184, type 1 (EV_KEY), code 200 (KEY_PLAYCD), value 1
Event: time 1611178696.913184, -------------- SYN_REPORT ------------
Event: time 1611178696.936221, type 1 (EV_KEY), code 200 (KEY_PLAYCD), value 0
Event: time 1611178696.936221, -------------- SYN_REPORT ------------
Event: time 1611178728.701315, type 1 (EV_KEY), code 201 (KEY_PAUSECD), value 1
Event: time 1611178728.701315, -------------- SYN_REPORT ------------
Event: time 1611178728.716422, type 1 (EV_KEY), code 201 (KEY_PAUSECD), value 0
Event: time 1611178728.716422, -------------- SYN_REPORT ------------
`

macmpi, 2021-01-20:
> I tried with a sony headset I have here and it seems to be working fine
oh good, thanks for checking: is it on Alpine 3.13 too (or 5.10.7 kernel), by chance?
I wonder what may cause that, as Alpine 3.11/Bluez 5.52/kernel 5.4.84 was fine...

Vudentz, 2021-01-21:
> > I tried with a sony headset I have here and it seems to be working fine
> 
> oh good, thanks for checking: is it on Alpine 3.13 too (or 5.10.7 kernel), by chance?
> I wonder what may cause that, as Alpine 3.11/Bluez 5.52/kernel 5.4.84 was fine...

Nope, I was running on FC31, I believe you have a device specific problem here, maybe it needs a dedicated dirver to handle the event and that is not being probed properly because pid/vid are set to 0, now I wonder how it would have worked before but perhaps the problem is that we are not actually fetching the pid/vid properly but Ive looked at the code and it doesn't seem we have changed that in a while.

macmpi, 2021-01-30:
Thanks.
It seems some modules are currently missing in recent Alpine releases, which probably is the root cause of this issue.
Closing for now.

macmpi, 2021-05-29:
Still struggling at this since 5.55; and 5.58 is same.
May be Pi specific with built-in cypress/brcm module, as the issue is not present on Alpine 3.13 x86-64 with my onboard bluetooth?

@gearhead, do you manage to get media key support from paired headset keys, and see events flowing in with `evtest`?
I see you are on Arch Linux, which may not be that far from my AlpineLinux setup...
(also have [pairing issues](https://github.com/bluez/bluez/issues/55) BTW)

gearhead, 2021-05-29:
It appears you are correct. I can now pair and trust my BT headphones, but cannot connect.
```
[bluetooth]# paired-devices
Device 70:BF:92:D8:82:84 Jabra Evolve2 65
Device 08:B7:38:11:A6:C2 HP Bluetooth Keyboard K4000
[bluetooth]# connect 70:BF:92:D8:82:84
Attempting to connect to 70:BF:92:D8:82:84
Failed to connect: org.bluez.Error.Failed
```
It has been a while since I tried this, but it may have been before my keyboard pairing incidents. So, now BT is quite broken. No audio in or out and many many attempts to connect and pair a keyboard. Not good. Tried connecting a phone as a source and a pair of BT headphones as a sink. Neither work.

macmpi, 2021-05-29:
In my case (AlpineLinux+Bluez+Bluealsa), audio is fine (Pi source to BT headphone). But media keys won't work.

gearhead, 2021-05-30:
I am in the process of trying older Bluez (5.55) and older firmwares. I will post if I find anything. As it is, with Bluez 5.55, I was able to pair the headphones but still not the keyboard.

macmpi, 2021-06-08:
Same on 5.58.
Worked last with 5.54-based setup

gearhead, 2021-06-16:
@macmpi I posted to the other issue #55. Bluez 5.59 recently dropped and I updated and tried again and it does the same thing.  As for evtest, I am not running a GUI, really, so I do not have evtest. I just want to use one of the first BT devices, a keyboard... I get BT audio in and out to work fine, it just takes hours and many many attempts to finally get the keyboard to pair and I have no idea why.

macmpi, 2021-06-25:
@Vudentz just noticed something: bug is not present on x86-64, but is present on x86 (same PC and headset in both case).
BTW my Pi is also on 32bit armhf build...
Would you mind replicating your previous test on x86 too?
Thanks in advance.

macmpi, 2021-07-02:
@bgix I noticed you were after 32-bit fixes recently, maybe you would have some clues on this?
(btw have many compiler warning about depreciated `g_memdup` in case it could matter)

bgix, 2021-07-02:
> @bgix I noticed you were after 32-bit fixes recently, maybe you would have some clues on this?
> (btw have [many compiler warning](https://gitlab.alpinelinux.org/macmpi/aports/-/jobs/423450/raw) about deprecated `g_memdup` in case it could matter)

It is certainly an un-ending struggle to get printf formatters to work reliably across architectures.  Answers online suggest using %ld and their variants (%06ld, %8ld etc) and then casting both time_t and suseconds_t  to (long) as the most portable solution for the most hardware.  It would be nice to have a %z type formatter that we use for size_t, but I guess there can't be a unique formatter for every non-portable data type.

macmpi, 2021-07-02:
Right, though I don't think that `printf` issue has any direct impact on AVRCP key input now failing on 32-bit (albeit fine on 64-bit.
Any thought on that part?
I could confirm on Alpine builds, but I'm curious if it's a distro issue or bluez issue: would be nice to have a confirmation on another 32bit build.

bgix, 2021-07-02:
So our primary development architecture is little endian, so if you are compiling for a big endian system, it could be we are missing a native <--> network order conversion somewhere. Do you have a specific example of this AVRCP key input entry point in the code (or specific DBus call, with a specific u64 type "t" parameter for example)?  If the key is originating as a u64 in a user app, it would typically be kept in native format all the way up to the point it is marshalled into an OTA packet in network format.

bgix, 2021-07-02:
Oh, I must have misread something...  Is it failing to *compile* on your 32 bit system, or is it compiling, but malfunctioning?

macmpi, 2021-07-02:
Thanks a lot for looking into it.
If does compile fine on x86-64, x86 and armhf (despite some suspicious `g_memdup` warnings).
In each case, I can pair a BT heaset (bluez-alsa) and playback sound just fine.
However, headset AVRCP key-presses only flow fine onto x86-64 input system, they just vanish somewhere on x86 and armhf (`evtest` won't see them on those 2 platforms, although we see them with `btmon`).
Is this summary description clearer?

macmpi, 2021-07-08:
same issue on 5.60 on 32-bit x86.
Let me know what kind of trace/debug info would help sort this out.
Thanks.

macmpi, 2021-07-22:
Tested on glibc-based Archlinux32: issue does not show-up (exact same HW environment).

It might probably be **32-bit compatibility issue between bluez and musl 1.2** ([since time64 release in particular](https://musl.libc.org/time64.html))
AlpineLinux shifted to musl 1.2 after Alpine 3.11, which corresponds to this problem poping-up...

At least there's simple a test usecase now.

baybal, 2021-08-16:
I think this is not a MUSL specific problem

macmpi, 2021-08-16:
@baybal did you experience the same issue in a different condition? Any clue appreciated.

baybal, 2021-08-17:
@macmpi this is what I am getting as of either recent kernel version 5.13, or bluez 5.60

```
Aug 16 18:39:36 xps bluetoothd[16960]: profiles/audio/avctp.c:avctp_control_confirm() Control: Refusing unexpected connect
Aug 16 18:39:37 xps bluetoothd[16960]: profiles/audio/avctp.c:uinput_create() Can't open input device: No such file or directory (2)
Aug 16 18:39:37 xps bluetoothd[16960]: profiles/audio/avctp.c:init_uinput() AVRCP: failed to init uinput for GS
```

macmpi, 2021-08-17:
Hum, never got any error message in my case.
I see you are probably on `xps`, so I imagine this is a 64-bit build?
I did not get any problem on 64-bit, nor on Archlinux32. Issue showed-up only on 32-bit Alpine (x86 & armhf).

baybal, 2021-08-18:
Maybe a different issue then.

macmpi, 2021-08-23:
I enclose bluetooth logs below on Alpine 64-bit (OK) and Alpine 32-bit (KO): in both cases I connect BT headset and press play/pause headset keys and observe them filling-in (or not) PC input system.

Both logs look pretty much similar: the only noticeable difference I can see may be the reported **registered endpoint sender**
- **1.12** on 64-bit (and key events flow fine into input system)
-  **1.5** on 32-bit (and key events vanish)

Both tests are made on very same hardware (x86 PC and BT headset)
I assume reported sender should be same with both implementations, right?
Could such difference adversely affect keys event pass-through on 32-bit case?

[LOG64_OK.txt](https://github.com/bluez/bluez/files/7032514/LOG64_OK.txt)
[LOG32_KO.txt](https://github.com/bluez/bluez/files/7032515/LOG32_KO.txt)

macmpi, 2021-09-08:
>  I believe you have a device specific problem here

@Vudentz I did test again with few other AVRCP devices (Headsets, Speakers) and same problem occurs.
Problem does not show-up with a remote control device like original Amazon FireTV remote (no mic): all 3 inputs do work as expected (media remote, keyboard, mouse).
It does indeed seem an AVRCP-specific issue.

macmpi, 2021-09-15:
Referring to [glibc Y2038 Proofness guideline](https://sourceware.org/glibc/wiki/Y2038ProofnessDesign), there possibly are some potential _time64_ compatibility issues in bluez (i.e some incompatible stucts like `timeval` or [`timespec` and time manipulations](https://github.com/bluez/bluez/blob/master/src/device.c#L2739-L2750)).

musl 1.2 has done [time64 transition on all arch](https://musl.libc.org/time64.html), and the issue I now see is only on 32-bit instances running bluez.
Could a `time64`-related timing issue swallow AVRCP events in the process (or some other non-explicit 64/32 int casting cancel data)?

No idea how I can track that down further, as base logs do not show any clue...
Suggestions?

macmpi, 2021-09-23:
BTW, it seems Jan 2020 [[RFC] y2038: HCI_TIME_STAMP with time64](https://www.spinics.net/lists/y2038/msg04953.html)  thread did not materialize into actual commits on this mater ([`/monitor/hcidump.c`, `/tools/hcidump.c`](https://www.spinics.net/lists/linux-bluetooth/msg82773.html) Arnd Bergmann patches, [`libpcap`](https://www.spinics.net/lists/y2038/msg04959.html) @guyharris)

arndb, 2021-10-13:
There is a general problem with any interface reading 'input_event' structures from the /dev/input/event device, this is unfortunately not fixable in the kernel, which has no idea whether the task that processes the event uses 32-bit or 64-bit time_t.

In old kernels, the structure was defined as

```
struct input_event {
        struct timeval time;
        __u16 type;
        __u16 code;
        __s32 value;
};
```

In new versions, this has become

```
struct input_event {
#if (__BITS_PER_LONG != 32 || !defined(__USE_TIME_BITS64)) && !defined(__KERNEL__)
        struct timeval time;
#define input_event_sec time.tv_sec
#define input_event_usec time.tv_usec
#else
        __kernel_ulong_t __sec;
#if defined(__sparc__) && defined(__arch64__)
        unsigned int __usec;
#else
        __kernel_ulong_t __usec;
#endif
#define input_event_sec  __sec
#define input_event_usec __usec
#endif
        __u16 type;
        __u16 code;
        __s32 value;
};
```

Any application or library that uses the old version with a private copy of the old definition must be updated to the new definition in order to be compatible with the kernel's layout regardless of the 'timeval' structure definition provided by the C library. At the same time, it should be changed to not be affected by the y2038 overflow of input_event_sec. This can be done using unsigned values consistently (which moves the limit to year 2106) or (better) by forcing the timestamps to use CLOCK_MONOTONIC with the EVIOCSCLOCKID ioctl command.

I don't know if this is what you are hitting here, but would be the most likely root cause.

Vudentz, 2021-11-22:
```
diff --git a/src/uinput.h b/src/uinput.h
index 75d791bc6..c304fdc21 100644
--- a/src/uinput.h
+++ b/src/uinput.h
@@ -698,7 +698,22 @@ struct uinput_dev {
 };
 
 struct uinput_event {
+#if (__BITS_PER_LONG != 32 || !defined(__USE_TIME_BITS64)) \
+       && !defined(__KERNEL__)
        struct timeval time;
+#define input_event_sec time.tv_sec
+#define input_event_usec time.tv_usec
+#else
+       __kernel_ulong_t __sec;
+#if defined(__sparc__) && defined(__arch64__)
+       unsigned int __usec;
+       unsigned int __pad;
+#else
+       __kernel_ulong_t __usec;
+#endif
+#define input_event_sec  __sec
+#define input_event_usec __usec
+#endif
        uint16_t type;
        uint16_t code;
        int32_t value;

```

Looks like the above change is required, but Im starting to wonder if it wouldn't have been better to just include <linux/input.h> instead of coming up with our own definitions @holtmann.

closed 2021-11-22 by BluezTestBot

macmpi, 2021-11-24:
@Vudentz I confirm this commit does fix the original AVRCP issue on 32-bit x86, on Alpinelinux/musl1.2
Thanks to you and @arndb .