bluez/bluez #85: profile caching causes issues with some devices

rene-dev, 2021-01-18:
sony wh-1000xm4 changes profile when multipoint is enabled in the app.
this stops a2dp from working while the old profile is cached.
https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/479

Vudentz, 2021-01-18:
Looks like we are not removing the endpoits when they disappear which is probably why they still appear as valid, I will investigate how we can detect when endpoits disappear like this.

Vudentz, 2021-01-21:
@rene-dev Can you test the above patch? I don't have a WH-1000XM4 to test this.

sheepdestroyer, 2021-01-29:
I have the same headset and have been trying bluez master with this commit.
I veryfied working state when connecting with multipoint enabled on the headset.
I then disconnected, disabled multipoint and reconnected to the computer. This time I don't see A2DP or HSP profiles and sound is not working.

messages in logs :
````
janv. 29 19:59:27 sheepora-X230 bluetoothd[3201]: profiles/audio/avdtp.c:avdtp_parse_rej() SET_CONFIGURATION request rejected: Invalid Capabilities (26)
janv. 29 19:59:27 sheepora-X230 bluetoothd[3201]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
````

Vudentz, 2021-03-10:
> We just discovered on Amlogic 4.9.113 kernel and bluez 5.56 with Kodi v19 that this commit: https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=3c65f88d47346ff04a1201a6274dac3dd96247d9
> does break the headphone bluetooth connection. It just connect the device and one second later it get disconnected.

Do you have the bluetoothd logs?

Portisch, 2021-03-10:
Not yet, I need to check one part tomorrow. Maybe we block the interface-added callback function too long. This could be a reason as we have a one second delay there. I will come back with more info.

Portisch, 2021-03-11:
Looks like the device just get removed by `remove_disappeared` after connect.

5.56 with revert 3c65f88d47346ff04a1201a6274dac3dd96247d9

```text
CoreELEC:~ # journalctl --unit=bluetooth -f
-- Journal begins at Thu 2021-03-11 06:33:51 UTC. --
Mar 11 06:34:04 CoreELEC bluetoothd[4421]: plugins/hostname.c:update_name() name: CoreELEC
Mar 11 06:34:04 CoreELEC bluetoothd[4421]: src/adapter.c:adapter_set_name() name: CoreELEC
Mar 11 06:34:04 CoreELEC bluetoothd[4421]: plugins/hostname.c:property_changed() chassis:
Mar 11 06:34:08 CoreELEC bluetoothd[4421]: src/agent.c:add_default_agent() Default agent set to :1.15 /CoreELEC/bt_agent
Mar 11 06:34:08 CoreELEC bluetoothd[4421]: src/adapter.c:set_mode() sending set mode command for index 0
Mar 11 06:34:08 CoreELEC bluetoothd[4421]: src/agent.c:agent_ref() 0x124c570: ref=1
Mar 11 06:34:08 CoreELEC bluetoothd[4421]: src/agent.c:register_agent() agent :1.15
Mar 11 06:34:08 CoreELEC bluetoothd[4421]: src/adapter.c:new_settings_callback() Settings: 0x00000ad1
Mar 11 06:34:08 CoreELEC bluetoothd[4421]: src/adapter.c:settings_changed() Changed settings: 0x00000010
Mar 11 06:34:08 CoreELEC bluetoothd[4421]: src/adapter.c:settings_changed() Pending settings: 0x00000000
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: src/adapter.c:connected_callback() hci0 device F4:B6:88:0E:A8:53 connected eir_len 16
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/a2dp.c:confirm_cb() AVDTP: incoming connect from F4:B6:88:0E:A8:53
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_F4_B6_88_0E_A8_53: SINK_STATE_DISCONNECTED -> SINK_STATE_CONNECTING
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avdtp.c:avdtp_connect_cb() AVDTP: connected signaling channel to F4:B6:88:0E:A8:53
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avdtp.c:avdtp_connect_cb() AVDTP imtu=672, omtu=895
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 1 type 1 media 0 delay_reporting false
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_F4_B6_88_0E_A8_53/sep1
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 2 type 1 media 0 delay_reporting false
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_F4_B6_88_0E_A8_53/sep2
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 3 type 1 media 0 delay_reporting false
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_F4_B6_88_0E_A8_53/sep3
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avdtp.c:avdtp_ref() 0x124a6e8: ref=1
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avdtp.c:set_disconnect_timer() timeout 1
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avctp.c:avctp_confirm_cb() AVCTP: incoming connect from F4:B6:88:0E:A8:53
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avctp.c:avctp_set_state() AVCTP Connecting
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avctp.c:avctp_connect_cb() AVCTP: connected to F4:B6:88:0E:A8:53
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avctp.c:init_uinput() AVRCP: uinput initialized for PLT Focus
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avrcp.c:controller_init() 0x1252398 version 0x0104
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: src/service.c:change_state() 0x12445d0: device F4:B6:88:0E:A8:53 profile audio-avrcp-target state changed: disconnected -> connected (0)
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avrcp.c:target_init() 0x124b168 version 0x0104
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: src/service.c:change_state() 0x1240c20: device F4:B6:88:0E:A8:53 profile avrcp-controller state changed: disconnected -> connected (0)
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avctp.c:avctp_set_state() AVCTP Connected
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avrcp.c:handle_vendordep_pdu() AVRCP PDU 0x10, company 0x001958 len 0x0001
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avrcp.c:avrcp_handle_get_capabilities() id=2
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avrcp.c:handle_vendordep_pdu() AVRCP PDU 0x10, company 0x001958 len 0x0001
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avrcp.c:avrcp_handle_get_capabilities() id=3
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avrcp.c:handle_vendordep_pdu() AVRCP PDU 0x31, company 0x001958 len 0x0005
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avrcp.c:handle_vendordep_pdu() AVRCP PDU 0x31, company 0x001958 len 0x0005
Mar 11 06:34:34 CoreELEC bluetoothd[4421]: profiles/audio/avrcp.c:handle_vendordep_pdu() AVRCP PDU 0x11, company 0x001958 len 0x0000
Mar 11 06:34:36 CoreELEC bluetoothd[4421]: profiles/audio/avdtp.c:avdtp_ref() 0x124a6e8: ref=2
Mar 11 06:34:36 CoreELEC bluetoothd[4421]: profiles/audio/avdtp.c:avdtp_ref() 0x124a6e8: ref=3
Mar 11 06:34:36 CoreELEC bluetoothd[4421]: profiles/audio/a2dp.c:setup_ref() 0x12470f8: ref=1
Mar 11 06:34:36 CoreELEC bluetoothd[4421]: profiles/audio/avdtp.c:avdtp_unref() 0x124a6e8: ref=2
Mar 11 06:34:36 CoreELEC bluetoothd[4421]: profiles/audio/a2dp.c:discover_cb() version 0x0102 err (nil)
[... cut ...]
Mar 11 06:39:55 CoreELEC bluetoothd[4446]: src/adapter.c:adapter_remove_connection()
Mar 11 06:39:55 CoreELEC bluetoothd[4446]: plugins/policy.c:disconnect_cb() reason 3
Mar 11 06:39:55 CoreELEC bluetoothd[4446]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr F4:B6:88:0E:A8:53 type 0 status 0xe
Mar 11 06:39:55 CoreELEC bluetoothd[4446]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
Mar 11 06:39:55 CoreELEC bluetoothd[4446]: src/device.c:device_bonding_failed() status 14
Mar 11 06:39:55 CoreELEC bluetoothd[4446]: src/adapter.c:resume_discovery()
```

lpilz, 2021-08-03:
On Arch, bluez-git was just updated to 5.60 and now HSP is unavailable on my headphones even though HFP is available. I am seeing the same error:
```
Aug 03 12:57:05 <pc> bluetoothd[1433]: /org/bluez/hci0/dev_70_BF_92_38_2A_99/sep1/fd0: fd(28) ready
Aug 03 14:06:01 <pc> bluetoothd[1433]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
```
What is the status on a fix for this issue and do you know when this will be released? Thanks!

closed 2021-09-01 by Vudentz